openthread/openthread #9888: Add ICMPv6 Destination Unreachable sending for Border Router (SPEC-1218)

EskoDijk, 2024-03-01:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
Problem stated in https://threadgroup.atlassian.net/browse/SPEC-1218 : 
a Thread Device that only has a ULA OMR address (BR-generated prefix), may try to contact a cloud server continuously even though it will never work with such address type. This happens in case the BR publishes a ::/0 prefix on the mesh that suggests the cloud server is reachable, while that ::/0 only is meant to communicate reachability of a non-ULA-prefix on the AIL.

Similarly, regardless of Thread Device address type, if the BR doesn't have a route to the destination, but as before the ::/0 route was published, the message will be dropped but the sender not notified - sender may assume congestion, so may keep trying forever.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
As proposed in SPEC-1218: BR to send ICMPv6 Destination Unreachable message in such cases.
* If there's no route possible to destination, Code 0 can be sent
* If there's a route possible, but BR knows destination is not on the AIL, so not reachable using the ULA OMR address, Code 1 can be sent (see SPEC-1218)

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
See SPEC-1218

**Additional context** Add any other context or screenshots about the feature request here.
I didn't check the code in detail on what is already implemented with ICMPv6 error messages - but it seems there's only some code related to ICMPv6 errors for on-mesh destination and not for this case.

abtink, 2024-07-30:
Submitted 
- https://github.com/openthread/openthread/pull/10548

EskoDijk, 2024-08-29:
#10548 was merged, closing this issue.

closed 2024-08-29 by EskoDijk

abtink, 2024-08-29:
One thing that may still make sense to add related to this is "rate limiting" logic.

> In case the same source Thread Device (as identified by IPv6 source address) repeatedly sends packets to the same destination address, causing ICMPv6 Destination Unreachable messages to be sent back to the sender every time, a Border Router MAY temporarily suppress the sending of further ICMPv6 error messages to that source. This is allowed to avoid excessive traffic load on the mesh.

I may add it in a PR at a later point.