openthread/openthread #9910: SRP client registers mesh-local IP address in some topology scenarios

ndyck14, 2024-03-09:
SRP client registers mesh-local IP address in some topology scenarios

Information to reproduce the behavior, including:

1. Git commit id e6df00d (https://github.com/SiliconLabs/matter/releases/tag/v2.2.0-1.2)
2. IEEE 802.15.4 hardware platform MG24
3. Build steps
4. Network topology 2 BRs, ~10 FTDs.

I had an OTBR device as leader with SRP disabled, and another BR. I disabled BR function on the leader, and the other BR prefix has not arrived in the netdata after some time.

I believe addresses registered by SRP clients should only be OMR type addresses? Or is it intended that mesh-local addresses are registered if there are no OMRs

I observe this from proxied records: 
```
Nanoleaf A19 751-BF5C
E243C061865B58C8.local.
[fd00:cafe:1eaf:0:e3e3:4401:39a3:8935]:5653
```
The IP address is mesh-local on my network.

The precipitating pre condition here seems to be a misbehaving BR who isnt adding its OMR prefix (or a leader who isnt accepting it, not sure yet).

jwhui, 2024-03-10:
SRP Client will register the mesh local address if there are no global scope (e.g. OMR) addresses. This conforms to the Thread Specification - see [SPEC-1070](https://threadgroup.atlassian.net/browse/SPEC-1070).

ndyck14, 2024-03-12:
Makes sense. This does indeed happen with anycast, so the assumption is the SRP servers live on mesh and can at least make use of the services themselves.

closed 2024-03-12 by ndyck14