openthread/openthread #9930: [Thread 1.3.x / 1.4] related OT PRs and Thread JIRAs

EskoDijk, 2024-03-14:
Created this issue to track OT PRs and related Thread Technical Committee JIRAs, that are not yet listed in [8461](https://github.com/openthread/openthread/issues/8461).

TBD: SPEC Items that still need to be checked are SPEC-1149 to SPEC-1201; and any 1.4 items.

- [ ] [SPEC-1218](https://threadgroup.atlassian.net/browse/SPEC-1218) - default route when Thread has only ULA OMR prefix
   - https://github.com/openthread/openthread/issues/9888
   - No PR yet / Not verified if already implemented
- [ ] [SPEC-1086](https://threadgroup.atlassian.net/browse/SPEC-1086) - BR attempts DHCPv6-PD when only low-prio OMR prefixes present
   - No PR yet / Not verified if already implemented
- [ ] [SPEC-1133](https://threadgroup.atlassian.net/browse/SPEC-1133) - BR using DHCPv6-PD must monitor AIL for change
   - No PR yet / Not verified if already implemented
- [ ] [SPEC-1238](https://threadgroup.atlassian.net/browse/SPEC-1238)   - new reqs on TREL host name and instance name
   - No PR yet / Not verified if already implemented
- [ ] [SPEC-1229](https://threadgroup.atlassian.net/browse/SPEC-1229) - new rules for Unicast Dataset withdrawal
   - No PR yet 
- [ ] [SPEC-1211](https://threadgroup.atlassian.net/browse/SPEC-1211) - BR adv proxy KEY handling
   - No PR yet / Not verified if already implemented
- [ ] [SPEC-1202](https://threadgroup.atlassian.net/browse/SPEC-1202) - added bits in mDNS sb state bitmap
   - No PR yet / Not implemented yet

_Note: Access to the Thread JIRA links requires Thread group membership and Thread group login/account._

marius-preda, 2024-09-04:
Hi @abtink, when are you planning to change the version of Open Thread to 5 (1.4) ? From this list I see the DNS-over-TCP/TLS is last big missing item. Are you planning adding support for this feature?

jwhui, 2024-09-04:
> When are you planning to change the version of Open Thread to 5 (1.4)?

I think we can make the change soon.

> From this list I see the DNS-over-TCP/TLS is last big missing item. Are you planning adding support for this feature?

OpenThread DNS client supports TCP, but we still lack TLS support on the client and TCP/TLS on the server. Contributions are welcome!