openthread/openthread #9964: Intermittent CoAP ACK Reception Issues with Multiple nRF52840 Nodes in Thread Network

cedriczou13, 2024-03-25:
Hello,

I am experiencing an issue with my OpenThread network that consists of a Raspberry Pi acting as a border router and multiple nRF52840 nodes. The setup works perfectly with a single child node sending CoAP messages to a cloud server every 30 seconds, with acknowledgments (ACKs) being consistently received. However, when I expand the network by adding four more nRF52840 nodes, I start facing intermittent problems. Specifically, after some time, one or more nodes stop receiving ACKs for their CoAP messages, leading to unreliable data transmission.

### Network Setup:
- **Border Router**: Raspberry Pi (serving as the OpenThread Border Router)
- **Child Nodes**: 5 x nRF52840 boards, configured to send CoAP messages to a cloud server every 30 seconds.

### Observed Behavior:
- With a single child node, there are no issues.
- Adding four more nodes initially works fine, but eventually, one or more nodes intermittently fail to receive ACKs for their CoAP messages.
- Temporarily resolving the issue by resetting the affected device(s), after which they begin to receive ACKs again, but the problem eventually recurs.

I appreciate any guidance or insights you can provide to help resolve this intermittent ACK reception problem. Thank you for your support.

jwhui, 2024-03-25:
What git commit are you using?

Can you provide logs and packet captures to provide more visibility?

jwhui, 2024-06-10:
Closing stale issue.

closed 2024-06-10 by jwhui