openthread/openthread #10008: How should the payload data be packaged when using CoAP for PUT/POST?

sagexy, 2024-04-09:
**Describe the bug** A clear and concise description of what the bug is:

CoAP put command does not seem work, is the payload I entered  incorrect? What is the payload format should be?

**To Reproduce** Information to reproduce the behavior, including:

When I use the coap get command, it return the below message:

```
coap get fdaf:518:5fdc:1:8dc4:600c:8a4e:ba7f p/sdev/wakeupcycle
Done
> coap response from fdaf:518:5fdc:1:8c4:600c:8a4e:ba7f with payload: 363030
```
(Actually, the data is 600, decode "363030" is 600, it returns the correct value)

But when I use the put command, it does not return any payload and when use get command again, the configuration remains unchanged.

```
coap put fdaf:518:5fdc:1:8dc4:600c:8a4e:ba7f p/sdev/wakeupcycle con 363032
Done
> coap response from fdaf:518:5fdc:1:8c4:600c:8a4e:ba7f
coap get fdaf:518:5fdc:1:8dc4:600c:8a4e:ba7f p/sdev/wakeupcycle
Done
> coap response from fdaf:518:5fdc:1:8c4:600c:8a4e:ba7f with payload: 363030
```

The payload I entered, either '363032' or '602', does not work at all. So I want to transfer the payload with data "602", what should do?

1. Git commit id: 2793017a
3. IEEE 802.15.4 hardware platform: nRF52840 USB dongle
4. Build steps
5. Network topology: nRF52840 dongle <---> thread device

**Expected behavior** A clear and concise description of what you expected to happen.

Use CoAP put command with payload can work well.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

sagexy, 2024-04-10:
From the wireshark capture package, the payload format was application/octet-stream not application/json, so the server not unsupported this content-format
[image]

[image]

abtink, 2024-04-10:
@sagexy OT stack itself does not specify any format for Application CoAP payload.

The CLI `coap` commands are just a limited example of how CoAP APIs can be used. It is up to next layer (user of OT API) to specify the format (based on URI, etc).

closed 2024-04-12 by sagexy