openthread/openthread #10048: Request for a new hook that occurs after Host gets RCP capabilities

lmnotran, 2024-04-19:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.

RCPs don't have a hook that's executed after the Host finishes initializing RCP capabilities.

In my use case, I'm trying to print some crash logs from the RCP using `LOG_STREAM` spinel messages to the Host after recovering from a crash.

**Describe the solution you'd like** A clear and concise description of what you want to happen.

I think there are many ways to implement something that would address the feature request. Here are a couple independent solutions I have in mind:

- Add a new [Notifier event](https://github.com/openthread/openthread/blob/fa6477197c61a7e69e47e9e8dd9a032cd38a709b/src/core/common/notifier.hpp#L67-L101) `kRcpCapabilitiesSentToHost`.
  - On the RCP, this event would be emitted at the end of [`NcpBase::HandlePropertyGet<SPINEL_PROP_CAPS>()`](https://github.com/openthread/openthread/blob/fa6477197c61a7e69e47e9e8dd9a032cd38a709b/src/ncp/ncp_base.cpp#L2068-L2069) if no error has occurred.
  - This would allow a vendor to use the event to send crash logs or any other logs upon a Host establishing a connection to the RCP using [`ExtensionBase::HandleNotifierEvents()`](https://github.com/openthread/openthread/blob/fa6477197c61a7e69e47e9e8dd9a032cd38a709b/src/core/common/notifier.cpp#L175)
  - Nothing would need to be added to the Host code.

- Add a new spinel capability and property `SPINEL_PROP/CAP_RCP_INIT_COMPLETE`.
  - After the Host gets the RCP capabilities, if a RCP has this capability, the Host sends `SPINEL_PROP_RCP_INIT_COMPLETE` to RCP.
  - The handler for this in `ncp_base_radio.cpp` would call a new `otPlatRcpInitComplete()` platform API.
    - I recognize that Abtin [expressed some concern in my PR](https://github.com/openthread/openthread/pull/9829#discussion_r1478940471) about how `otPlat...()` APIs are intended for OT core modules use. 
    - It looks like [`otPlatMultipanGet/SetActiveInstance()`](https://github.com/search?q=repo%3Aopenthread%2Fopenthread%20otPlatMultipanGetActiveInstance&type=code) is being used in a similar way to how I invision `otPlatRcpInitComplete()` being used. `otPlatMultipanGet/SetActiveInstance()` is only being called in [`NcpBase::HandlePropertyGet<SPINEL_PROP_MULTIPAN_ACTIVE_INTERFACE>()`](https://github.com/openthread/openthread/blob/4737231d64d262eb850e0da729d4b2df130033e1/src/ncp/ncp_base_radio.cpp#L589)
    - Alternatively, a callback could be added to the `NcpBase` class instead of adding a new `otPlat...()` API

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.

- Adding a new spinel capability and property specifically for logging crash dumps
  - I proposed this in https://github.com/openthread/openthread/pull/9829
  - This approach would have allowed for the Host to request the latest crash dump from the RCP
- @zhanglongxia suggested in (https://github.com/openthread/openthread/pull/9829#issuecomment-1925563823) that using [`ExtensionBase::SignalNcpInit()`](https://github.com/openthread/openthread/blob/main/src/core/instance/extension_example.cpp#L79) might be a better approach.
  - I implemented this and upon initial testing using a `diag` command to trigger a fault, this solution works since the RCP crashes and resets while the Host has an existing/established connection to the RCP. See 
    - See log [ot-cli-syslog_ot-rcp-using-vendor-hook-to-print-logs.txt](https://github.com/openthread/openthread/files/15045096/ot-cli-syslog_ot-rcp-using-vendor-hook-to-print-logs.txt)
 
  - After more thorough testing, I discovered that there are still some scenarios where the RCP can send `LOG_STREAM` spinel messages to the Host before the Host is aware of the RCP capabilities, causing the logs to be missed on the Host.
  - Specifically, this occurs when I configure the RCP to crash every 2 seconds. Since the [`ExtensionBase::SignalNcpInit()`](https://github.com/openthread/openthread/blob/main/src/core/instance/extension_example.cpp#L79) hook is called at the end of the [NcpBase::NcpBase() constructor](https://github.com/openthread/openthread/blob/fa6477197c61a7e69e47e9e8dd9a032cd38a709b/src/ncp/ncp_base.cpp#L361), it's possible for the hook to be called before `ot-cli` has been started on the Host.

**Additional context** Add any other context or screenshots about the feature request here.

lmnotran, 2024-04-19:
@jwhui @zhanglongxia @abtink Could you take a look at this please?

zhanglongxia, 2024-04-22:
@lmnotran  If RCP uploads a lot of crash logs in the [ExtensionBase::SignalInstanceInit](https://github.com/openthread/openthread/blob/main/src/core/instance/extension_example.cpp#L79) after it restarts, it will take much time. The Host has a high possibility of sending a Spinel command to RCP during this time. The longer the time, the higher the possibility. When the RCP is uploading the logs and the Spinel module is not ready in the RCP side. The received Spinel command will be dropped, and no Spinel response will be sent to the host. Then the host will exit and restart. Some of the crash log will be lost.

I think there are to methods to resolve this issue:

<1> Remove useless logs and combine all crash logs into one log to reduce the uploading time. But this cannot completely solve the problem.

<2> Add a spinel command to fetch the crash logs from the RCP when the host is ready.

lmnotran, 2024-04-22:
@zhanglongxia I agree that Option 2 is probably the correct way to fix this. Should I reopen my PR #9829 since it implements this?

zhanglongxia, 2024-04-23:
> @zhanglongxia I agree that Option 2 is probably the correct way to fix this. Should I reopen my PR #9829 since it implements this?

Sure.

PR #10061 (2024-04-29): [spinel] add support for logging crash dumps
Supersedes #9829
Closes #10048
---

This PR adds a new feature that allows platforms to log crash logs.

### Additions
- `void otPlatLogCrashDump(void)` - API that logs a crash dump using OpenThread Logging APIs
- `SPINEL_PROP_RCP_LOG_CRASH_DUMP` - spinel prop that calls `otPlatLogCrashDump()` when `Set`
- `SPINEL_CAP_RCP_LOG_CRASH_DUMP` - spinel capability denoting that a RCP supports logging crash dumps

### Usage
- For `ot-cli-ftd|mtd`, `otPlatLogCrashDump()` is called at the end of app initialization, before the main loop. See [main.c]
- For Hos [... cut]
closed 2024-04-29 by jwhui