openthread/openthread #10253: Migrate Mbed TLS platform to PSA APIs

gilles-peskine-arm, 2024-05-16:
Mbed TLS is in the process of migrating its [cryptography API](https://github.com/Mbed-TLS/mbedtls/blob/mbedtls-3.6.0/docs/psa-transition.md) from “legacy” APIs to [PSA Crypto API](https://arm-software.github.io/psa-api/crypto/1.1/). The timeline for the transition is:

* Mbed TLS 2.28 LTS (the oldest supported version at this time) has PSA API support for every algorithm that OpenThread uses.
* Mbed TLS 4.0, currently planned to be released in Q2 2025, will drop support for many of the legacy crypto APIs that OpenThread currently uses.

So OpenThread should add PSA API support within the next year. I don't know if you need to retain legacy APIs for a transition period.

Here are a few tips about the features that OpenThread uses:

* You need to call `psa_crypto_init()` before anything else.
* As of Mbed TLS 2.x/3.x, there is no way to use the PSA API without an RNG. This will be possible in Mbed TLS 4.0 by calling [`psa_crypto_init_subsystem`](https://github.com/Mbed-TLS/mbedtls/issues/6007) instead of `psa_crypto_init`.
* `psa_{sign,verify}_hash` functions handle signatures in the fixed-size (r,s) format that OpenThread wants, rather than the ASN.1 format used in the legacy Mbed TLS APIs.
* Since Mbed TLS 3.5, PBKDF2-AES-CMAC-128 is available in Mbed TLS, only through the PSA API.

jwhui, 2024-05-16:
@gilles-peskine-arm , thanks for submitting this issue. We are planning to migrate to PSA APIs, so this is useful input.

nfarouk, 2025-04-20:
@jwhui when is this migration planned?

LuDuda, 2025-04-22:
Feature breakdown plan:

- [ ] Implement the default OpenThread support for OPENTHREAD_CONFIG_CRYPTO_LIB_PSA using the pure PSA API
- [ ] Align cryptographic unit tests to work with the PSA API
- [ ] Add support for the PSA Crypto API on the simulation platform
- [ ] Introduce a default Native ITS implementation for the simulation and POSIX platforms
- [ ] Enable the PSA Crypto API (by default) for the simulation platform (CI checks)
- [ ] Remove CI checks for the legacy mbedTLS 2.x

LuDuda, 2026-01-22:
@nbashirfr Just posted another PR. Apart of this, there are two more left from the original PR to enable PSA API for simulation platform. So hopefully, we can have them in soon.