openthread/openthread #10261: Coap Get blockwise request failed with Parse error

arnoldlv1976, 2024-05-17:
Hi There

I tried to execute Coap Get request to some block-wise resource which is for example 1025 bytes in size. I faced of the same wrong behaviour via the both CLI interface and Copa Api. If I try to send the Get block-wise request with the block length of 1024 bytes to upload it it succeeded (that's because the entire transaction takes only two consecutive requests) but if I use the block size less (128, 256,512,....) it failed with Parse error because it takes more then two consecutive requests. I used gecko_sdk_4.4.1
 

This example:
1. failed case!!!
>coap start
Done
>coap get 2001:db8:abcd:12:a882:297a:bbc4:8d91 oadimage block-128
Done
received block: Num 0 Len 128
4142434445464748494a4b4c4d4e4f50
5152535455565758595a2d3031323334
........
58595a414243444546474849474b4c4d
received block: Num 1 Len 128
4e4f505152535455565758595a414243
444546474849474b4c4d4e4f50515253
........
565758595a414243444546474849474b
coap receive response error 6: Parse

2. succeeded case!!!
> coap get 2001:db8:abcd:12:a882:297a:bbc4:8d91 oadimage block-1024
Done
received block: Num 0 Len 1024
4142434445464748494a4b4c4d4e4f50
5152535455565758595a2d3031323334
.........
474b4c4d4e4f50515253545556575859
received block: Num 1024 Len 1
coap response from 2001:db8:abcd:12:a882:297a:bbc4:8d91 with payload: 5a

 I investigated the files and could see the next:
1. The result of executing the Coap Get block-wise request is that the Coap engine starts sending sequential requests of the specified block length to read the entire amount of data.
2. For each subsequent request for the next data block, options are generated based on the options of the previous request. but always when generating the third sequential request, these options are read incorrectly. and this leads to this error
3!!!!. I see if I add payload marker to the end of the next subsequent request It parses properly. I made some temporary fix in ../gecko_sdk_4.4.1/util/third_party/openthread/src/core/coap/coap.cpp file.

Error CoapBase::PrepareNextBlockRequest(Message::BlockType aType,
bool aMoreBlocks,
Message &aRequestOld,
Message &aRequest,
Message &aMessage)
{
....
if (!isOptionSet)
{
// Write Block1 option to next message
SuccessOrExit(error = aRequest.AppendBlockOption(aType, aMessage.GetBlockWiseBlockNumber() + 1, aMoreBlocks,
aMessage.GetBlockWiseBlockSize()));
aRequest.SetBlockWiseBlockNumber(aMessage.GetBlockWiseBlockNumber() + 1);
aRequest.SetBlockWiseBlockSize(aMessage.GetBlockWiseBlockSize());
aRequest.SetMoreBlocksFlag(aMoreBlocks);
}

// here is my fix!!!
SuccessOrExit(error = aRequest.SetPayloadMarker());

exit:
return error;
}

Finally I think the problem is here in coap_message.cpp :

Error Option::Iterator::Advance(void)
{
  Error  error = kErrorNone;
  uint8_t headerByte;
  uint16_t optionDelta;
  uint16_t optionLength;

  VerifyOrExit(!IsDone());

 // read out options Not properly if we don't add payload marker as I did!!! I mean it will read out unexisting option after valid ones!!!
  error = Read(sizeof(uint8_t), &headerByte);
 
//error should be not kErrorNone after Read() reads out the last option from the prev request to go in this if() and then call ExitNow(error = kErrorNone). or we should add payload marke to the end of the each sequential request.

   if ((error != kErrorNone) || (headerByte == Message::kPayloadMarker))
  {
    // Payload Marker indicates end of options and start of payload.
    // Absence of a Payload Marker indicates a zero-length payload.

    MarkAsDone();
...
   ExitNow(error = kErrorNone);
  }
.....

 I've already asked about it in Silabs Thread forum: https://community.silabs.com/s/question/0D5Vm00000AmjhLKAR/coap-get-blockwise-request-failed-with-parse-error

Regards
Dmytro

jwhui, 2024-06-05:
Thanks for raising this issue. If you have a proposed fix, please feel free to submit a PR.

jwhui, 2025-01-07:
Closing stale issue.