openthread/openthread #10262: SRP Advertising Proxy Thread 1.3 Certification Question

Cristib05, 2024-05-17:
**Describe the bug** A clear and concise description of what the bug is.
Thread 1.3 Certification Test 2.1 is failing for the BR DUT on step 10 when an SRP Update is sent by the End Device.
Initially, SRP client is registering the host with 2 addresses, OMR and ML-EID, then it performs an SRP Update for the service, but this time only the ML-EID address is used.
SRP Update is refused two times [correlated with Response Timeout error from advertising proxy log] and it seems to be accepted only after the update announcements are performed. What would be the problem if the SRP Update will be accepted from first attempt?
SRP Advertising Proxy module is only adding addresses for a host if those are not link local or mesh local.
If there are no such addresses in the SRP Update, the mDNS module will then unregister that specific host and will announce the removal of the old addresses.
[image]

Second question is why the services that have been associated with that host are still kept and listed using "mdns services", when that host is not anymore present in "mdns hosts" output command?
[image]

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform : not platform dependent
3. Build steps
4. Network topology : As described in test plan.

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

abtink, 2024-05-17:
Thanks @Cristib05 for reporting this.

I think there may be an issue in native mDNS when registering host with no address (which is then treated as unregistering the host), that the callback may not be invoked. I will submit a PR to update the code to address this.

abtink, 2024-05-17:
Submitted
- https://github.com/openthread/openthread/pull/10264

closed 2024-05-20 by jwhui