openthread/openthread #10287: The state is always detached after change the channel

sagexy, 2024-05-23:
**Describe the bug** A clear and concise description of what the bug is.

**After change the channel, then the state is always detached can not be leader or router ...**

**To Reproduce** Information to reproduce the behavior, including:

```
thread stop
channel 15
state
```

1. Git commit id [8b04e9c](https://github.com/openthread/openthread/tree/8b04e9ca73ef3876748734cd1c4e002579bf5fce)
2. IEEE 802.15.4 hardware platform nRF52840 USB dongle
3. Build steps [OpenThread on nRF52840 examples](https://github.com/openthread/ot-nrf528xx/blob/main/src/nrf52840/README.md)
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Change channel successfully and after thread start, state can be leader or router**

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

abtink, 2024-05-23:
`thread stop` command will stop the Thread operation. 
Do you start it back (issuing `thread start`)?

sagexy, 2024-05-24:
> `thread stop` command will stop the Thread operation. Do you start it back (issuing `thread start`)?

Yes, of course, I executed the command `thread start`; otherwise the state will be `disabled`.

Even when I reconnect the dongle and execute
```
ifconfig up
thread start
```

The state still be detached

logs:
```
> state
state
disabled
Done
> ifconfig up
ifconfig up
Done
> thread start
thread start
Done
> state
state
detached
Done
> state
state
detached
Done
> state
state
detached
Done
> state
state
leader
Done
> channel
channel
16
Done
> thread stop
thread stop
Done
> channel 18
channel 18
Done
> thread start
thread start
Done
> state
state
detached
Done
> state
state
detached
Done
> state
state
detached
Done
>
.
.
.
.
```

abtink, 2024-05-24:
If you need to change the channel of a device that's already connected, using dataset commands and the API is the recommended approach. This typically involves working with the Pending Operational Dataset. Here are some resources that can help:
- [`dataset` CLI documentation](https://github.com/openthread/openthread/blob/e7535f7183db4a80ced0f4eb39e0c09bc89c6b7d/src/cli/README_DATASET.md#openthread-cli---operational-datasets) and in particular [Using the Dataset Updater to update Operational Dataset](https://github.com/openthread/openthread/blob/e7535f7183db4a80ced0f4eb39e0c09bc89c6b7d/src/cli/README_DATASET.md#using-the-dataset-updater-to-update-operational-dataset) section
- OpenThread also offers a "channel manager" module, which provides an API to initiate a network-wide channel change. This module handles the preparation of the Pending Dataset, simplifying the process for users. [`channel manager` CLI documention](https://github.com/openthread/openthread/blob/e7535f7183db4a80ced0f4eb39e0c09bc89c6b7d/src/cli/README.md#channel-manager-change-channel).

---

The `channel` CLI command, which corresponds to the `otLinkSetChannel()`, is a low-level API primarily designed for testing purposes. When this command is used, any previously saved Active or Pending Dataset is invalidated and erased ([API documenation](https://github.com/openthread/openthread/blob/e7535f7183db4a80ced0f4eb39e0c09bc89c6b7d/include/openthread/link.h#L504-L520
)). Employing it in the way you mention in the example can lead to unexpected behavior.

closed 2024-05-24 by sagexy