openthread/openthread #10302: SED loss of parent keeps radio on

krbvroc1, 2024-05-27:
I am performing power consumption profiling and testing different scenarios. I am using the nrf52840 MCU with the lastest stable NCS 2.6.1.  In our use case it is very likely our portable SED devices could be moved outside of the range of the thread network for several days and then return.

While testing I have a single SED and an OTBR. To simulate loss of parent, I am either killing the OTBR or just switching the OTBR to a detached mode (ot-ctl mode -)

I am finding really bad behavior. When the SED loses the parent, it goes into a cycle of child update request, data request, parent request.  However the radio is turned on and not turned off when expected! This will obviously deplete the battery.

Looking at the logs, I see the node transition to the 'detached' state and it changes to a 'RxOnWhenIdle 1' state.

It then sends two Parent requests to Routers, then five Parent requests to Routers and REEDs.
Next, while it is listening on Channel 26, I see it sending an Announce on Channel 11 through 26. This is about 10 seconds of the radio on. 

Next, it starts an exponential backoff repeating the above sequence. Here is where the problem gets really bad...

I am seeing the following in the logs...

```
[00:04:23.101,409] <inf> [I] Mle-----------: AttachState Announce -> Idle
[00:04:23.105,865] <dbg> [D] SubMac--------: RadioShortAddress: 0xfffe
[00:04:23.106,018] <inf> [I] Mle-----------: AttachState Idle -> Start
[00:04:23.106,201] <inf> [N] Mle-----------: Attach attempt 7 unsuccessful, will try again in 16.100 seconds
[00:04:23.106,353] <dbg> [D] SubMac--------: RxOnWhenIdle: 0
[00:04:23.106,475] <dbg> [D] SubMac--------: RadioState: Receive -> Sleep
[00:04:23.106,597] <dbg> [D] Mac-----------: Idle mode: Radio sleeping

 *** Radio is actually NOT turned off here ***

[00:04:39.206,451] <inf> [N] Mle-----------: Attach attempt 8, AnyPartition 
[00:04:39.206,604] <inf> [I] Mle-----------: AttachState Start -> ParentReq

```
Note that above the Radio is NOT turned off. It isn't until later that I see the following debug print and then I see the power consumption drop indicating the radio was actually turned off.
```
[00:05:27.096,557] <dbg> [D] Mac-----------: Frame rx failed, error:DestinationAddressFiltered, len:63, seqnum:49, type:Data, src:76d8030c1e93f953, dst:0xffff, sec:no, ackreq
```

In each case during this cycle where I see a log message indicate the radio moves to 'Sleep', it does not actually turn off until I see one of the 'Frame rx failed, error:DestinationAddressFiltered' message much later. I don't know what this message means. As the exponential backoff increases, the time it takes for the above DestinationAddressFiltered to show up and turn off the radio also increases.

Here is some info from logs later in the sequence when the exponential backoff has reach 20 minutes, the radio is on for 17 minutes.

```
[00:38:47.743,103] <dbg> [D] Mac-----------: Idle mode: Radio receiving on channel 26
[00:38:47.815,612] <inf> [I] Mle-----------: AttachState Announce -> Idle
[00:38:47.820,098] <dbg> [D] SubMac--------: RadioShortAddress: 0xfffe
[00:38:47.820,251] <inf> [I] Mle-----------: AttachState Idle -> Start
[00:38:47.820,404] <inf> [N] Mle-----------: Attach attempt 14 unsuccessful, will try again in 1200.084 seconds
[00:38:47.820,556] <dbg> [D] SubMac--------: RxOnWhenIdle: 0
[00:38:47.820,709] <dbg> [D] SubMac--------: RadioState: Receive -> Sleep
[00:38:47.820,800] <dbg> [D] Mac-----------: Idle mode: Radio sleeping

RADIO STILL ON 17 Minutes!

[00:55:40.194,519] <dbg> [D] Mac-----------: Frame rx failed, error:DestinationAddressFiltered, len:63, seqnum:125, type:Data, src:76d8030c1e93f953, dst:0xffff, sec:no, ackre

```
I have no idea what is going on here but this is a show stopper for a battery powered product. As a 'hack', I did wait for the thread_role 'detached' notification and call otPlatRadioSleep() which does shut the radio off.

abtink, 2024-05-27:
Thanks @krbvroc1.

I think this may be a radio driver (platform) issue. Here are two possibilities:

1. **Recent PR ([#9554](https://github.com/openthread/openthread/pull/9554)):** This change allows the radio platform to directly manage when to put the radio to sleep.
   - The radio indicates this capability by including `OT_RADIO_CAPS_RX_ON_WHEN_IDLE` in `otPlatRadioGetCaps()`.
   - It's worth checking if your platform is enabling and using this capability.
   - If so, there might be an issue in the radio platform implementation.

2. **OT `SubMac` Management:** If the platform isn't managing sleep, OT `SubMac` handles it.
   - The log message `"RadioState: Receive -> Sleep"` is emitted only after a successful call to `otPlatRadioSleep()`, which instructs the radio to enter sleep mode. ([SubMac::Sleep()](https://github.com/openthread/openthread/blob/a54f4c485071c4909a46483e89ba527789130357/src/core/mac/sub_mac.cpp#L230-L249))
   - If the radio doesn't enter sleep mode despite this log message, there could be an issue in the platform's implementation of the `otPlatRadioSleep()` API.

@LuDuda , @edmont , would you be able to take a look and provide your input on this? Thank you!

krbvroc1, 2024-05-27:
> 1. **Recent PR ([#9554](https://github.com/openthread/openthread/pull/9554)):** This change allows the radio platform to directly manage when to put the radio to sleep.
>
>        * The radio [...]

I have looked at the build I am using. While OT_RADIO_CAPS_RX_ON_WHEN_IDLE is set in the return value (0x1FB) from otPlatRadioGetCaps(), I do not see that OPENTHREAD_CONFIG_MAC_SOFTWARE_RX_ON_WHEN_IDLE_ENABLE was chosen during the compilation.

I am confused by the following code:
https://github.com/openthread/openthread/blob/a54f4c485071c4909a46483e89ba527789130357/src/core/config/mac.h#L408-L416

I don't see the define used anywhere in openthread other than above.  And even the comment is confusing. Is says enable software RX OFF, yet the define is about RX ON.

In sub_mac.cpp, I see ShouldHandleTransitionToSleep() calls RadioSupportsRxOnWhenIdle() which returns true if the capability is set. Even though the capability is set, I don't think OPENTHREAD_CONFIG_MAC_SOFTWARE_RX_ON_WHEN_IDLE_ENABLE was set during compilation.

Setting some breakpoints in the software, even though there is the print of RadioState: Receive -> Sleep, the call to otPlatRadioSleep() is NOT triggering. Maybe due to the above ShouldHandleTransitionToSleep()?

abtink, 2024-05-27:
If the radio claims to provide the `OT_RADIO_CAPS_RX_ON_WHEN_IDLE` capability, then sleep management is handled by the radio platform itself. This suggests the issue you are observing is within the Nordic radio platform implementation.
(@LuDuda , @edmont can you please help check?)

----

> Setting some breakpoints in the software, even though there is the print of RadioState: Receive -> Sleep, the call to otPlatRadioSleep() is NOT triggering. Maybe due to the above ShouldHandleTransitionToSleep()?

This behavior is expected when `OT_RADIO_CAPS_RX_ON_WHEN_IDLE` is enabled. 

With `OT_RADIO_CAPS_RX_ON_WHEN_IDLE`, the radio platform layer is responsible for putting the radio to sleep. This is the feature and behavior added in the PR https://github.com/openthread/openthread/pull/9554.  Under this,  `ShouldHandleTransitionToSleep()` would return `false`, and the OT `SubMac` layer would correctly skip calling `otPlatRadioSleep()`.

---

Regarding `OPENTHREAD_CONFIG_MAC_SOFTWARE_RX_ON_WHEN_IDLE_ENABLE`, you can disregard it in relation to this issue. It's primarily intended for testing and/or use by RCP firmware. The fact that it's set to zero is appropriate here.

krbvroc1, 2024-05-27:
One last question before hopefully someone else helps... where in the 'Nordic radio platform' does it have the knowledge about the openthread state machine to make these decisions?

I am seeing breakpoints hit in places such as:
SubMac::StartTimerForBackoff()->SubMac::Sleep()->otPlatRadioSleep()->nrf5_stop() which implies that openthread is handling the sleep in at least some cases.

Is there some mixed responsibility?

I don't really comprehend the PR in [#9554.](https://github.com/openthread/openthread/pull/9554) and why I would want the RX to be ON when idle on a SED. Is turning off the OT_RADIO_CAPS_RX_ON_WHEN_IDLE something I can do or am I stuck with this huge current consumption behavior until it is fixed?

```
bool SubMac::ShouldHandleTransitionToSleep(void) const { return (mRxOnWhenIdle || !RadioSupportsRxOnWhenIdle()); }
```
I am confused by the above terminology? If mRxOnWhenIdle==true, then openthread calls sleep()? Shouldn't if be when mRxOnWhenIdle==false, that it calls sleep() since I don't want the receiver to be on when idle? Also when a device becomes detached, which is my scenario because it is looking for the parent / network, mRxOnWhenIdle does not stay the same. See below log where RxOnWhenIdle is changed from 1 to 0. And note that is is 0 before the call to sleep.

```
uart:~$ [00:03:12.421,936] <dbg> [D] SubMac--------: RadioState: Transmit -> Receive
uart:~$ [00:03:12.422,058] <dbg> [D] Mac-----------: Finishing operation "TransmitPoll"
uart:~$ [00:03:12.427,062] <inf> [N] Mle-----------: Role child -> detached
uart:~$ [00:03:12.427,185] <dbg> [D] SubMac--------: RxOnWhenIdle: 1
uart:~$ [00:03:12.427,459] <dbg> [D] Mac-----------: Idle mode: Radio receiving on channel 26
uart:~$ [00:03:12.427,673] <inf> [N] Mle-----------: RLOC16 f81b -> fffe
uart:~$ [00:03:12.427,917] <dbg> [D] SubMac--------: RadioShortAddress: 0xfffe
uart:~$ [00:03:12.428,070] <inf> [I] Mle-----------: AttachState Idle -> Start
uart:~$ [00:03:12.428,222] <dbg> [D] SubMac--------: RxOnWhenIdle: 0
uart:~$ [00:03:12.428,344] <dbg> [D] SubMac--------: RadioState: Receive -> Sleep
uart:~$ [00:03:12.428,466] <dbg> [D] Mac-----------: Idle mode: Radio sleeping
uart:~$ [00:03:12.428,588] <dbg> [D] Mac-----------: Idle mode: Radio sleeping
uart:~$ [00:03:12.428,802] <inf> [I] Notifier------: StateChanged (0x00000044) [Role Rloc-]
uart:~$ [00:03:12.428,924] <inf> [I] AnnounceSender: Stopped

```

LuDuda, 2024-05-28:
@krbvroc1 Thanks for reaching out.
@edmont is analyzing your finding and trying to reproduce.

I would recommend to move the discussion to our [DevZone](https://devzone.nordicsemi.com/) portal to get support. Please refer to this discussion in the description.

edmont, 2024-05-28:
Yes, I'm able to reproduce.

As you mention, NCS makes use of `OT_RADIO_CAPS_RX_ON_WHEN_IDLE` capability which implements IEEE 802.15.4 feature. The confusion comes on the interpretation of the capability, since setting RxOnWhenIdle to 0 doesn't mean that the radio goes to sleep immediately, but rather under these circumstances documented for `otPlatRadioSetRxOnWhenIdle` API:

```
 * There are a few situations that the radio can enter sleep state if the device is in rx-off-when-idle state but
 * it's hard and costly for the SubMac to identify these situations and instruct the radio to enter sleep:
 *
 * - Finalization of a regular frame reception task, provided that:
 *   - The frame is received without errors and passes the filtering and it's not an spurious ACK.
 *   - ACK is not requested or transmission of ACK is not possible due to internal conditions.
 * - Finalization of a frame transmission or transmission of an ACK frame, when ACK is not requested in the transmitted
 *   frame.
 * - Finalization of the reception operation of a requested ACK due to:
 *   - ACK timeout expiration.
 *   - Reception of an invalid ACK or not an ACK frame.
 *   - Reception of the proper ACK, unless the transmitted frame was a Data Request Command and the frame pending bit
 *     on the received ACK is set to true. In this case the radio platform implementation SHOULD keep the receiver on
 *     until a determined timeout which triggers an idle period start.`OPENTHREAD_CONFIG_MAC_DATA_POLL_TIMEOUT` can be
 *     taken as a reference for this.
 * - Finalization of a stand alone CCA task.
 * - Finalization of a CCA operation with busy result during CSMA/CA procedure.
 * - Finalization of an Energy Detection task.
 * - Finalization of a radio reception window scheduled with `otPlatRadioReceiveAt`.
```
So in this case no other operation is requested to the radio driver and radio will remain receiving until an spurious frame is received (like the one with `DestinationAddressFiltered` error mentioned by @krbvroc1).

IMO, OT should actively request the radio to sleep in this case, but not sure which API will be a better fit here. @abtink any ideas?

abtink, 2024-05-28:
@edmont, 

I would suggest that the call to `otPlatRadioSetRxOnWhenIdle(false)` itself trigger the radio platform to re-evaluate the idle state and put the radio to sleep.
- If the radio is in receive mode, `otPlatRadioSetRxOnWhenIdle(false)` should put it to sleep immediately.
- If in any other state, the rules specified in the API documentation would be followed (e.g., if in the middle of transmission, wait for it to finish and/or for an acknowledgment to be received or timed out).

----

> OT should actively request the radio to sleep in this case, but not sure which API will be a better fit here. @abtink any ideas?

Introducing a new API or having OpenThread call the radio platform might defeat the purpose of your original PR. This seems similar to my [proposal in the PR to have `Sleep()` called unconditionally](https://github.com/openthread/openthread/pull/9554#discussion_r1371234493) and ignored by the radio platform if already asleep, which you evaluated and found some drawbacks with.

abtink, 2024-05-28:
> I am seeing breakpoints hit in places such as:
> SubMac::StartTimerForBackoff()->SubMac::Sleep()->otPlatRadioSleep()->nrf5_stop() which implies that openthread is handling the sleep in at > least some cases.

This behavior is unexpected and not desirable.

The Sub-MAC layer's `StartTimerForBackoff()` will be used when the radio platform does not support CSMA backoff and/or MAC frame retries. In such cases, the Sub-MAC layer implements these mechanisms.

If the radio platform supports `OT_RADIO_CAPS_RX_ON_WHEN_IDLE`, it's reasonable to expect it also supports `OT_RADIO_CAPS_TRANSMIT_RETRIES` and `OT_RADIO_CAPS_CSMA_BACKOFF`.

@edmont, could you please verify if these capabilities are set on the radio platform implementation? If not, should we consider making these capabilities mandatory when `OT_RADIO_CAPS_RX_ON_WHEN_IDLE` is enabled?

krbvroc1, 2024-05-28:
FYI, I think I mentioned above the caps are reported as 0x1FB.

Could you respond to my question about:
```
bool SubMac::ShouldHandleTransitionToSleep(void) const { return (mRxOnWhenIdle || !RadioSupportsRxOnWhenIdle()); }
```
Is this logic correct or backwards?

I am confused by the above terminology? If mRxOnWhenIdle==true, then openthread calls sleep()? Shouldn't if be when mRxOnWhenIdle==false, that it calls sleep() since I don't want the receiver to be on when idle? Also when a device becomes detached, which is my scenario because it is looking for the parent / network, mRxOnWhenIdle does not stay the same. I was trying to understand the intent here and whether it was correctly implemented.

abtink, 2024-05-28:
> One last question before hopefully someone else helps... 

@krbvroc1 Answers to your questions (well, technically multiple "last questions")... :)

> where in the 'Nordic radio platform' does it have the knowledge about the openthread state machine to make these decisions?

The PR added the new API `otPlatRadioSetRxOnWhenIdle` to inform the radio platform of this flag.

----

> I don't really comprehend the PR in #9554. and why I would want the RX to be ON when idle on a SED. 

Traditionally, radio APIs were kept simple, with functions like `otPlatRadioReceive()` to start receiving and `otPlatRadioSleep()` to go to sleep. The OpenThread stack's MAC and sub-MAC layers handled all other radio operations.

Over time, more and more functions (like acknowledgment timeouts, frame retries, security processing, etc.) have been delegated to the radio platform itself, for various reasons. The radio platform indicates its capabilities in `otPlatRadioGetCaps()`, and the OpenThread stack adjusts its behavior accordingly. Any functions the radio platform can handle are delegated to it; otherwise, the OpenThread stack uses its internal implementations.

The PR #9554 attempts to add sleep management to the radio platform for two reasons:
1.  To enable faster sleep transitions, potentially reducing power consumption.
2.  To reduce the number of commands issued to the radio. This is related to the hardware architecture. The idea is that if the radio goes to sleep independently after transmitting/receiving is done, we can avoid issuing another command, especially when commands to the radio need to go through an interface (e.g., when radio operates on its own MCU core or on a separate device (RCP model)).

---

>I am confused by the above terminology? 

"Rx On When Idle" can be considered a MAC configuration flag.

The MAC layer implements support for a set of operations (e.g., direct frame transmission, indirect frame transmission, energy scan). The next layers use MAC APIs to request different operations, and the MAC layer processes these requests.

If there is no ongoing operation and no new request, the MAC layer is considered to be in "idle state." The "Rx On When Idle" flag specifies whether the MAC should keep the radio in receive mode or sleep mode while in this idle state.

----

> Also when a device becomes detached, which is my scenario because it is looking for the parent / network, mRxOnWhenIdle does not stay the same. See below log where RxOnWhenIdle is changed from 1 to 0. And note that is is 0 before the call to sleep.

The behavior you observed, where `RxOnWhenIdle` changes while the device is detached, is correct and intended.

While detached, the sleepy device periodically tries to attach. It sends "Parent Requests" and needs to listen for some time for "Parent Response" messages from potential parent candidates to attach to. During this process, the device doesn't act as sleepy and therefore sets `RxOnWhenIdle` to `true`. If the attachment process fails, it sets `RxOnWhenIdle` back to `false`.

-----

> Is this logic correct or backwards?

```c++
bool SubMac::ShouldHandleTransitionToSleep(void) const { return (mRxOnWhenIdle || !RadioSupportsRxOnWhenIdle()); }
```

This logic is correct. The key condition is `!RadioSupportsRxOnWhenIdle()`.

If the radio does not support `OT_RADIO_CAPS_RX_ON_WHEN_IDLE`, the OpenThread stack manages sleep and `ShouldHandleTransitionToSleep()` will be `true`.

If the radio supports `OT_RADIO_CAPS_RX_ON_WHEN_IDLE`:

- When `mRxOnWhenIdle` is `false`, the radio platform is expected to manage sleep. The OpenThread stack won't issue `Radio::Sleep()` commands, and `ShouldHandleTransitionToSleep()` will be `false`.
- When `mRxOnWhenIdle` is `true`, `SubMac::Sleep()` should not be called under normal circumstances. If it is called (e.g., during testing), we still want to inform the radio platform. In this case, `ShouldHandleTransitionToSleep()` will be `true`.

krbvroc1, 2024-05-28:
@abtink Thanks for the education... I saw a lot of this activity when sniffing using pyspinel and trying to figure out what is going on before reporting this. Crossing referencing it to the Thread specification has been enlightening but time consuming...I have spent a lot of hours on this.

Let me know if you need more info, thanks.

edmont, 2024-05-29:
> I would suggest that the call to `otPlatRadioSetRxOnWhenIdle(false)` itself trigger the radio platform to re-evaluate the idle state and put the radio to sleep.

Thanks @abtink, I agree that's a good compromise. Please see the API documentation update: https://github.com/openthread/openthread/pull/10307

@krbvroc1, this commit should fix your issue: https://github.com/nrfconnect/sdk-zephyr/pull/1751/commits/49f31986c4c716c32e5fa453eb2493d27fed3728

edmont, 2024-05-29:
> @edmont, could you please verify if these capabilities are set on the radio platform implementation? If not, should we consider making these capabilities mandatory when `OT_RADIO_CAPS_RX_ON_WHEN_IDLE` is enabled?

@abtink NCS radio driver already supports `OT_RADIO_CAPS_CSMA_BACKOFF` and in fact it's already a requirement in `otPlatRadioSetRxOnWhenIdle` doc.

When comes to `OT_RADIO_CAPS_TRANSMIT_RETRIES`, it's not currently supported by NCS platform since there was no special performance requirement so far. But I agree that together with `OT_RADIO_CAPS_RX_ON_WHEN_IDLE` it could slightly improve power consumption.

edmont, 2024-05-30:
> @edmont I see some other activity related to your change in drivers/ieee802154/ieee802154_nrf5.c... should I hold off before testing?

I've tested your scenario on my desk and it works as expected. Currently running some other tests with higher coverage before merging it.

edmont, 2024-06-03:
Hi @krbvroc1 the commit has been merged. Could you please verify on your end?

krbvroc1, 2024-06-04:
@edmont I manually applied the following commit
https://github.com/nrfconnect/sdk-zephyr/pull/1751/commits/9c34d6943d91a7a8d23d8e3ccc20c1d05479f22b
and it solved the issue. Thanks!

Any idea how long before this makes it into an official release such as NCS 2.6.2?
Also someone might want to update the 2.6.1 known issues for Thread in the existing SDK release notes page.

edmont, 2024-06-06:
> @edmont I manually applied the following commit [nrfconnect/sdk-zephyr@9c34d69](https://github.com/nrfconnect/sdk-zephyr/commit/9c34d6943d91a7a8d23d8e3ccc20c1d05479f22b) and it solved the issue. Thanks!

Great to hear!

> Any idea how long before this makes it into an official release such as NCS 2.6.2? Also someone might want to update the 2.6.1 known issues for Thread in the existing SDK release notes page.
Known issue is already documented here: https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/releases_and_maturity/known_issues.html#thread

When comes to 2.6.2 it's not planned yet. But fix will be backported to `v2.6-branch` soon.

jwhui, 2025-01-07:
Closing stale issue.

closed 2025-01-07 by jwhui