openthread/openthread #10315: CLI `udp bind` does not work on a multicast IPv6 address

EskoDijk, 2024-05-30:
**Describe the bug** A clear and concise description of what the bug is.

The documentation for `udp bind` mentions an IPv6 address can be provided. However, it doesn't work it seems for multicast IPv6 addresses.  Not clear if this is intentional, or a bug, or just a feature not implemented.

If we agree the feature is not available at this time and it would be some effort to add it, I can as a solution submit a PR on the documentation to mention that multicast is out of scope.

**To Reproduce** Information to reproduce the behavior, including:

In the CLI: 
```
> ipmaddr add ff05::1234
Done
> udp bind ff05::1234 17001
Error 7: InvalidArgs
```

**Expected behavior** A clear and concise description of what you expected to happen.
Would expect 'bind' to succeed. Because multicast IPv6 addresses are not explicitly ruled out.

**Console/log output** If applicable, add console/log output to help explain your problem.
See above.

**Additional context** Add any other context about the problem here.
None

jwhui, 2024-05-30:
OpenThread does not support binding a UDP socket to an IPv6 multicast address. I think this is consistent with standard sockets APIs. Feel free to submit a PR to update the docs!

EskoDijk, 2024-05-30:
@jwhui Thanks, will do that! On PC / Linux I think it's quite common to bind a UDP socket to IPv6 multicast, but I expect on embedded it's not.   (See eg https://stackoverflow.com/questions/10692956/what-does-it-mean-to-bind-a-multicast-udp-socket)

jwhui, 2024-05-30:
I'm used to using `IPV6_JOIN_GROUP` socketopt for joining a multicast group and then `bind()` for the port and interface.

What does `bind()` do differently than `IPV6_JOIN_GROUP`?

EskoDijk, 2024-06-12:
@jwhui It basically allows creating a socket that only receives datagrams that were sent to a particular multicast group, and not datagrams sent to other groups (which may have been requested by other processes on the same host) or to unicast address.

In this book text ([link](https://litux.nl/mirror/unixnetworkprogramming/0131411551_ch21lev1sec6.html) ) quoted below there's some more explanation. It seems to be a relatively 'modern' feature as it mentions older hosts don't support it. And the `IPV6_JOIN_GROUP` step is still needed. 

```
To receive a multicast datagram, a process must join the multicast group and it must also bind a UDP socket to the port number that will be used as the destination port number for datagrams sent to the group. The two operations are distinct and both are required. Joining the group tells the host's IP layer and datalink layer to receive multicast datagrams sent to that group. Binding the port is how the application specifies to UDP that it wants to receive datagrams sent to that port. Some applications also bind the multicast address to the socket, in addition to the port. This prevents any other datagrams that might be received for that port to other unicast, broadcast, or multicast addresses from being delivered to the socket.

Historically, the multicast service interface only required that some socket on the host join the multicast group, not necessarily the socket that binds the port and then receives the multicast datagrams. There is the potential, however, with these implementations for multicast datagrams to be delivered to applications that are not multicast-aware. Newer multicast kernels now require that the process bind the port and set any multicast socket option for the socket, the latter being an indication that the application is multicast-aware. The most common multicast socket option to set is a join of the group. Solaris differs slightly and only delivers received multicast datagrams to a socket that has both joined the group and bound the port. For portability, all multicast applications should join the group and bind the port.

The newer multicast service interface requires that the IP layer only deliver multicast packets to a socket if that socket has joined the applicable group and/or source. This was introduced with IGMPv3 (RFC 3376 [Cain et al. 2002]) to permit source filtering and source-specific multicast. This reinforces the requirement to join the group, but relaxes the requirement to bind to the group address. However, for maximum portability to both the old and new multicast service interfaces, applications should both join the group and bind to the group address.

Some older multicast-capable hosts do not allow the bind of a multicast address to a socket. For portability, an application may wish to ignore a bind error for a multicast address and try again using INADDR_ANY or in6addr_any.
```

PR #10370 (2024-06-18): [cli] `udp bind` doc update to clarify that multicast address cannot be used as argument
close #10315
closed 2024-06-18 by jwhui