openthread/openthread #10373: Allow the 'service add' CLI command with a zero-length S_server_data field in the Server Data TLV

EskoDijk, 2024-06-13:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
For cert tests, the `service add` command is used to mimic various DNS/SRP Anycast and Unicast Datasets. This often requires a zero-length S_server_data field to be used in the Server Data TLV. But it seems the CLI command doesn't allow to leave out the final argument or to indicate it's a zero-length field.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
Easiest would be to make the final parameter (S_server_data) optional.

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
A special character to indicate 'zero length field' for the final argument.

**Additional context** Add any other context or screenshots about the feature request here.
JIRA TESTPLAN-608

abtink, 2024-06-13:
I took care of it -> https://github.com/openthread/openthread/pull/10377

Also can suggest checking [`netdata publish dnssrp`](https://github.com/openthread/openthread/blob/6bc3b4de8a9899675707e264a0ab9194b4499a57/src/cli/README_NETDATA.md#publish-dnssrp) which is easier to use.

EskoDijk, 2024-06-14:
@abtink Thanks for this & for the pointer!  It may be that the `netdata publish dnssrp` command works somewhat differently because I was expecting it to follow the standard publishing rules for Thread.  And for testing purposes we sometimes want to force a particular service to be published, even when it doesn't follow these publishing rules. But I haven't checked if for the specific cert test here this "forcing" is needed or not.

closed 2024-06-14 by jwhui