openthread/openthread #10396: Set DNS client ServiceMode without changing any other DNS config variables

EskoDijk, 2024-06-17:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
Maybe the feature is already possible, but I didn't find yet how :)   Maybe it's something new.

For cert tests, we need an OT reference device to send DNS queries with only 1 question inside (either SRV, TXT, PTR, maybe others in the future).  However, currently the default ServiceMode is send SRV and TXT questions in a single DNS query (mode `srv_txt_opt`) causing a message to be sent with 2 questions.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
A solution is a way to set the ServiceMode without changing/touching any of the other parameters.
E.g. `dns configsm srv` or `dns config srv`.

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
Instead of changing this config, it may be even more versatile / useful to have a DNS query command that works with any numeric DNS RRTYPE.

For example a minor variant of the `dns service` command that reuses the same code and has the format:

`dns rr <rr-number> <service-instance-label> <service-name> [DNS server IP] [DNS server port] [response timeout (ms)] [max tx attempts] [recursion desired (boolean)]`

where rr-number is an integer. If I use RR `33` I can query for SRV records, and with RR `16` I query TXT records. This would allow also future tests for more RRTypes that we so far didn't do because of CLI constraints.

**Additional context** Add any other context or screenshots about the feature request here.
TSTBED-358

abtink, 2024-06-17:
@EskoDijk, I submitted this PR to change the default config: [https://github.com/openthread/openthread/pull/10397](https://github.com/openthread/openthread/pull/10397)

Without this PR, we can use the existing `dns config` command. We can set some of the earlier parameters to zero (or unspecified) value so to use the default values defined by the `OPENTHREAD_CONFIG_DNS_CLIENT_DEFAULT_{}` configuration options.

For example:

```
> dns config
Server: [2001:4860:4860:0:0:0:0:8888]:53
ResponseTimeout: 6000 ms
MaxTxAttempts: 3
RecursionDesired: yes
ServiceMode: srv_txt_opt
Nat64Mode: allow
Done

> dns config :: 0 0 0 1 srv_txt_sep
Done

> dns config
Server: [2001:4860:4860:0:0:0:0:8888]:53
ResponseTimeout: 6000 ms
MaxTxAttempts: 3
RecursionDesired: yes
ServiceMode: srv_txt_sep
Nat64Mode: allow
Done
```

The `dns config` format:
```
dns config [DNS server IP] [DNS server port] [response timeout (ms)] [max tx attempts] [recursion desired (boolean)] [service mode] ...
```

EskoDijk, 2024-06-18:
@abtink Thanks, this works!

PR #10614 (2024-08-16): [cli] clarify docs - parameter use `dns config` to change service mode only
Also adds missing CLI outputs in examples.
close #10396
closed 2024-08-16 by jwhui