openthread/openthread #10404: Serial Monitor Displays Garbage When Connecting to RCP Joiner

Glenyz, 2024-06-18:
When attempting to connect to my RCP (Restful Commissioning Protocol) Joiner using the command sudo screen /dev/ttyACM0 115200, the serial monitor only displays garbage characters. This happens consistently despite testing multiple USB ports, cables, and ensuring the boards themselves are operational.

I followed the "Thread Border Router - Bidirectional IPv6 Connectivity and DNS-Based Service Discovery" [codelab](https://openthread.io/codelabs/openthread-border-router#1) to try and get everything working. 
I expected the serial monitor to display readable data or status messages from the RCP Joiner when connected.

Console/log output:
�������#��"T�W�&�#�F�������#�<l�0lV�������3�

1. Tested multiple USB ports and cables to rule out hardware issues.
2. Verified board functionality by testing in different configurations with OpenThread software.
3. Running on Linux x86 Ubuntu 22.04.1 LTS with GNOME Terminal 3.38.1.
4. Tested every baud rate possible.

abtink, 2024-06-18:
[RCP (Radio CoProcessor)](https://openthread.io/platforms/co-processor) is a radio controlled by a host running the OpenThread stack.

The RCP/host interface utilizes the [Spinel protocol](https://github.com/openthread/openthread/blob/main/src/lib/spinel/spinel.h) over a serial bus (e.g., UART or SPI). The Spinel protocol encodes messages in a non-textual format for efficient communication between the two components.

Glenyz, 2024-06-18:
My mistake on the RCP interpretation. I am using a USB to Micro-USB cable to connect to the nRF USB port on my nrf52840 dev board. This is after I flashed the firmware onto the board. When I enter the GNU screen process and try to check ip addresses, and other info, etc. the monitor is unresponsive. I can't type into it and sometimes it displays the garbage values I mentioned before. Any ideas on why this might be happening?

Glenyz, 2024-06-18:
50% of the time the screen shows this:

~�qge~

and is unresponsive. Also this is only happening when I flash the RCP joiner firmware. Both of my FTD devices have no issues running GNU screen processes.

jwhui, 2024-06-19:
As mentioned in https://github.com/openthread/openthread/issues/10404#issuecomment-2177253603, the RCP serial protocol is a binary serial protocol and not human readable.

Glenyz, 2024-06-19:
Sorry, I am not trying to read the binary itself, but trying to ensure that the serial connection allowed my dev board to connect with my linux machine. I am following this [codelab](https://openthread.io/codelabs/openthread-hardware#6) and I am connecting to the RCP Joiner using:

sudo screen /dev/ttyACM0 115200

Can you confirm if this is the correct procedure? If not, I'm not sure how the codelab produced step 7.

jwhui, 2024-06-19:
`screen` is only used with the CLI builds. Regarding the codelab you referenced, this is related to the `ot-cli-ftd` binary.

Glenyz, 2024-06-19:
I see I wasn't working with the ```ot-cli-ftd``` binary correctly. The basic commands work now, thanks.

closed 2024-06-19 by Glenyz