openthread/openthread #10410: implementation of `mCslFrameRequestAheadUs` calculation can be improved

maciejbaczmanski, 2024-06-19:
in `InitFrameRequestAhead()`, `mCslFrameRequestAheadUs` value is calculated as sum of `OPENTHREAD_CONFIG_MAC_CSL_REQUEST_AHEAD_US` and `busTxTimeUs` (which is calculated from `busSpeedHz`). 

For RCP architecture there is one other factor which should be taken into account- there could be some additional delays on communication between host and RCP device (for example for nRF52840 DK we suspect that it might be caused by USB-UART converters), while `busSpeedHz` is calculated from baudrate value. This can be aligned by increasing `OPENTHREAD_CONFIG_MAC_CSL_REQUEST_AHEAD_US`, but:

- given option is not descriptive enough- it doesn't state that communication delays should be taken into account for RCP.
- we need to increase the value while building host's app- then if someone wants to compile a host which should work with multiple types of devices, they will need to use the highest possible value, as host will not be able to recognize expected latency for connected device.

We came up with two improvement ideas:
- implement new commands to read and set `mCslFrameRequestAheadUs` value
- implement API like `otPlatformGetCommLatency` which could return expected latency from the device and could further be added to `mCslFrameRequestAheadUs` on host on initialization

I'm opening this issue to start discussion on proposed ideas (or maybe find some other ways for mitigating the issue, which we didn't think about)

Irving-cl, 2024-06-20:
Hi @maciejbaczmanski,

I think the extra delay is related with both host and RCP. For example, even we use nrf52840 DK but we use two different host devices, the communication latency could also be different. So I think we should probably provide a way to read and set  `mCslFrameRequestAheadUs`.

My idea is that, we add a command to set the communication latency. The current calculation is kept. And when we set the latency, this latency will be added to `mCslFrameRequestAheadUs `. In this way, even nothing is set, there's still a chance that CSL can work.

maciejbaczmanski, 2024-06-21:
> My idea is that, we add a command to set the communication latency. The current calculation is kept. And when we set the latency, this latency will be added to `mCslFrameRequestAheadUs `. In this way, even nothing is set, there's still a chance that CSL can work.

Sounds good to me! @abtink do you think we can proceed with this approach?

abtink, 2024-06-21:
Yes. Adding an API to set this make sense to me. Thanks. 
Probably as a platform `otPlat` API (an `otPlatRadio`)?

maciejbaczmanski, 2024-06-24:
> Yes. Adding an API to set this make sense to me. Thanks. Probably as a platform `otPlat` API (an `otPlatRadio`)?

@Irving-cl what do you think about implementing an API instead of a command? Maybe while setting `mCslFrameRequestAheadUs` we could add a value returned from the API + new macro like `OPENTHREAD_CONFIG_RCP_HOST_COMMUNICATION_LATENCY` to also cover hosts latency, ensuring better interoperability?

Irving-cl, 2024-06-24:
@maciejbaczmanski Adding an API sounds good to me. Even if we add a command we need the API first right?

As I mentioned, I think the delay may depend on both sides. However we hope that one host program can work with different co-processors. Then we should be able to adjust it at running time. So the value (delay) shouldn't be decided at compile-time.(though it can have a default value) I think the best way is that we add an argument when starting host program. And in the handling of this argument we call the new platform API we added to set the value.

We can also add a cli command to call this platform API to tune the value. It will help in debugging.

maciejbaczmanski, 2024-06-28:
Hi!
I've started working on that. Combining given ideas I've come up with an approach:
- from device's side: add `otPlatRadioGetBusLatency` with a weak implementation to return  a precompiled `OPENTHREAD_CONFIG_BUS_LATENCY` (I'm leaving it as a weak implementation on purpose if some vendor decides to calculate the latency in the runtime)
- from host's side: add an optional argument when starting ot-daemon with host's latency
- add some logging in `InitFrameRequestAhead` for better debugging.

The downside of this approach is that we cannot change the value after initialization, but I think that when both host and device "know" their communication latencies before initialization, we cover all interoperability issues (host with different devices and device with different hosts) and should not require additional tweaking in the runtime.

Irving-cl, 2024-06-28:
Hi @maciejbaczmanski,

Sorry I think there are some misunderstandings.

My suggestion is that we only needs 'Get/SetBusLatency' on the host side. This latency contains the total latency value of both sides. This assumes that we know the total 'additional latency' for a specific combination of host+RCP. I think it's easier to tune this value as a whole, instead of configuring a latency on the RCP and adding it to the host latency.

So I mean, we can implement `otPlatRadioGetBusLatency` and `otPlatRadioSetBusLatency` but they don't need to interact with the RCP. They simply set and get a value. After we  calls `otPlatRadioSetBusLatency`, the RadioSpinel will adds this latency to the CSL_AHEAD time.

And we can also add a cli command to change this value. So this value can be changed in the runtime. But the idea is that the cli command is used to tune. And the value should be obtained by tuning in advance and set during initialization. And it should not change in runtime.

As an example, we have a new product with a combo of host A and Thread chip B. And before launch we do some tunings for this value. (Cli command help us tune this value) After we have this value, we change the start script of Thread daemon for this product to pass an argument to set this latency value. Then it's done. This value won't change for this product (in the runtime). Not sure if I understand this correctly.

maciejbaczmanski, 2024-06-28:
Thanks @Irving-cl

OK, that makes sense. I went a step forward with assumption that we might want to plug an RCP device that has been tuned by vendor without a need to know/ tune latency on our own, but this indeed might overcomplicate everything as it would require getting the value from RCP and adding it to host's.

I'll proceed with just adding for host's side then:
- optional parameter with latency
- getting/setting latency value via `otPlat` API (with default value from above parameter)
- CLI command to use `otPlat` API for tweaking the value

closed 2024-07-20 by jwhui