openthread/openthread #10437: "thread start" CLI command should not create an arbitrary active dataset

EskoDijk, 2024-06-25:
**Describe the bug** A clear and concise description of what the bug is.
In the CLI, the `thread start` command alone will create a new (arbitrary) active dataset if none is defined yet. This behavior is not documented in the CLI documentation or in the API documentation for `otThreadSetEnabled`.

In one example, on a router node, if the higher layer initiates a join operation (and that one failed or wasn't completed yet) and then a "thread start" command expecting that the join was completed/successful, then the node will start a new Thread network with arbitrary dataset.

**To Reproduce** Information to reproduce the behavior, including:
```
ifconfig up
thread start
```

**Expected behavior** A clear and concise description of what you expected to happen.
Expected: thread cannot be started without a dataset. Only with explicit commands like `dataset init new` there should be a new (arbitrary) dataset created.
For the `dataset` CLI commands we now add warning messages that it's not suitable for production use. It would be strange to allow similar behavior in "thread start" without even warning about it.

If there's a reason to keep the behavior: then the APIs and CLI commands should get some note/warnings about this behavior.

**Console/log output** If applicable, add console/log output to help explain your problem.
none

**Additional context** Add any other context about the problem here.
none

jwhui, 2024-06-25:
> In the CLI, the `thread start` command alone will create a new (arbitrary) active dataset if none is defined yet. This behavior is not documented in the CLI documentation or in the API documentation for `otThreadSetEnabled`.

What git commit are you using? We correct this behavior a while ago.

abtink, 2024-06-25:
@EskoDijk can you check the [`OPENTHREAD_CONFIG_OPERATIONAL_DATASET_AUTO_INIT` config](https://github.com/openthread/openthread/blob/cc8f66c56f9675648c3d2c47cd6aa7f4e92102b1/src/core/config/misc.h#L590-L600)?  It is by default disabled. If it is enabled, then dataset can be auto generated.

EskoDijk, 2024-06-26:
@abtink Yes, that was the issue indeed! Thanks. (I enabled this config last year because at the time, my code to configure datasets was not using the  `dataset` command yet.  Now it is. And by now I forgot I was even using it...) Closing this issue.

closed 2024-06-26 by EskoDijk