openthread/openthread #10447: Help: router limit

dakhnod, 2024-06-27:
While the mesh ability of Thread would be perfect for our application, the router limit of 32 poses a limitation for us.
We potentially have thousands of devices over a large area, thus 32 routers could not suffice.

Hence, I am seeking for new Ideas on this question:

How can I build my huge Thread network while adhering to the Standard as much as possible?

My only Idea right now is to have multiple Thread networks, bridging them over IP.
The downside I see is that that limits roaming through the while are though, or can my nodes jump from one network to another somehow?

konradderda, 2024-07-01:
Hello @dakhnod ,
The limit of 32 routers in a single network cannot be increased as some of mesh management messages must fit into a single 802.15.4 radio frame. At the moment, it seems that the best solution for your use case would be to use Thread Domains. With this feature you connect multiple Thread partitions and use them as a single being. 

You can test it by adding a Domain Prefix (reference: https://github.com/openthread/openthread/blob/main/src/cli/README.md#prefix). If you add the same prefix for multiple Thread partitions they should be able to communicate with each other using DUA addresses (Domain Unicast Address). Also, DUA addresses can be globally routable and are accessible from external interface thanks to ND proxy feature.

As far as I know there is ongoing work towards improving handling of such scenarios but at the moment it looks like the best shot for you.

dakhnod, 2024-07-02:
Thanks, that answers it!

dakhnod, 2024-07-02:
Btw, is OpenThread capable of supporting your suggestion at the moment?

konradderda, 2024-07-02:
Yes, it's a part of Thread 1.2 Specification and is available. In order to utilize DUA routing you have to use Thread Border Routers connected through an external interface (e.g. Ethernet or Wi-Fi). Then you can propagate the Domain information and they should handle the rest :)

dakhnod, 2024-07-05:
Thanks! Could you please give me a hint on how to configure my border router for a domain?
Or rather, which steps need to be done on the border router side and which on the nrf52 side, so the devices can roam around partitions?

I find it a bit hard to find documentation on this part...

dakhnod, 2024-07-15:
Also,

7. Would >32 routers with the same dataset split into partitions?
8. Do I need the same dataset on all routers in order to allow for roaming between partitions?

jwhui, 2024-09-27:
Closing stale issue.

closed 2024-09-27 by jwhui

dakhnod, 2024-10-29:
@EskoDijk Thanks for the explanation!
1. Am I correct that the DUI, in this case, behaves like the LL-address in the sense that it remains throughout different networks?
2. Can I rely on the LL-address of a device when addressing it within one network?
3. What limitations does the LL-address pose? I assume it isn't routed by BR between networks...
4. 
Thanks for your time!

EskoDijk, 2024-10-29:
@dakhnod 
1. Yes, the DUA is designed to remain the same as long as the Thread device stays in the same Thread Domain. It should be randomly generated using the same algorithm as SLAAC (https://datatracker.ietf.org/doc/html/rfc7217#autoid-5) with specific choices for input parameters as defined by Thread.  However in practice OpenThread leaves that generation to the caller of the API I think so any random and stable identifier (IID) is possible.  Link-local IPv6 also has this property in practice in Thread.
2. In Thread, the LL address should be stable within one Thread Network, however it only allows local (one radio hop) of communication. What you're looking for is the Thread ML-EID (Mesh-Local Endpoint Identifier) which is routable only within one Thread Network Partition.
3. So LL is only a single radio hop, while ML-EID is routed only within the current Thread Network Partition.  The DUA is routed within a Thread Network and also between multiple Thread Networks (within a Domain).

DUA addressing is only really valuable if you have devices that are allowed to "roam" between Thread Networks i.e. you don't know in advance always in what network a device is located.  It's designed for relatively low mobility so i.e. only a few Thread devices that may occassionally move to another network due to RF coverage issues or rare "device was moved" event.

If your devices are each fixed to one Thread Network, then it's easier (and maybe more reliable) to make use of fixed ULA addresses for each device and let them communicate this way. This way, you don't need Border Routers that specifically support all the DUA features. When using DUA, it requires specific Border Routers that can support configuration of the DUA (Domain Prefix) and that run the ND-proxy function requires for locating the DUA devices.