openthread/openthread #10457: [coap] partial matching of resource's URI

konradderda, 2024-07-01:
**Is your feature request related to a problem? Please describe.**
Currently, CoAP module matches the whole `mUri` value. Such behavior limits usages of a single resource instance for scenarios where multiple URIs share a fine portion of the path, e.g. device id's selector URI-Path: `/device/<id>`. With current implementation one must to have an otCoapResource instance for every <id> where much simpler approach would be to have a single structure instance regardless of a number of hypothetical devices.

Also, it totally blocks usage of URI-Query options, as in example URI `/discover?version=1&type=thread&diag=true` implementation cannot handle values provided by the other party and one would also need to handle every possible combination of these.

**Describe the solution you'd like**
I would like to propose a change with partial URI matching but I would like to know if such change is welcomed and also discuss if such partial matching should be a default behavior or it should be controlled by a flag.

abtink, 2024-07-03:
Thanks, @konradderda. I think adding this feature could be beneficial.

As for whether to make partial URI matching the default behavior or control it with a flag, I suggest we base the decision on the size of the code change required to implement it. If the change is small, including it by default might make sense. If it introduces significant overhead, it might be better to put it behind a build flag.

jwhui, 2025-01-07:
Closing stale issue.

closed 2025-01-07 by jwhui