openthread/openthread #10458: [routing-manager] recent PRs updating/enhancing BR functionality

abtink, 2024-07-01:
Using this issue to track recent PRs related to Border Router (BR) function in OT core `RoutingManager` module and its APIs.
In particular simplifications, enhancements, and new features/functions (e.g. tracking peer BRs).

- https://github.com/openthread/openthread/pull/10453
- https://github.com/openthread/openthread/pull/10448
- https://github.com/openthread/openthread/pull/10445
- https://github.com/openthread/openthread/pull/10452
- https://github.com/openthread/openthread/pull/10389
- https://github.com/openthread/openthread/pull/10388
- https://github.com/openthread/openthread/pull/10382
- https://github.com/openthread/openthread/pull/10379
- https://github.com/openthread/openthread/pull/10343
- https://github.com/openthread/openthread/pull/10342
- https://github.com/openthread/openthread/pull/10330
- https://github.com/openthread/openthread/pull/10317
- https://github.com/openthread/openthread/pull/10316
- https://github.com/openthread/openthread/pull/10310
- https://github.com/openthread/openthread/pull/10309
- https://github.com/openthread/openthread/pull/10291
- https://github.com/openthread/openthread/pull/10284
- https://github.com/openthread/openthread/pull/10281
- https://github.com/openthread/openthread/pull/10270
- https://github.com/openthread/openthread/pull/10265
- https://github.com/openthread/openthread/pull/10257
- https://github.com/openthread/openthread/pull/10256
- https://github.com/openthread/openthread/pull/10255
- https://github.com/openthread/openthread/pull/10250
- https://github.com/openthread/openthread/pull/10244
- https://github.com/openthread/openthread/pull/10240
- https://github.com/openthread/openthread/pull/10239
- https://github.com/openthread/openthread/pull/10230
- https://github.com/openthread/openthread/pull/10229
- https://github.com/openthread/openthread/pull/10216
- https://github.com/openthread/openthread/pull/10214
- https://github.com/openthread/openthread/pull/10147

abtink, 2025-03-31:
New BR related PRs (in 2025):
- https://github.com/openthread/openthread/pull/11365

jwhui, 2026-11-04:
Closing as all tracked PRs have been merged.

closed 2026-11-04 by jwhui