openthread/openthread #10464: Help: multiple thread stacks on the same RCP

dakhnod, 2024-07-02:
We are utilizing an [nRF82840 Dongle](https://www.mouser.de/ProductDetail/Nordic-Semiconductor/nRF52840-Dongle?qs=gTYE2QTfZfTbdrOaMHWEZg%3D%3D&mgh=1&vip=1&utm_id=20979042634&gad_source=1&gclid=CjwKCAjwyo60BhBiEiwAHmVLJRnO0xlkDxNx5nXY9E1kvKszwMoVzIt_67ZHoXBt5nIVe6DG3F1tnxoCwrwQAvD_BwE) at the moment to run the OpenThread agent.

That agent, if I understand correctly, utilizes the nRF52 as a Radio Co-processor (RCP) (or NCP? not sure...).

In any case, is there a way to start multiple Thread router processes on the same host with the same dongle?
We want to spin up multiple networks on the same raspi and, if possible the same Dongle.
Is there a way to share the serial port between multiple processes?

abtink, 2024-07-02:
> In any case, is there a way to start multiple Thread router processes on the same host with the same dongle?

The RCP (or any radio platform) can generally only act as the radio for a single Thread network (single OpenThread instance).

dakhnod, 2024-07-04:
okay, thanks.

closed 2024-07-04 by dakhnod