openthread/openthread #10469: Potential double free of messages

CodingRays, 2024-07-03:
For context this is a bug i encountered in a modified version of openthread. I am unsure if it is possible to trigger this specific code path in an unmodified version of openthread. However, it seems like something that should be checked.

The bug is a double free of a message which is both marked as a direct transmission as well as a indirect transmission. In particular it is triggered if the following sequence happens:
1. Message is set as the next indirect transmission for a child
2. `MeshForwarder::PrepareNextDirectTransmission` is called
3. Route update for the direct transmission fails
4. The [default error handler case](https://github.com/openthread/openthread/blob/main/src/core/thread/mesh_forwarder.cpp#L657) in `MeshForwarder::PrepareNextDirectTransmission` frees the message without checking if it has any indirect transmissions pending.
5. Indirect transmission finishes. Causes a second free of the already freed message.

As i said i am unsure if it is possible to trigger this without modifications. However other direct transmit functions ([for example](https://github.com/openthread/openthread/blob/main/src/core/thread/mesh_forwarder.cpp#L1302) `MeshForwarder::UpdateSendMessage`) do correctly check for pending indirect transmissions so it seems like something that should be covered.

jwhui, 2024-07-03:
@CodingRays , thanks for raising this issue.

It looks like the line you highlighted:
https://github.com/openthread/openthread/blob/aebecca9649704c1318228e6da562eaa7c1ed156/src/core/thread/mesh_forwarder.cpp#L657

Should be replaced with a call to:
https://github.com/openthread/openthread/blob/aebecca9649704c1318228e6da562eaa7c1ed156/src/core/thread/mesh_forwarder.cpp#L1351

abtink, 2024-07-03:
Thanks @CodingRays for debugging, finding and reporting this. I just submitted https://github.com/openthread/openthread/pull/10471. 

I agree it would be definitely better/safer to use the `RemoveMessageIfNoPendingTx()` instead.

I expect this scenario to be unlikely. An `error` in `PrepareNextDirectTransmission()` usually comes from `error = UpdateIp6Routes()` failing to find a next hop towards the destination of message. Typically, only multicast messages would be queued for both indirect tx to children and direct tx, and then `UpdateIp6Routes()` wouldn't fail for broadcast destination.

closed 2024-07-03 by jwhui