openthread/openthread #10470: [Infra If] Consider deprecating `otPlatInfraIfHasAddress`?

Irving-cl, 2024-07-03:
The only usage of `otPlatInfraIfHasAddress` is for `RoutingManager` to find out if a received Router Advertisement is for this device or another BR: https://github.com/openthread/openthread/blob/main/src/core/border_router/routing_manager.cpp#L745 

What about we add a parameter in `otPlatInfraIfRecvIcmp6Nd` to indicate if the received ND message is for this device or not?

Because implementing `otPlatInfraIfHasAddress` under NCP design (on NCP side) brings extra complexity. For example, we should let the host sync the addresses of the infra interface to the NCP. This seems unnecessary. Thoughts? @abtink

abtink, 2024-07-03:
I agree that combining it with `otPlatInfraIfRecvIcmp6Nd()` seems simpler, but keeping `InfraIfHasAddress` as a separate API might be a better approach for future flexibility. There may be potential uses of this API in the future (e.g., in native mDNS, we can make additional optimizations if we can determine that a message originates from the device itself).

> Because implementing otPlatInfraIfHasAddress under NCP design (on NCP side) brings extra complexity. For example, we should let the host sync the addresses of the infra interface to the NCP. This seems unnecessary.

Regarding implementation complexity, I believe both options would require a similar level of effort, whether we determine address ownership within a separate command or as part of `otPlatInfraIfRecvIcmp6Nd`.

Instead of syncing infra-interface addresses to the NCP, an alternative approach can be for the NCP to query the host for this information as needed. Would this work?

Irving-cl, 2024-07-04:
@abtink 

Thanks! Then let's keep `InfraIfHasAddress` for future flexibility. 

Regarding the implementation, I prefer using the approach to syncing infra-interface addresses to the NCP. Because:
1. The update of addresses on the infra interface is expected to be less frequent. And then the NCP can always do the queries locally.
2. If we let NCP query the host, it would be more difficult to write the code. For example, the routing manger (on NCP) tries to now if the Infra has an address, it needs to send a query to the host. And the following code has to be executed after it gets a response.

We can further discuss the implementation when I start working on this :)

closed 2024-07-04 by Irving-cl