openthread/openthread #10478: [tests] unexpected test_advertising_proxy failing for openthread#10318

librasungirl, 2024-07-04:
**Describe the bug** 
Unexpected test_advertising_proxy failing for https://github.com/openthread/openthread/pull/10318

**Console/log output**
https://github.com/openthread/openthread/actions/runs/9779759835/job/27023842229?pr=10318

FAIL: test (__main__.SingleHostAndService)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/border_router/test_advertising_proxy.py", line 179, in test
    self.check_host_and_service(server, client, '2001::2', 'my-service')
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/border_router/test_advertising_proxy.py", line 301, in check_host_and_service
    self.assertEqual(client_service['state'], 'Registered')
AssertionError: 'Refreshing' != 'Registered'
- Refreshing
+ Registered

librasungirl, 2024-07-04:
Another failing https://github.com/openthread/openthread/actions/runs/9791676882/job/27035979267?pr=10318#step:8:6748

abtink, 2024-07-04:
Thanks, @librasungirl.

I noticed the same CI failures and submitted #10474 yesterday (before you filed this issue and submitted #10479). I guess you probably didn't see it. 🙂

closed 2024-07-05 by jwhui