openthread/openthread #10483: Performing an address query immediately after becoming a router can lead to some unusual phenomena

gytxxsy, 2024-07-05:
**Describe the bug** 

Shortly after a child device becomes a router device through address solicitation, some connectivity issues may arise due to the order of sending and receiving advertisements.

**To Reproduce**

1. Git commit id: be7d36e4ff9cf7df6dfce54e58a31163c87b93f7
2. platform: ESP32-H2
4. Build steps: reffer to [here](https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_cli/README.md)
5. Network topology: device1(leader)---device2(firstly child, then become router via `address solicit`)

**Expected behavior**

Device 1 forms a network and becomes the leader. Device 2 joins the network and becomes a child. Then, Device 2 becomes a router through `address solicit` and immediately pings Device 1's mleid address after becoming a router.

In expectation, Device 2 should be able to ping Device 1's mleid address. However, I have discovered two unexpected situations:

1. After Device 2 becomes a router, it immediately sends out an advertisement packet. During the first ping, Device 2 sends out an address query packet. It may lead to a situation: Device 1 receive and process the address query first, but at this point, Device 1 has not received and processed the advertisement. Therefore, it considers the address query to be from an invalid router and does not respond with an address notification. In this scenario, the first ping operation will fail.
pacp

I tried delaying 1 second after Device 2 becomes a router to ensure that the advertisement is received by Device 1 before the address query. This solves issue 1, but at the same time, the following problem is exposed.

2. After Device 2 becomes a router, it immediately sends out an advertisement. Device 1 also sends out an advertisement, but there is no strict order between these two actions. It may lead to a situation: Device 1 sends out an advertisement (ad1_1) before receiving the advertisement from Device 2 (ad2_1). At this point, the LinkqualityIn and LinkqualityOut corresponding to Device 2 in ad1_1 will be set to 0. When Device 2 receives ad1_1, it will update its LinkqualityOut corresponding to Device 1 based on the route64 TLV (updating it to 0). As a result, Device 2 considers the link quality to Device 1 to be 0. When it receives the address notification response from Device 1, it assumes there is no route to Device 1 and does not reply with a CoAP ACK. This causes the address query process to fail.

**Console/log output**
[pcap_file.zip](https://github.com/user-attachments/files/16109781/pcap_file.zip)
* `advertisement_later_than_address_query.pcap` is for situation 1, the critical packets are packet 32 and packet 45.

* `send_ad1_with_wrong_linkquality_tlv.pcap` is for situation 2, the critical packet are packet 32, 45, 46.

***Note: The order of packets in the pcap file only represents the order in which they were captured, not the order in which they were actually received and processed.***

**Additional context**

In fact, if there is a delay of around 4 seconds after the network is formed, Device 1 and Device 2 will send multiple advertisements, thereby establishing a stable network connection. Subsequent address queries and pings work normally. However, based on the scenario I described above, performing an address query right after the network is established can indeed cause some unusual behavior in the network.

abtink, 2024-07-05:
Thanks @gytxxsy.

After a child is promoted to a router role, the route table (path costs) need to be discovered and updated by all nodes in the network. This is done through the MLE Link Advertisement process. It's expected that during this transition, some messages may be dropped.

There have been numerous changes and enhancements to improve this process, both in the Thread specification and additional optimizations in OpenThread:

- **Immediate Link Advertisement:** The newly promoted router (upon receiving an "Address Solicit Response") immediately sends a Link Advertisement and resets its advertisement trickle timer. This accelerates the dissemination of the new router ID allocation throughout the network.
- **Parent Acceptance of Child's Link Advertisement:** The parent of the newly promoted router accepts its Link Advertisement since they have an established link and can validate/authenticate the message. Previously, this information had to be distributed via a Link Advertisement from the Leader, which could take longer.
- **Established Link After Promotion:** The (former) parent considers the link established with the newly promoted router upon receiving its Link Advertisement, avoiding the need for an extra Link Request/Response exchange.
- **Next Hop Selection:** The newly promoted router initially uses its parent as the next hop towards all other nodes until it discovers better paths.

There are also optimizations to skip performing Address Queries when possible:
- **Child Retains Address Cache:** The child keeps its Address Cache table after promotion.
- **Parent Updates Address Cache:** The (former) parent updates its Address Cache entries for the newly promoted router to use its new RLOC16.
- **Network Data (SRP) Optimization:** Devices skip Address Queries if they can use information from the Network Data (SRP) service info. This helps them reach Border Routers (BRs) without needing Address Queries.
- **BR Address Cache Snooping:** BRs proactively add snoop Address Cache entries for the set of addresses registered by devices using SRP.

There is also an optimization to allow the (former) child to accept its old RLOC16 for a short interval after promotion. This requires support from the Radio Platform (to allow "alternate RLOC16"). This optimization is in #10006 (not yet merged).

Link Advertisements are sent as broadcasts and may be missed by other devices. So the scenarios you described are possible but should be atypical and resolve automatically after a short period (e.g., through retries). Ideas/solutions to enhance are of course welcome.

Here are some related PRs and issues for further reference:
- #8307
- #8318
- #9881
- #9956
- #10006
- [SPEC-1131](https://threadgroup.atlassian.net/browse/SPEC-1131)

jwhui, 2025-01-07:
Closing stale issue.

closed 2025-01-07 by jwhui