openthread/openthread #10529: Diagnostic MLE Counters TLV: New Parent Counter value after initial attach

EskoDijk, 2024-07-19:
**Describe the bug** A clear and concise description of what the bug is.
For MLE Counters TLV, the New Parent Counter field is expected to be `1` after the initial attach of a device to a Thread Network. During cert test development, we see the value `0` however.

(The issue might be related to the method of bringing up the DUT during the test, but having this issue if still useful to verify if the code is the problem or something else.)

**To Reproduce** Information to reproduce the behavior, including:
OPENTHREAD/b9dcdbc; Zephyr

**Expected behavior** A clear and concise description of what you expected to happen.
Value `1` after the initial attach.
See spec text  10.11.4.15; I assume a node always starts as "detached" if it's not yet attached to the network. (If this is not correct let me know) So the counter should go +1 on initial attach.

**Console/log output** If applicable, add console/log output to help explain your problem.
-

**Additional context** Add any other context about the problem here.
-

abtink, 2024-07-19:
This corresponds to the MLE counter `mParentChanges`:
[https://github.com/openthread/openthread/blob/602167f94ad4318a26d73bb7ecb639cbf54df633/include/openthread/thread.h#L194-L202](https://github.com/openthread/openthread/blob/602167f94ad4318a26d73bb7ecb639cbf54df633/include/openthread/thread.h#L194-L202)

This is intended to count the number of "changes", and the very first time a device (from a factory-reset state) attaches to a parent is not counted as a change.

We can either change the code to count the first-time attachment as well, or modify the specification. I don't have a preference either way.

EskoDijk, 2024-07-21:
@abtink Thanks, I see now how the difference of `1` got introduced. The term "New parent" can be seen as "can only get a new parent if you had a current/previous/old one" or as "your first parent is also (by definition) a new parent".
I think we can update the spec text on this - created [SPEC-1303](https://threadgroup.atlassian.net/browse/SPEC-1303)

abtink, 2024-07-21:
Thanks @EskoDijk.

EskoDijk, 2024-08-15:
Thanks. Resolved in SPEC-1303, no code change needed - closing this issue!

closed 2024-08-15 by EskoDijk