openthread/openthread #10532: otCoapAddResource执行后，会改变otCoapResource的mNext属性为 NULL？

wnylei, 2024-07-22:
### 问题：
**otCoapAddResource** 执行后，会改变otCoapResource的mNext属性为 NULL，这是预期行为吗？

### 环境：
硬件平台：乐鑫（Espressif）
软件库版本：ESP-IDF v5.1.3

### 复现代码：
```

static otCoapResource u_coap_resource1 = {
    .mUriPath = "thread/test1",
    .mHandler = u_common_thread_coap_request_handler1,
    .mContext = NULL,
    .mNext = NULL,
};

static otCoapResource u_coap_resource = {
    .mUriPath = "thread/test",
    .mHandler = u_common_thread_coap_request_handler,
    .mContext = NULL,
    .mNext = &u_coap_resource1,
};

// U_LOCKED_OT_CODE 为加锁代码，无需关注
static void u_add_all_resources(otCoapResource *resource_head_p)
{
    U_LOCKED_OT_CODE(
        otInstance *instance_p = u_thread_get_instance();

        otCoapResource *resource_loop_p = resource_head_p;
        while (resource_loop_p != NULL) {
            U_LOGE("添加资源: %s\n", resource_loop_p->mUriPath);
            U_LOGE("下一个资源: %s\n", resource_loop_p->mNext ? resource_loop_p->mNext->mUriPath : "NULL");

            otCoapAddResource(instance_p, resource_loop_p);

            U_LOGE("资源添加后，当前资源: %s\n", resource_loop_p->mUriPath);
            U_LOGE("资源添加后，下一个资源: %s\n", resource_loop_p->mNext ? resource_loop_p->mNext->mUriPath : "NULL");

            resource_loop_p = resource_loop_p->mNext;
        }

    );
}

```

### 执行 **u_add_all_resources(&u_coap_resource);**  后的日志：

```
E (660) U_LOG: 添加资源: thread/test
E (670) U_LOG: 下一个资源: thread/test1
E (690) U_LOG: 资源添加后，当前资源: thread/test
E (710) U_LOG: 资源添加后，下一个资源: NULL

```

### 请求指教，万分感谢！

zwx1995esp, 2024-07-22:
@wnylei Hi
`otCoapAddResource`底层实现是基于openthread common 中实现的类 LinkedList 来做的。 LinkedList 需要一个entry一个entry的加，不支持加一个list. 具体可以看源码实现: [void otCoapAddResource(otInstance *aInstance, otCoapResource *aResource)](https://github.com/openthread/openthread/blob/main/src/core/api/coap_api.cpp#L279) -> [void CoapBase::AddResource(Resource &aResource)](https://github.com/openthread/openthread/blob/main/src/core/coap/coap.cpp#L99) -> [Error Add(Type &aEntry)](https://github.com/openthread/openthread/blob/main/src/core/common/linked_list.hpp#L300) -> [void Push(Type &aEntry)](https://github.com/openthread/openthread/blob/main/src/core/common/linked_list.hpp#L164 ) -> [void SetNext(Type *aNext)](https://github.com/openthread/openthread/blob/main/src/core/common/linked_list.hpp#L94)

wnylei, 2024-07-22:
### 看到了，非常感谢！

closed 2024-07-22 by wnylei