openthread/openthread #10538: [pull/10454](https://github.com/openthread/openthread/pull/10454) causes the nrf52840dongle usb-acm device fail to start

ihidchaos, 2024-07-24:
**Describe the bug** A clear and concise description of what the bug is.
I found that after using the latest commit code, my nrf52840 dongle can no longer be used normally, and the error message is as follows. After gradually narrowing the commit scope, I found that this is caused by pr 10454. After I manually reverted this commit, my thread network started up normally.

```Shell
root@OpenWrt:~# otbr-agent -I wpan0 -B br-lan -d7 -v spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=1000000 trel://br-lan --auto-attach
otbr-agent[4987]: [NOTE]-AGENT---: Running 0.3.0
otbr-agent[4987]: [NOTE]-AGENT---: Thread version: 1.3.0
otbr-agent[4987]: [NOTE]-AGENT---: Thread interface: wpan0
otbr-agent[4987]: [NOTE]-AGENT---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=1000000
otbr-agent[4987]: [NOTE]-AGENT---: Radio URL: trel://br-lan
otbr-agent[4987]: [NOTE]-ILS-----: Infra link selected: br-lan
otbr-agent[4987]: [INFO]-RCP_HOS-: OpenThread log level changed to 5
otbr-agent[4987]: 49d.21:49:54.141 [D] P-SpinelDrive-: Sent spinel frame, flg:0x2, iid:0, tid:0, cmd:RESET
otbr-agent[4987]: 49d.21:49:54.141 [D] P-SpinelDrive-: Waiting response: key=0
otbr-agent[4987]: 49d.21:49:56.143 [W] P-SpinelDrive-: Wait for response timeout
otbr-agent[4987]: 49d.21:49:56.143 [I] P-SpinelDrive-: co-processor self reset successfully
otbr-agent[4987]: 49d.21:49:56.143 [D] P-SpinelDrive-: Sent spinel frame, flg:0x2, iid:0, tid:1, cmd:PROP_VALUE_GET, key:PROTOCOL_VERSION
otbr-agent[4987]: 49d.21:49:56.143 [D] P-SpinelDrive-: Waiting response: key=1
otbr-agent[4987]: 49d.21:49:58.146 [W] P-SpinelDrive-: Wait for response timeout
otbr-agent[4987]: 49d.21:49:58.146 [D] P-SpinelDrive-: Sent spinel frame, flg:0x2, iid:0, tid:1, cmd:PROP_VALUE_GET, key:PROTOCOL_VERSION
otbr-agent[4987]: 49d.21:49:58.146 [D] P-SpinelDrive-: Waiting response: key=1
otbr-agent[4987]: 49d.21:50:00.148 [W] P-SpinelDrive-: Wait for response timeout
otbr-agent[4987]: 49d.21:50:00.148 [C] Platform------: Init() at spinel_driver.cpp:82: Failure
otbr-agent[4987]: 49d.21:50:00.148 [D] P-SpinelDrive-: Sent spinel frame, flg:0x2, iid:0, tid:1, cmd:PROP_VALUE_GET, key:PROTOCOL_VERSION
otbr-agent[4987]: 49d.21:50:00.148 [D] P-SpinelDrive-: Waiting response: key=1
otbr-agent[4987]: 49d.21:50:02.150 [W] P-SpinelDrive-: Wait for response timeout

```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
I tested it using the following commit for **ot-br-posix**:
**Wokring:**

- 2024-07-10
#PKG_SOURCE_VERSION:=88403c841c9d7fbaa0606ef480a06181f8c99904

- 2024-07-09
#PKG_SOURCE_VERSION:=04a9963ac8d0165338abadb80da968b26050cacb

- 2024-07-02
#PKG_SOURCE_VERSION:=a994884f099b7765b91a3ea35ce2fbd1a5453d51

- 2024-06-24
#PKG_SOURCE_VERSION:=261a0a1bdce240b7dedc01a765237d444b718549

**Not Working:**
- 2024-07-16
#PKG_SOURCE_VERSION:=8c7b36b3fa5b265b5897a44e3fcffdcd380f251a

- 2024-07-11-1027
PKG_SOURCE_VERSION:=05ca8d966abccc444c3c3961b1174486a7ddf2e9

- 2024-07-24
#PKG_SOURCE_VERSION:=2325a356761c14095cce8a19c38191c6c187c854

2. IEEE 802.15.4 hardware platform **nrf52840dongle**
3. Build steps **The construction process of openwrt platform was used, and the hardware was Raspberry PI 4b.**
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.
The nrf52840dongle should work normally and has nothing to do with flow control
**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2024-07-24:
@tl-sl , can you help take a look?

tl-sl, 2024-07-25:
I dont have this device to test, however I have looked through the datasheets and can't see any indication that DTR/RTS are used for any special function on this device. That makes me suspect that is in fact expecting flow control, and the previous behaviour where RTS was asserted, essentially made it appear like flow control was enabled on the host.

@ihidchaos can you try add `&uart-flow-control` to the spinel url and see if that makes any difference?

tl-sl, 2024-07-25:
it looks like the shell ate the `&`, you might need to quote the entire url string to avoid this:

`"spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=1000000&uart-flow-control"`

ihidchaos, 2024-07-25:
Ok, thanks for the tip, I can now successfully run otbr with the following command.👍
```
./otbr-agent-20240716 -I wpan0 -B br-lan -d7 -v "spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=1000000&uart-flow-control" trel://br-lan
```
My questions are:
1. Before this commit, I didn't need to enable flow control to work.
2. When I compiled the rcp according to the nrf sdk documentation, my config file also didn't have any configuration for flow control
3. So I don't think my nrf52840dongle actually needs to rely on flow control.
4. This new commit requires me to enable flow control, but my hardware and software do not support flow control, so is this enabled necessary, is it correct?

tl-sl, 2024-07-26:
> 4\. This new commit requires me to enable flow control, but my hardware and software do not support flow control, so is this enabled necessary, is it correct?

there are only two possibilities here:
1. the nrf dongle is in fact expecting flow control, which happened to work previously as RTS was asserted, which only makes it appear that flow control is supported on the host, and thus allows data to flow, but without actual flow control support. In this case enabling flow control configuration would be correct thing to do.

2. The dongle is using DTR/RTS for some special function such as to enter bootloader/DFU mode, however there is not any mention of this in the datasheets. Further `nrfutil` appears to use native USB requests to enter DFU mode and not signals on the virtual com port. So this seems unlikely to be the case.

I will however get a nrf dongle to test further next week.

ihidchaos, 2024-07-26:
> there are only two possibilities here:
>
> 1. the nrf dongle is in fact expecting flow control, which happened to work previously as RTS was asserted, which only makes it appear that flow control is [...]

Thank you for your explanation! If conjecture 1 is true, I think we need to add documentation for this situation, otherwise it will appear to the user that the previously available device just updated the commit and won't work properly. In particular, the CHIP community's TestHarnnes suite relies on nrf52840dongle by default and does not have a flow control configuration enabled.

rgillan, 2024-07-26:
Have confirmed the same scenario with the nRF52840 dev kit as well as the USB dongle version.  Both otbr-agent and ot-daemon both start correctly:
```/usr/sbin/otbr-agent -I wpan0 -B eth0 "spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=1000000&uart-flow-control" trel://eth0```

ihidchaos, 2024-08-05:
> there are only two possibilities here:
> 1. the nrf dongle is in fact expecting flow control, which happened to work previously as RTS was asserted, which only makes it appear that flow control is [...]

Hi tl, Is there any progress on testing, should we keep flow control on and update the documentation to account for the changes?

tl-sl, 2024-08-09:
> Hi tl, Is there any progress on testing, should we keep flow control on and update the documentation to account for the changes

I just received the dongle, will do some testing in the next few days.

LuDuda, 2024-08-09:
I completely agree with @suveshpratapa and support his suggestion. The new behavior has already resulted in some customer support cases on our side, even though it was only introduced just a month ago. If possible, reverting to the previous behavior by default would be ideal.

jwhui, 2024-08-09:
@tl-sl, given the comments from @suveshpratapa and @LuDuda , would you be willing to make the default behavior the same as before #10454?

tl-sl, 2024-08-09:
> since we (Silicon Labs) are also dealing with the ramifications of this change.

I tested this change on Silicon labs adapters before submitting, for both devices with and without hardware flow control and it worked as expected (with the relevant flow control setting). Or is it that flow control devices used to work without setting flow control?

tl-sl, 2024-08-09:
> would you be willing to make the default behavior the same as before #10454?

I guess can add a flag to control behaviour.

tl-sl, 2024-08-12:
> I assume that is due to default driver settings or auto-negotiation during startup when hardware flow control is enabled.

No, this only happens because RTS happens to be asserted (which in itself is platform dependant as to default behaviour of RTS), as such it appears that the host is ready to receive data.  Data will flow but you are still likely to get buffer overruns though as any flow control requests will be ignored. Normally with hardware flow control enabled, RTS will be deasserted until the host is actually ready to receive data, as which point RTS will become asserted.

> I think the behavior you want to impose -- i.e., deasserting DTR/RTS on init, should be independent of the flow control setting anyway.

It is inherently linked to flow control though, when flow control is disabled there shouldn't be any dependency on the state of RTS.

romacdon, 2024-08-12:
> > would you be willing to make the default behavior the same as before #10454?
> 
> I guess can add a flag to control behaviour.

@tl-sl any more thoughts on this?  I would be great if we could great the change in soon.  Thanks.

romacdon, 2024-08-16:
@tl-sl Any update on this issue?  Are you still planning on adding a flag to control behaviour?

jwhui, 2024-09-27:
Closing stale issue.

closed 2024-09-27 by jwhui