openthread/openthread #10557: Occasional failure to register DUA as REED

Cristib05, 2024-07-30:
**Describe the bug** A clear and concise description of what the bug is.
Occasional failure to register DUA as REED. This was observed in Thread 1.2 Test Harness, DUA-TC-01: Basic DUA registration and communication.
**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id cf5637f3cabf70b39d153cc4ba61be6eb1b6461b
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology: As described in test spec.

We have observed some occasional failures for this specifc test, and from our debugging log, we have seen a dependency between fail/pass status and `mDelay.mFields.mReregistrationDelay` value from `dua_manager.cpp`.

It seems that in this specific test case, `mDelay.mFields.mRegistrationDelay` is set to `kNewDuaRegistrationDelay` from `UpdateRegistrationDelay(kNewDuaRegistrationDelay)`, meaning `mDelay.mFields.mRegistrationDelay` will have a value of 1.
`mDelay.mFields.mReregistrationDelay` will be set from `DuaManager::UpdateReregistrationDelay` and this value will be between `[1,config.mReregistrationDelay], or 1`,  `delay = config.mReregistrationDelay > 1 ? Random::NonCrypto::GetUint16InRange(1, config.mReregistrationDelay) : 1`.

`DuaManager::UpdateRegistrationDelay(uint8_t aDelay)` will call `UpdateTimeTickerRegistration` which will either register or unregister a receiver in `TimeTicker` module.

My thought is that as `DuaManager::HandleTimeTick` decrements both `mDelay.mFields.mRegistrationDelay` and `mDelay.mFields.mReregistrationDelay`, and as they are both equal to 1, they will become 0 and as `UpdateTimeTickerRegistration()` will be called, `TimeTicker::kDuaManager` will be unregistered from `TimeTicker` module. This will never allow `DuaManager::HandleTimeTick` to be executed, thus `mRegistrationTask` will never get to be posted and eventually attempt a DUA registration if all conditions are met.

Just for validation's sake, I've did a dummy code, that will set the `mDelay.mFields.mReregistrationDelay` to 2 if the chosen value was 1, and the test had no failure. Initially we had a 40% fail rate out of 10 runs.

What are your thoughts on this scenario?

**Console/log output** If applicable, add console/log output to help explain your problem.
Left side log illustrates a pass test and left side a failure.
[image]

**Additional context** Add any other context about the problem here.

librasungirl, 2024-07-31:
@Cristib05 Thanks for your triage and so much details here. 

When both `mDelay.mFields.mRegistrationDelay` and `mDelay.mFields.mReregistrationDelay` reaches 0 at the same time in `HandleTimeTick()` , the task has already posted before  unregistering from `TimeTicker` module in `UpdateTimeTickerRegistration`.  Intentionally update reregdelay as 2 just gives it another chance to register.

https://github.com/openthread/openthread/blob/5493815126897db1dc2ad493543aa992d1944178/src/core/thread/dua_manager.cpp#L411-L416

Would you please help to try if https://github.com/openthread/openthread/pull/10559 helps?

Cristib05, 2024-07-31:
> @Cristib05 Thanks for your triage and so much details here.
> 
> When both `mDelay.mFields.mRegistrationDelay` and `mDelay.mFields.mReregistrationDelay` reaches 0 at the same time in `HandleTimeTick()` , the task has already posted before unregistering from `TimeTicker` module in `UpdateTimeTickerRegistration`. Intentionally update reregdelay as 2 just gives it another chance to post/register.
> 
> https://github.com/openthread/openthread/blob/5493815126897db1dc2ad493543aa992d1944178/src/core/thread/dua_manager.cpp#L411-L416
> 
> Would you please help to try if #10559 helps?

@librasungirl I had the chance to test using https://github.com/openthread/openthread/pull/10559 and I couldn't reproduce the issue anymore. Thanks for your help!

jwhui, 2024-07-31:
@Cristib05 , thanks for raising this issue and validating the fix!

closed 2024-08-01 by jwhui