openthread/openthread #10567: [tests] unexpected failing of MATN_15_ChangeOfPrimaryBBRTriggersRegistration

librasungirl, 2024-08-01:
https://github.com/openthread/openthread/actions/runs/10171961935/job/28133812905?pr=10559

PASS 0 FAIL 1 ./tests/scripts/thread-cert/border_router/MATN/MATN_15_ChangeOfPrimaryBBRTriggersRegistration.py

librasungirl, 2024-08-06:
another occurence

https://github.com/openthread/openthread/actions/runs/10256837839/job/28376746403?pr=10503#step:8:8035

librasungirl, 2024-08-28:
another one
[MATN_15_ChangeOfPrimaryBBRTriggersRegistration](https://github.com/openthread/openthread/actions/runs/10590920799/job/29347461202?pr=10641#step:8:8178)

librasungirl, 2024-09-29:
another one [MATN_15_ChangeOfPrimaryBBRTriggersRegistration](https://github.com/openthread/openthread/actions/runs/11089666892/job/30811081685?pr=10750#step:8:8324)
```
FAIL: test (__main__.MATN_15_ChangeOfPrimaryBBRTriggersRegistration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/border_router/MATN/MATN_15_ChangeOfPrimaryBBRTriggersRegistration.py", line 141, in test
    self.assertTrue(br2.is_primary_backbone_router)
AssertionError: False is not true
```

superwhd, 2024-09-30:
If I remember correctly, I previously looked at the BBR code but I failed to find the code that specifically implemented the timeout mechanism referred by MATN 15 test case. I wonder if that was really implemented. 

cc @sunytt