openthread/openthread #10568: [tests] unexpected faliing test_publish_meshcop_service

librasungirl, 2024-08-01:
Two occurrences 
https://github.com/openthread/openthread/actions/runs/10171961935/job/28191374337?pr=10559
https://github.com/openthread/openthread/actions/runs/10171961935/job/28188997616?pr=10559

```
======================================================================
ERROR: test (__main__.PublishMeshCopService)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/border_router/test_publish_meshcop_service.py", line 135, in test
    br1.factory_reset()
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 2569, in factory_reset
    self._reset('factoryreset')
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 2575, in _reset
    self.send_command('version', expect_command_echo=False)
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/node.py", line 905, in send_command
    self.pexpect.send(cmd + '\n')
  File "/home/runner/.local/lib/python3.8/site-packages/pexpect/popen_spawn.py", line 142, in send
    return self.proc.stdin.write(b)
BrokenPipeError: [Errno 32] Broken pipe
```

librasungirl, 2024-08-05:
Another failing for different reason

https://github.com/openthread/openthread/actions/runs/10213412027/job/28258738097?pr=10549

```
FAIL: test (__main__.PublishMeshCopService)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/border_router/test_publish_meshcop_service.py", line 153, in test
    self.assertEqual(len(host.browse_mdns_services('_meshcop._udp')), 2)
AssertionError: 3 != 2

----------------------------------------------------------------------
Ran 1 test in 269.505s

FAILED (failures=1)
File "./tests/scripts/thread-cert/run_cert_suite.py", line 84, in run_cert
2024-08-02 09:59:23,321 - ERROR - Run test test_publish_meshcop_service_0 failed, please check the log file: ot_testing/test_publish_meshcop_service_0.log
Traceback (most recent call last):
  File "./tests/scripts/thread-cert/run_cert_suite.py", line 76, in run_cert
    subprocess.check_call(abs_script,
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '/home/runner/work/openthread/openthread/tests/scripts/thread-cert/border_router/test_publish_meshcop_service.py' returned non-zero exit status 1.
PASS 0 FAIL 1 ./tests/scripts/thread-cert/border_router/test_publish_meshcop_service.py
```

abtink, 2024-10-09:
This test was failing when run with the changes in this pull request: 
- https://github.com/openthread/openthread/pull/10798.

The failure appears to be due to a short wait time after restarting the leader (stopping MLE and restarting it). Upon restart, a device will first try to restore its previous role by sending multicast Link Requests (a leader would send 6, each about 5 seconds apart, with a 10% random jitter added to every 5-second interval). After this, 6 Parent Requests are sent before the device becomes the leader.

The wait time in this test was marginal, meaning the test would pass if the random intervals picked were on the lower end and could fail if a higher random intervals were picked.

This is now addressed in the aforementioned pull request (a long enough wait time is used to accommodate the worst-case scenario).

abtink, 2025-05-20:
This test is still flaky and fails often.

An instance can be found here: [https://github.com/openthread/openthread/actions/runs/15146823648/job/42584455174?pr=11513](https://github.com/openthread/openthread/actions/runs/15146823648/job/42584455174?pr=11513)

```
======================================================================
FAIL: test (__main__.PublishMeshCopService)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/border_router/test_publish_meshcop_service.py", line 196, in test
    self.assertEqual(len(host.browse_mdns_services('_meshcop._udp')), 2)
AssertionError: 3 != 2
```

This seems to be the same issue mentioned in @librasungirl's comment at [https://github.com/openthread/openthread/issues/10568#issuecomment-2267995186](https://github.com/openthread/openthread/issues/10568#issuecomment-2267995186).

The failing step in the test is the following (at the end):

```python
        br1.factory_reset()

        dataset = {
            'timestamp': 1,
            'channel': config.CHANNEL,
            'channel_mask': config.CHANNEL_MASK,
            'extended_panid': config.EXTENDED_PANID,
            'mesh_local_prefix': config.MESH_LOCAL_PREFIX.split('/')[0],
            'network_key': binascii.hexlify(config.DEFAULT_NETWORK_KEY).decode(),
            'network_name': 'ot-br-1-3',
            'panid': config.PANID,
            'pskc': config.PSKC,
            'security_policy': config.SECURITY_POLICY,
        }

        br1.set_active_dataset(**dataset)
        self.simulator.go(10)

        self.assertEqual(len(host.browse_mdns_services('_meshcop._udp')), 2)
        self.check_meshcop_service(br1, host)
        self.check_meshcop_service(br2, host)
```

The failing line is `self.assertEqual(len(host.browse_mdns_services('_meshcop._udp')), 2)`.

We factory reset `br1` and then expect to see two `_meshcop._udp` services, but we somehow see `3`. It seems the old service (before the factory reset) might still be present or seen.

One thing I noticed is the `self.simulator.go(10)` after the `br1` factory reset. We usually use longer start-up times; e.g., `config.LEADER_REBOOT_DELAY` may be a better value than the hardcoded `10`. I will submit a PR to change this.

However, it is not clear how we end up seeing `3` services (while expecting `2`).

abtink, 2025-05-20:
Another observation: it seems the failure is always seen with the "**BR (mDNSResponder, TREL =1)**" config. Somehow, enabling TREL seems to cause this (perhaps due to TREL also registering an mDNS service?).

abtink, 2025-05-22:
I managed to reproduce this, with more logs available at [https://github.com/openthread/openthread/actions/runs/15148523964/job/42733387942](https://github.com/openthread/openthread/actions/runs/15148523964/job/42733387942).

We see the following:

```
======================================================================
FAIL: test (__main__.PublishMeshCopService)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/openthread/openthread/tests/scripts/thread-cert/border_router/test_publish_meshcop_service.py", line 198, in test
    self.assertEqual(len(browse_result), 2, browse_result)
AssertionError: 3 != 2 : ['OpenThread\\032BorderRouter\\032#E0CE', 'OpenThread\\032BorderRouter\\032#0001', 'OpenThread\\032BorderRouter\\032#1622']
```

We expect to see 2 services on mDNS but end up seeing 3. The three services seen are indeed of the "OpenThread BorderRouter #{xyz}" format, which shows they were published by the OTBRs.

**This seems to be an issue with the underlying mDNS still caching earlier service entries (which should have been removed)**. I assume that in the test step where `br1` is factory reset, it is an abrupt reset, and therefore the Border Agent and other functions are not given the chance to stop properly and consequently remove any previously registered mDNS entries and advertise their removal. Consequently, such entries may remain in the mDNS cache.

I will relax this test to allow for more entries to accommodate such situations.