openthread/openthread #10588: Unicast NS sent from should contain source link-layer address option.

zwx1995esp, 2024-08-07:
**Describe the bug** A clear and concise description of what the bug is.
In OT implement https://github.com/openthread/openthread/blob/main/src/core/net/nd6.hpp#L1057, the NS packets never contains source link-layer address option. According to [RFC-4861 section-4.3](https://www.rfc-editor.org/rfc/rfc4861#section-4.3), unicast NS should contain the source link-layer address option.

**Expected behavior** 
The unicast NS sent from OT should contain source link-layer address option.

abtink, 2024-08-07:
Thanks @zwx1995esp.

`RoutingManager` uses unicast NS to validate that other discovered routers on AIL are still present.

Per RFC-4861, unlike multicast NS where the source link-layer address option is a MUST requirement, for unicast NS it's SHOULD, meaning it can be omitted under certain circumstances (RFC2119). Therefore, NS receivers should not expect this option to always be present.

Does the absence of this option cause issues in certain cases? Could you elaborate on the reason behind this?

The challenge is that within the OT core (`RoutingManager`), we don't have direct access to the link-layer address as it belongs to the AIL netif (e.g., WiFi netif on the device). Obtaining this would require a new platform  API to be defined to retrieve the AIL netif's link-layer address, which would then need to be implemented on various platforms. This is technically feasible but adds complexity and additional effort if not strictly necessary.

zwx1995esp, 2024-08-08:
Hi, @abtink  thanks for your explaination. Base our test, for the [lwIP stack](https://savannah.nongnu.org/projects/lwip/), it will ignore the unicast NS without source link-layer address option. I also know that the OpenThread SDK currently does not require the MAC address when the infrastructure network interface initializes. 

Would it be possible to add some APIs to set this MAC address for the AIL link? If the platform provides it, we could include the source link-layer address option in the ND6 packets; otherwise, we could leave the ND6 packets as they are now. As you mentioned, I think incorporating this logic could enhance the compatibility and robustness of the OpenThread stack.

abtink, 2024-08-08:
> Would it be possible to add some APIs to set this MAC address for the AIL link? If the platform provides it, we could include the source link-layer address option in the ND6 packets; otherwise, we could leave the ND6 packets as they are now

This can work. Would you like to contribute this? If not, I can help add it later when I have some free time (might take a week or more).

Regardless, I strongly recommend changing/fixing the lwIP stack to accept NS even without the option, especially if the lwip stack is being used in a Thread Border Router.

zwx1995esp, 2024-08-09:
Ok, Thanks @abtink I will  create a PR to add this feature, and for lwIP, I will also create an issue for them.

EskoDijk, 2024-08-15:
I have now seen more cases like this, where an open source component doesn't strictly comply to IETF requirements, and often there's 1) a huge installed/legacy base of devices with that component and 2) unwillingness by maintainers to change because "it has been like this for years/decades and it always worked".  As a practical solution often we adapt the tests or the code under our control to cope with the status quo of implementations. 

To me it sounds strange to ignore the SHOULD requirement, normally it's expected to include SLLAO if the sender has a valid link-layer address. To not send the option requires a very good reason I think! Especially if we know of stacks in the field that (mistakenly) depend on it.

The other cases I mentioned where issues like this appeared are BSD (MLDv2 protocol bugs explicitly introduced as required features) and mDNS (Avahi).

zwx1995esp, 2024-08-28:
Hi, @abtink I created a PR for fixing this issue, could you please take a look?

EskoDijk, 2024-10-15:
@zwx1995esp Possibly this issue can be closed now?

zwx1995esp, 2024-10-15:
Yes, I think this issue can be closed due to [PR-10647](https://github.com/openthread/openthread/pull/10647) merged.

closed 2024-10-15 by zwx1995esp