openthread/openthread #10612: Issue in method of storing/retrieving BBR Sequence Number

EskoDijk, 2024-08-15:
**Describe the bug** A clear and concise description of what the bug is.
For the BR, there's a requirement to store the BBR Sequence Number (and increase by +1) when a BR reboots. Instead of storing that in NVM, an allowed alternative method is to retrieve the old (previous) BBR Seq Nr from network data when the BR boots up, increasing it +1, publishing it and then storing it in RAM again.

However, in OT BR it seems that the following problem may occur in the procedure:

* the OT BR waits with restoring the BBR Seq Nr from network data until the AIL infrastructure link is fully active;
* and the AIL infrastructure link may take very long to be activated again after the reboot (in some products/situations);
* and the old lingering BBR Seq Nr info may at some point be purged by the Leader;
* therefore by the time the BR wants to restore the BBR Seq Nr from network data, it's already deleted from there.

**To Reproduce** Information to reproduce the behavior, including:

See [TESTPLAN-598](https://threadgroup.atlassian.net/browse/TESTPLAN-598) for some more details.

**Expected behavior** A clear and concise description of what you expected to happen.

What's needed:
* when BR boots up and first network data is received, it stores/buffers old BBR Seq Nr in RAM. (Even though it may never be used later on.)
* when BR activates its BBR function and publishes the BBR Dataset, it retrieves the old BBR Seq Nr from this RAM location.
* when BR increases the BBR Seq Nr +1, it writes the new Seq Nr into the RAM location.
* when the BBR function gets deactivated and activated again while the BR was just running (no reboots), the BBR function gets the "old BBR Seq Nr" from the RAM location and not from network data. (Because it might not be anymore in the network data.)
* said RAM location exists independent of whether the BBR function is enabled or disabled, etc.

**Console/log output** If applicable, add console/log output to help explain your problem.
None

**Additional context** Add any other context about the problem here.
See TESTPLAN-598.
Also I didn't check how the current implementation works, sorry! This issue is based on behavior described in TESTPLAN-598 only.

librasungirl, 2024-08-20:
Some initial update:

> the OT BR waits with restoring the BBR Seq Nr from network data until the AIL infrastructure link is fully active;

It's device/product specific behavior, that is likely override [`OTBR_ENABLE_BACKBONE_ROUTER_ON_INIT`](https://github.com/openthread/ot-br-posix/blob/6fcf269e28182b2fb6ba7f4f4d81cc49abaf7fdb/src/backbone_router/backbone_agent.hpp#L40) as 0 intentionally, which should not impact this BBR DUT's restoring  mSequenceNumber as in https://github.com/openthread/openthread/blob/fb7b457ab225f9a5b32c359b47eaf51a9f73a46a/src/core/backbone_router/bbr_local.cpp#L277-L280

(_though the `AddService` will fail as BBR functionality not enabled yet due to infra not ready yet). In theory, the increased sequence number restored from netdata should be used when infra link is ready and BBR is enabled. 

Need to further figure out if there are some unnoticed reset/local config in the TESTPLAN issue.

EskoDijk, 2024-08-28:
@librasungirl Thanks for looking into this. To me it seems that the retrieval (attempt) of prior sequence number in RAM needs to be done always when the BR boots up, even when the BBR function only gets enabled later on (e.g. some seconds or minutes later). Not sure if the current code does this, or that it is dependent on build configuration?

librasungirl, 2024-09-02:
Ah, good catch, @EskoDijk !

There might be no chance to restore sequence number from network data if it's not enabled before attached.
https://github.com/openthread/openthread/blob/fb7b457ab225f9a5b32c359b47eaf51a9f73a46a/src/core/backbone_router/bbr_local.cpp#L254

Will submit a fix

abtink, 2025-06-20:
@librasungirl , do you know if this is fixed? Can we close this?

closed 2026-11-12 by jwhui