openthread/openthread #10633: MAC security frame counter can be saved / restored incorrectly during RCP recovery

jcblomquist, 2024-08-23:
**Describe the bug**

During the RestoreProperties() step of RCP recovery, the MAC security frame counter is [restored from the global variable mMacFrameCounter](https://github.com/openthread/openthread/blob/1c5ad3403d624c0e57b7a6462aa82d67bf46c690/src/lib/spinel/radio_spinel.cpp#L2166). This variable is set in several places, but most notably in [HandleTransmitDone](https://github.com/openthread/openthread/blob/main/src/lib/spinel/radio_spinel.cpp#L1581). In the current implementation of this callback, mMacFrameCounter is overwritten for every incoming MAC frame counter value. This is an issue because Openthread uses two different MAC security frame counters, depending on the MAC Key ID Mode that is used to transmit the packet. Most packets use MAC Key ID Mode 0/1, but (at least) the MLE announce packet uses MAC Key Mode 2, which maintains a different MAC Security frame counter global. In rare cases where the last packet you transmitted was an MLE announce packet, and RCP recovery is triggered, the RCP recovery will restore the Key ID Mode 2 frame counter (+ 1000, the guard value), which will effectively brick the device until the power is cycled and the correct frame counter is restored from the non-volatile memory. It's possible that the incorrect frame counter could be written back to the non-volatile memory causing greater issues, but I have not confirmed this, maybe there is a check which only updates the non-volatile memory if the value is greater.

Note: this is specifically for the "Unexpected RCP reset" which restores values of the frame counter from RAM since the host service is not reset. There are other types of resets, such as restarting otbr-agent, but those will restore the frame counter from non-volatile memory.

**To Reproduce** 

1. Git commit id
Should be able to reproduce on TOT

3. IEEE 802.15.4 hardware platform
Any platform which you can force an "Unexpected RCP reset: RESET_SOFTWARE" test flow

5. Build steps
N/A, I'm using Amazon's platform

7. Network topology
DUT == network leader
node_1 == any peripheral which you can ping

**Expected behavior** 
1. Form network with DUT, join node_1, confirm ping works
2. Perform reboot / RCP recovery on the DUT several times in order to increment the MAC Security frame counter to a sufficiently large value (>2000)
3. Monitor the logging system and wait for the MLE announce timer to fire (I increased it to a 15s cycle in my test setup)
4. The moment an MLE announce is sent, trigger the "Unexpected RCP reset"
5. Observe that the MAC frame counter is restored incorrectly. Try to ping the peripheral to confirm that it does not respond.

**Console/log output** 
Shell output
```
sh:/ # ./rcp_reset.sh
sh:/ # ping6 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621                                                                                       
PING 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621(96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621) 56 data bytes
64 bytes from 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621: icmp_seq=1 ttl=64 time=43.9 ms
64 bytes from 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621: icmp_seq=2 ttl=64 time=26.2 ms
64 bytes from 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621: icmp_seq=3 ttl=64 time=49.5 ms
64 bytes from 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621: icmp_seq=4 ttl=64 time=25.3 ms
64 bytes from 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621: icmp_seq=5 ttl=64 time=26.9 ms
64 bytes from 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621: icmp_seq=6 ttl=64 time=26.8 ms
64 bytes from 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621: icmp_seq=7 ttl=64 time=26.1 ms
64 bytes from 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621: icmp_seq=8 ttl=64 time=28.3 ms

--- 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621 ping statistics ---
8 packets transmitted, 8 received, 0% packet loss, time 7009ms
rtt min/avg/max/mdev = 25.374/31.695/49.551/8.850 ms

# wait for MLE announce

sh:/ # ./rcp_reset.sh
sh:/ # ping6 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621                                                                                       
PING 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621(96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621) 56 data bytes
^C
--- 96e1:74a8:3325:8ca0:4fe2:f7a5:daa8:9621 ping statistics ---
10 packets transmitted, 0 received, 100% packet loss, time 9210ms
```

Logging system output
```
D otbr-agent: 00:04:53.999 [D] P-RadioSpinel-: Sent spinel frame, flg:0x2, tid:7, cmd:PROP_VALUE_SET, key:RCP_MAC_FRAME_COUNTER, counter:3052
E otbr-agent: 00:05:01.853 [C] P-RadioSpinel-: Unexpected RCP reset: RESET_SOFTWARE
I otbr-agent: 00:05:01.859 [I] P-RadioSpinel-: RCP reset: RESET_POWER_ON
D otbr-agent: 00:05:01.907 [D] P-RadioSpinel-: Sent spinel frame, flg:0x2, tid:7, cmd:PROP_VALUE_SET, key:RCP_MAC_FRAME_COUNTER, counter:4056
I otbr-agent: 00:06:47.045 [I] Mle-----------: Send Announce on channel 20
E otbr-agent: 00:06:49.928 [C] P-RadioSpinel-: Unexpected RCP reset: RESET_SOFTWARE
I otbr-agent: 00:06:49.936 [I] P-RadioSpinel-: RCP reset: RESET_POWER_ON
D otbr-agent: 00:06:49.988 [D] P-RadioSpinel-: Sent spinel frame, flg:0x2, tid:7, cmd:PROP_VALUE_SET, key:RCP_MAC_FRAME_COUNTER, counter:1001
```

You can see above that the MAC security frame counter is restored to 1 (key mode 2 frame counter) + 1000 = 1001 after the RCP reset.

abtink, 2024-08-23:
@jcblomquist , thanks for finding, debugging, and reporting this with detailed information. 👍 

Checking on this, noticed that earlier versions did read the counter from non-volatile settings to restore it on RCP. 
- This was changed in #9407 where the MAC frame counter was tracked directly in `RadioSpinel` in a local variable.

Some possible solutions:
- Revert back to the earlier version that reads from non-volatile storage.
- Update the code in `RadioSpinel::HandleTransmitDone()` where `mMacFrameCounter` is updated to ensure it is only done when we see  proper key-id and key-id mode.

@zhanglongxia, @bukepo any thoughts?

abtink, 2024-08-23:
@jcblomquist I saw you already submitted [https://github.com/openthread/openthread/pull/10634](https://github.com/openthread/openthread/pull/10634). Thanks! This can work.

Another idea:

The frame counter is already tracked by `SubMac` in the OT core (which validates the key id to match). There's no need for `RadioSpinel` to redo the same. We can add a new API to get the current frame counter from the OT core (e.g. `otLinkGetFrameCounter()`) and then use this API in `RadioSpinel` during RCP restoration. This way, the logic of tracking the frame counter is common, and if it is changed, it won't impact `RadioSpinel` and RCP restoration code.

jcblomquist, 2024-08-23:
For what it's worth, with the current implementation in RadioSpinel::HandleTransmitDone(), the aBuffer parameter seems to only contain 5 bytes, keyID + 32-bit frame counter, so it's only possible to check it against the keyID in there.

abtink, 2024-08-23:
> the relevant part of the aBuffer parameter seems to only contain 5 bytes, keyID + 32-bit frame counter, so it's only possible to check it against the keyID in there.

Exactly. The alternative would be to parse the frame in `RadioSpinel` to remember/check the Key ID Mode and Key ID, which makes it more complex and repeats the work already done in `SubMac`.

I've submitted [https://github.com/openthread/openthread/pull/10635](https://github.com/openthread/openthread/pull/10635), which adds `otLinkGetFrameCounter()` and has `RadioSpinel` use it instead of tracking the counter on its own.

jcblomquist, 2024-08-23:
Great, I checked your PR out locally and tested it and I can confirm it also works. Thanks for the quick support.

PR #10635 (2024-08-27): [radio-spinel] fix tracking of current MAC frame counter
This commit fixes an issue in `RadioSpinel` related to the tracking of the MAC frame counter used during RCP restoration. Previously, `RadioSpinel` tracked the last seen counter on any secured received/transmitted frame. However, the frame counter should only be tracked for frames that use Key ID Mode 1 and where the included Key ID in the frame matches the current Key ID being used. The `SubMac` module also tracks the current MAC frame counter, and it does perform Key ID Mode and Key ID checks.

This commit adds a new public API `otLink [... cut]
closed 2024-08-27 by jwhui