openthread/openthread #10673: What situations may cause otPlatSettingsGet Failure?

lancersky2017, 2024-09-05:
**Describe the bug** A clear and concise description of what the bug is.
When otPlatSettingsGet Failure occurs, otbr-agent will restart. We don't know what situations will cause otPlatSettingsGet Failure.
```
Thu Sep  5 13:57:25 2024 user.warn otbr-agent[2911]: 1d.00:11:43.738 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:57:27 2024 user.warn otbr-agent[2911]: 1d.00:11:46.191 [W] P-Daemon------: Daemon read: Connection reset by peer
Thu Sep  5 13:57:27 2024 user.warn otbr-agent[2911]: 1d.00:11:46.211 [W] P-Daemon------: Daemon read: Connection reset by peer
Thu Sep  5 13:57:27 2024 user.warn otbr-agent[2911]: 1d.00:11:46.238 [W] P-Daemon------: Daemon read: Connection reset by peer
Thu Sep  5 13:57:34 2024 user.warn otbr-agent[2911]: 1d.00:11:53.303 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:57:48 2024 user.warn otbr-agent[2911]: 1d.00:12:07.191 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:58:03 2024 user.warn otbr-agent[2911]: 1d.00:12:21.592 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:58:15 2024 user.warn otbr-agent[2911]: 1d.00:12:34.406 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:58:16 2024 user.warn otbr-agent[2911]: 1d.00:12:35.327 [W] Mle-----------: Failed to process UDP: Duplicated
Thu Sep  5 13:58:17 2024 user.warn otbr-agent[2911]: 1d.00:12:36.383 [W] Mle-----------: Failed to process UDP: Duplicated
Thu Sep  5 13:58:26 2024 user.warn otbr-agent[2911]: 1d.00:12:45.014 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:58:36 2024 user.warn otbr-agent[2911]: 1d.00:12:55.167 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:58:47 2024 user.warn otbr-agent[2911]: 1d.00:13:06.443 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:59:03 2024 user.warn otbr-agent[2911]: 1d.00:13:22.240 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:59:15 2024 user.warn otbr-agent[2911]: 1d.00:13:34.335 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:59:25 2024 user.warn otbr-agent[2911]: 1d.00:13:44.330 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:59:34 2024 user.warn otbr-agent[2911]: 1d.00:13:52.860 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:59:50 2024 user.warn otbr-agent[2911]: 1d.00:14:09.522 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 13:59:59 2024 user.warn otbr-agent[2911]: 1d.00:14:18.290 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:00:13 2024 user.warn otbr-agent[2911]: 1d.00:14:31.821 [W] P-RadioSpinel-: Handle transmit done failed: ChannelAccessFailure
Thu Sep  5 14:00:15 2024 user.warn otbr-agent[2911]: 1d.00:14:34.030 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:00:25 2024 user.warn otbr-agent[2911]: 1d.00:14:44.156 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:00:35 2024 user.warn otbr-agent[2911]: 1d.00:14:53.705 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:00:46 2024 user.warn otbr-agent[2911]: 1d.00:15:04.732 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:00:58 2024 user.warn otbr-agent[2911]: 1d.00:15:16.852 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:01:10 2024 user.warn otbr-agent[2911]: 1d.00:15:29.408 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:01:22 2024 user.warn otbr-agent[2911]: 1d.00:15:41.452 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:01:34 2024 user.warn otbr-agent[2911]: 1d.00:15:53.461 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:01:46 2024 user.warn otbr-agent[2911]: 1d.00:16:05.017 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:02:02 2024 user.warn otbr-agent[2911]: 1d.00:16:21.238 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:02:13 2024 user.warn otbr-agent[2911]: 1d.00:16:32.371 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:02:24 2024 user.warn otbr-agent[2911]: 1d.00:16:43.302 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:02:35 2024 user.warn otbr-agent[2911]: 1d.00:16:53.798 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:02:49 2024 user.warn otbr-agent[2911]: 1d.00:17:07.788 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:03:02 2024 user.warn otbr-agent[2911]: 1d.00:17:21.457 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:03:10 2024 user.warn otbr-agent[2911]: 1d.00:17:29.108 [W] Mle-----------: Failed to process UDP: Security
Thu Sep  5 14:03:17 2024 user.warn otbr-agent[2911]: 1d.00:17:36.198 [W] P-Netif-------: ADD [U] fd36:ff76:692:a2ba:0:ff:fe00:fc37 failed (InvalidArgs)
Thu Sep  5 14:03:17 2024 user.warn otbr-agent[2911]: 1d.00:17:36.198 [W] P-Netif-------: Failed to process event, error:InvalidArgs
Thu Sep  5 14:03:17 2024 user.err otbr-agent[2911]: [ERR ]-MDNS----: Failed to register service OpenThread BorderRouter #A858 (51733)._meshcop._udp: Name Conflict
Thu Sep  5 14:03:17 2024 user.warn otbr-agent[2911]: [WARN]-BA------: Result of publish meshcop service OpenThread BorderRouter #A858 (51733)._meshcop._udp.local: Duplicated
Thu Sep  5 14:03:17 2024 user.crit otbr-agent[2911]: 1d.00:17:36.257 [C] Platform------: otPlatSettingsGet() at settings.cpp:259: Failure
Thu Sep  5 14:03:17 2024 user.crit otbr-agent[2911]: 1d.00:17:36.257 [C] Platform------: otPlatSettingsGet() at settings.cpp:259: Failure

```
**To Reproduce** Information to reproduce the behavior, including:
No exact steps

1. Git commit id [simplicity_sdk v2024.6.1](https://github.com/SiliconLabs/simplicity_sdk/releases/tag/v2024.6.1-0)
3. IEEE 802.15.4 hardware platform **EFR32MG21F768**
4. Build steps
5. Network topology

bukepo, 2024-09-05:
From the code, it can happen when the settings file was corrupted.

lancersky2017, 2024-09-06:
Thank you for your reply. I will continue to observe what API call causes the file to be corrupted.

lancersky2017, 2024-09-09:
Hi @bukepo 
How to interpret OT_POSIX_SETTINGS? I would add some debug logs to see what is failing to read.

jwhui, 2024-09-10:
> Hi @bukepo How to interpret OT_POSIX_SETTINGS? I would add some debug logs to see what is failing to read.

The process dies whenever `ot::Posix::PlatformSettingsGet()` returns an `OT_ERROR_PARSE` error:
https://github.com/openthread/openthread/blob/07a1b7bd985d8d1cd9fab5d45bc62ae8fc1358c9/src/posix/platform/settings.cpp#L260

I would try to understand why `PlatformSettingsGet()` is returning `OT_ERROR_PARSE`:
https://github.com/openthread/openthread/blob/07a1b7bd985d8d1cd9fab5d45bc62ae8fc1358c9/src/posix/platform/settings.cpp#L333

lancersky2017, 2024-09-11:
Hi @jwhui 
We haven't found the steps to reproduce it yet. After `otbr-agent` is restarted, otPlatSettingsGet failure does not occur immediately again. We want to understand what information is stored in OT_POSIX_SETTINGS, which may help us locate the problem.
```
Sep 09 05:51:44 192.168.251.18 Sep  9 05:51:26 KNBA5F00001E otbr-agent[7878]: 2d.08:30:50.877 [W] SrpServer-----: Name conflict: host name knx-00fd10f1c763.default.service.arpa. has already been allocated
Sep 09 05:51:44 192.168.251.18 Sep  9 05:51:26 KNBA5F00001E otbr-agent[7878]: 2d.08:30:50.877 [W] SrpServer-----: Failed to Process DNS Update section: Duplicated
Sep 09 05:51:44 192.168.251.18 Sep  9 05:51:26 KNBA5F00001E otbr-agent[7878]: 2d.08:30:50.878 [W] SrpServer-----: Send fail response: 6
Sep 09 05:51:45 192.168.251.18 Sep  9 05:51:26 KNBA5F00001E otbr-agent[7878]: 2d.08:30:51.349 [W] P-Daemon------: Daemon read: Connection reset by peer
Sep 09 05:51:48 192.168.251.18 Sep  9 05:51:30 KNBA5F00001E otbr-agent[7878]: 2d.08:30:54.693 [W] SrpServer-----: Name conflict: service name 00FD10F1C763._knx._udp.default.service.arpa. has already been allocated
Sep 09 05:51:48 192.168.251.18 Sep  9 05:51:30 KNBA5F00001E otbr-agent[7878]: 2d.08:30:54.693 [W] SrpServer-----: Failed to Process DNS Update section: Duplicated
Sep 09 05:51:48 192.168.251.18 Sep  9 05:51:30 KNBA5F00001E otbr-agent[7878]: 2d.08:30:54.694 [W] SrpServer-----: Send fail response: 6
Sep 09 05:51:55 192.168.251.18 Sep  9 05:51:37 KNBA5F00001E otbr-agent[7878]: 2d.08:31:01.896 [W] Mle-----------: Failed to process UDP: Security
Sep 09 05:51:59 192.168.251.18 Sep  9 05:51:41 KNBA5F00001E otbr-agent[7878]: 2d.08:31:06.308 [W] SrpServer-----: Name conflict: service name 00FD10F1C763._knx._udp.default.service.arpa. has already been allocated
Sep 09 05:51:59 192.168.251.18 Sep  9 05:51:41 KNBA5F00001E otbr-agent[7878]: 2d.08:31:06.308 [W] SrpServer-----: Failed to Process DNS Update section: Duplicated
Sep 09 05:51:59 192.168.251.18 Sep  9 05:51:41 KNBA5F00001E otbr-agent[7878]: 2d.08:31:06.309 [W] SrpServer-----: Send fail response: 6
Sep 09 05:52:06 192.168.251.18 Sep  9 05:51:48 KNBA5F00001E otbr-agent[7878]: 2d.08:31:13.229 [W] Mle-----------: Failed to process UDP: Security
Sep 09 05:52:11 192.168.251.18 Sep  9 05:51:53 KNBA5F00001E otbr-agent[7878]: 2d.08:31:17.945 [W] SrpServer-----: Name conflict: service name 00FD10F1C763._knx._udp.default.service.arpa. has already been allocated
Sep 09 05:52:11 192.168.251.18 Sep  9 05:51:53 KNBA5F00001E otbr-agent[7878]: 2d.08:31:17.945 [W] SrpServer-----: Failed to Process DNS Update section: Duplicated
Sep 09 05:52:11 192.168.251.18 Sep  9 05:51:53 KNBA5F00001E otbr-agent[7878]: 2d.08:31:17.946 [W] SrpServer-----: Send fail response: 6
Sep 09 05:52:17 192.168.251.18 Sep  9 05:51:59 KNBA5F00001E otbr-agent[7878]: 2d.08:31:23.773 [W] Mle-----------: Failed to process UDP: Security
Sep 09 05:52:23 192.168.251.18 Sep  9 05:52:05 KNBA5F00001E otbr-agent[7878]: 2d.08:31:29.599 [W] SrpServer-----: Name conflict: service name 00FD10F1C763._knx._udp.default.service.arpa. has already been allocated
Sep 09 05:52:23 192.168.251.18 Sep  9 05:52:05 KNBA5F00001E otbr-agent[7878]: 2d.08:31:29.600 [W] SrpServer-----: Failed to Process DNS Update section: Duplicated
Sep 09 05:52:23 192.168.251.18 Sep  9 05:52:05 KNBA5F00001E otbr-agent[7878]: 2d.08:31:29.601 [W] SrpServer-----: Send fail response: 6
Sep 09 05:52:31 192.168.251.18 Sep  9 05:52:13 KNBA5F00001E otbr-agent[7878]: 2d.08:31:38.016 [W] Mle-----------: Failed to process UDP: Security
Sep 09 05:52:34 192.168.251.18 Sep  9 05:52:16 KNBA5F00001E otbr-agent[7878]: 2d.08:31:41.201 [W] SrpServer-----: Name conflict: service name 00FD10F1C763._knx._udp.default.service.arpa. has already been allocated
Sep 09 05:52:34 192.168.251.18 Sep  9 05:52:16 KNBA5F00001E otbr-agent[7878]: 2d.08:31:41.204 [W] SrpServer-----: Failed to Process DNS Update section: Duplicated
Sep 09 05:52:34 192.168.251.18 Sep  9 05:52:16 KNBA5F00001E otbr-agent[7878]: 2d.08:31:41.205 [W] SrpServer-----: Send fail response: 6
Sep 09 05:52:35 192.168.251.18 Sep  9 05:52:16 KNBA5F00001E otbr-agent[7878]: 2d.08:31:41.342 [C] Platform------: otPlatSettingsGet() at settings.cpp:259: Failure
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: [NOTE]-AGENT---: Running 0.3.0-v19.07.7-45-gc4b47c59e4-dirty
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: [NOTE]-AGENT---: Thread version: 1.3.0
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: [NOTE]-AGENT---: Thread interface: wpan0
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: [NOTE]-AGENT---: Radio URL: spinel+hdlc+uart:///dev/ttyCH343USB0?uart-baudrate=1000000&uart-flow-control
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: [NOTE]-AGENT---: Radio URL: trel://eth0
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: [NOTE]-ILS-----: Infra link selected: eth0
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: [INFO]-NCP-----: OpenThread log level changed to 5
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: 52d.03:41:42.574 [D] P-SpinelDrive-: Sent spinel frame, flg:0x2, iid:0, tid:0, cmd:RESET
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: 52d.03:41:42.574 [D] P-SpinelDrive-: Waiting response: key=0
[... cut ...]
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: 00:00:00.077 [I] BbrLocal------: Add Domain Prefix: ::/0, NotFound
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: 00:00:00.078 [I] BbrLocal------: Add BBR Service: seqno (3), delay (5s), timeout (3600s), InvalidState
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: [INFO]-ADPROXY-: Started
Sep 09 05:52:40 192.168.251.18 Sep  9 05:52:22 KNBA5F00001E otbr-agent[12884]: [INFO]-DPROXY--: Started

```

jwhui, 2024-09-11:
All of the non-volatile settings are defined here:

https://github.com/openthread/openthread/blob/main/src/core/common/settings.hpp

lancersky2017, 2024-09-18:
Hi @jwhui 
The left side shows the hexdump data of OT_POSIX_SETTINGS during normal operation, and the right side shows the hexdump data after otbr-agent exits when otPlatSettingsGet Failure occurs. Can you help explain the differences?
[image]

jinpeng1989, 2024-09-18:
This issue is the same issue as the following.
https://github.com/openthread/ot-br-posix/issues/2477

bukepo, 2024-09-18:
@lancersky2017 could you attach the file?

jinpeng1989, 2024-09-19:
Hi @bukepo ,
The screenshot of OT_POSIX_SETTINGS @lancersky2017 provided is not from the time this issue occurred. We will try to reproduce the issue and record the OT_POSIX_SETTINGS before and after it occurred.

jinpeng1989, 2024-09-19:
It is worth mentioning that after switching back to [GSDK 4.4.3](https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.4.3). The issue will still come up. So please give priority to this issue.

lancersky2017, 2024-09-23:
Hi @bukepo 
This is OT_POSIX_SETTINGS after `otPlatSettingsGet Failure` occurs.
[0_94deb8fffef12db9.zip](https://github.com/user-attachments/files/17092237/0_94deb8fffef12db9.zip)

bukepo, 2024-09-23:
I use the file you provided to overwrite my node's settings file, it started successfully and the network is established successfully as `KNBA5F000043`. No crashes occurred. Could you double check whether this is the settings file after `otPlatSettingsGet` failed?

jinpeng1989, 2024-09-23:
According to the log, otbr-agent has been restarted. So the above Settings file is not the content of the process crash. Because it has been overwritten by the otbr-agent settings file after the restart. We will disable the automatic restart function of otbr-agent, reproduce the issue again.

jinpeng1989, 2024-09-27:
@bukepo 
We reproduced the otbr-agent crash again.

This is the settings file that was backed up before the otbr-agent crash.
[0_94deb8fffef12db9_before_crash.zip](https://github.com/user-attachments/files/17158818/0_94deb8fffef12db9_before_crash.zip)
This is the settings file that was backed up after the otbr-agent crash. Can you check if this file will cause otbr-agent crash?
[0_94deb8fffef12db9_after_crash.zip](https://github.com/user-attachments/files/17158822/0_94deb8fffef12db9_after_crash.zip)

Unfortunately, the remote logging system crashed. I can't confirm whether this crash is the same as the crash we are concerned about.

jinpeng1989, 2024-09-29:
@bukepo 
Have you checked the 0_94deb8fffef12db9_after_crash.zip file?

bukepo, 2024-10-08:
I tried loading the settings file after crashing. I saw it loaded successfully.

bukepo, 2024-10-08:
@jinpeng1989 is there any modification to the otbr-agent?

jwhui, 2024-10-08:
> This is the settings file that was backed up before the otbr-agent crash.

@jinpeng1989 , can you share how you backed up the settings file before the crash?

jinpeng1989, 2024-10-08:
> > This is the settings file that was backed up before the otbr-agent crash.
> 
> @jinpeng1989 , can you share how you backed up the settings file before the crash?
What's wrong with this file? The tester should have copied the file directly to the local computer using "scp".

jinpeng1989, 2024-10-08:
> @jinpeng1989 is there any modification to the otbr-agent?

Yes, there are some modifications. @lancersky2017 is the developer of this otbr-agent module.

bukepo, 2024-10-08:
Could you check whether the modifications can cause multiple threads to access OpenThread instance? OpenThread core stack is designed to run in single thread. I suspect the error occurs in some race conditions.

jinpeng1989, 2024-10-08:
@bukepo 
Thanks for the reminder, there is a modification that call the otDatasetGetActiveTlvs(). @lancersky2017 has removed it. We'll verify the fix. It will take about a week. I will get back to you with the results.

jinpeng1989, 2024-10-21:
@bukepo 
There are two new pieces of information:
1 It's not the part of the ot-br-posix code that we changed that caused this issue. Because after removing the code, the issue can still be reproduced.
2 These few commands can easily reproduce the issue.
ubus call otbr commissioner start, ubus call otbr commissioner stop, ubus call otbr status

bukepo, 2024-10-22:
I think this is caused by `ubus` is running in a separate thread. The `ubus` implementation is not well maintained. To address this issue, we have to post all OpenThread API calls to the main thread from the `ubus` thread.

jinpeng1989, 2024-10-22:
The settings file seems to be designed for single-thread access. Problems may occur if multiple threads access the API of the Settings file.  Since the swap file will be renamed to the data file. If thread 1 is traversing the settings (data) file and the contents of the file are modified by thread 2, offset may not point to the position of Key. In this case, the Key, Lenth, and Value will not be resolved correctly.
https://github.com/openthread/openthread/blob/2bc86850606601db0d565cd958540a2790c08f62/src/posix/platform/settings.cpp#L135-L148
Is there a case that multiple threads are accessing the settings API at the same time?

jinpeng1989, 2024-10-23:
Do you have any plans to fix this issue?

jwhui, 2024-10-23:
OpenThread APIs are not thread safe - they need to be called from a single execution context.

jinpeng1989, 2024-10-25:
There is progress on this issue. Our extended ubus method of otbr calls the otDatasetGetActive and otDatasetGetActiveTlvs APIs. We've removed them and started a new round of tests. Thank you for your support. I will update if there is any progress.

jinpeng1989, 2024-10-25:
After verification, the issue has been solved. I think we can close the issue.

closed 2024-10-28 by jwhui