openthread/openthread #10674: FP=1 set in ACK even no pending message available for end device

aashu216, 2024-09-05:
We have RCP design where host and radio controller communicated over SPI through spinel interface.

We have observed, Frame pending bit set to 1 in ACK of reply received from end-device. 

During Debugging, Host added entry in SAM table when it has message for child. Host is removed entry from SAM table after received ACK of sent message. Due to high latency from received ACK to host and send remove entry from SAM table command, Radio received response from child before removing entry from SAM table and that is why ACK has frame pending bit is set. 

[image]

**Expected behavior** : Host should clear entry from SAM table before received reply from end device.

**Proposed solution**:  Handling remove entry from SAM table from Sub mac itself instead of handling from host.

jwhui, 2024-09-05:
In the scenario you describe above, if the SED receives an ACK with FP=1 and the parent does not have any messages queued, the host should send an empty data frame indicating that it has no more messages queued.

For your proposed solution, if the radio clears the SAM table entry and the host has more messages queued, this will cause the SED to stop polling for messages and cause high latency.

aashu216, 2024-09-06:
Yes,  you are correct. Host sent empty data frame indicating that is has no more messages queued but my concern is for RCP design regarding to clear quickly entry from SAM table. 

I think it is not good to set FP=1 as we don't have messages in queue. It is just due to entry is removed lately.  Can't we have optimize on host side or some way to clear entry quickly?

jwhui, 2024-09-06:
The radio co-processor needs explicit indication from the host that there are no more frames pending.

Maybe that explicit signal can just be the host setting/clearing the FP bit in the data frame that it is sending via indirect transmission. If the radio co-processor is sending an indirect transmission with FP bit clear, then it can remove the Source Address Table entry.

@abtink , thoughts?

jwhui, 2025-01-07:
Closing stale issue.

closed 2025-01-07 by jwhui