openthread/openthread #10677: Why can ot-ctl ping work, but sys ping fails?

lancersky2017, 2024-09-06:
**Describe the bug** A clear and concise description of what the bug is.
In the Thread network, there are two TBRs, some Routers, and more than 50 Child nodes. 
Both ot-ctl ping and sys ping work fine from another TBR, but not from this TBR.
```
root@openwrt:~# ot-ctl dataset active
Active Timestamp: 17
Channel: 25
Channel Mask: 0x07fff800
Ext PAN ID: da0b36fb69a70115
Mesh Local Prefix: fda0:d336:e5c3:cd0a::/64
Network Key: 843b072062a3d13ea152d5bbb3b1d50f
Network Name: OpenThread-f1ea
PAN ID: 0xf1ea
PSKc: fdcd3108e4b3a4c385fa5ca1630140ad
Security Policy: 672 onrc 0
Done

root@openwrt:~# ot-ctl netdata show
Prefixes:
fdb3:9162:9bb3:1::/64 paros low 1c00
Routes:
::/0 s med 1c00
::/0 s med b800
fdb3:9162:9bb3:2:0:0::/96 sn low 1c00
Services:
44970 5d fda0d336e5c3cd0aa453971ef19dad3ad122 s 1c00
44970 5d fda0d336e5c3cd0aa66e61897425b50bd120 s b800
44970 01 51000500000e10 s 1c00
Contexts:
fdb3:9162:9bb3:1::/64 2 c
Commissioning:
38041 - - -
Done

root@openwrt:~# ot-ctl br omrprefix
Local: fdcb:533f:e332:1::/64
Favored: fdb3:9162:9bb3:1::/64 prf:low
Done

root@openwrt:~# ip -6 route
dd1f:953c:ddc0::/64 dev br-lan proto static metric 1024 pref medium
unreachable dd1f:953c:ddc0::/48 dev lo proto static metric 2147483647 pref medium
dde3:9ca7:ac3::/64 dev eth0 proto static metric 256 pref medium
unreachable dde3:9ca7:ac3::/64 dev lo proto static metric 2147483647 pref medium
fd0d:e699:f095::/64 dev eth0 proto static metric 256 pref medium
unreachable fd0d:e699:f095::/64 dev lo proto static metric 2147483647 pref medium
fd1b:6a0b:4b57:e874::/64 dev eth0 proto static metric 256 pref medium
unreachable fd1b:6a0b:4b57:e874::/64 dev lo proto static metric 2147483647 pref medium
fd3b:dcfd:d29b:1::/64 via fe80::9683:c4ff:fe36:3662 dev eth0 proto static metric 512 pref medium
fd91:7d2a:4419:672e::/64 dev eth0 proto static metric 256 pref medium
unreachable fd91:7d2a:4419:672e::/64 dev lo proto static metric 2147483647 pref medium
fda0:d336:e5c3:cd0a::/64 dev wpan0 proto kernel metric 256 pref medium
fdb3:9162:9bb3:1::/64 via fe80::2a0:3ff:fe97:8a dev eth0 proto static metric 512 pref medium
fdc6:ed9c:8844:1::/64 via fe80::56ef:44ff:fe60:cb94 dev eth0 proto static metric 512 pref medium
fdeb:de:25a9:be8f::/64 dev eth0 proto static metric 256 pref medium
unreachable fdeb:de:25a9:be8f::/64 dev lo proto static metric 2147483647 pref medium
fe80::/64 dev br-lan proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev wlan0 proto kernel metric 256 pref medium
fe80::/64 dev wpan0 proto kernel metric 256 pref medium
default via fe80::e695:6eff:fe45:12a3 dev eth0 proto static metric 512 pref medium
default via fe80::e695:6eff:fe45:12a3 dev eth0 proto ra metric 1024 expires 1675sec hoplimit 64 pref medium
root@openwrt:~# 
root@openwrt:~# ot-ctl meshdiag topology ip6-addrs
id:46 rloc16:0xb800 ext-addr:96a1d6878b79b792 ver:4 - me - br
    3-links:{ 07 }
    2-links:{ 24 25 26 }
    ip6-addrs:
        fda0:d336:e5c3:cd0a:0:ff:fe00:fc10
        fda0:d336:e5c3:cd0a:0:ff:fe00:b800
        fda0:d336:e5c3:cd0a:a66e:6189:7425:b50b
        fe80:0:0:0:94a1:d687:8b79:b792
        fdb3:9162:9bb3:1:fa4c:7ba2:9177:32ee
id:07 rloc16:0x1c00 ext-addr:5a9380ccd45d9164 ver:4 - br
    3-links:{ 26 46 }
    2-links:{ 24 25 }
    ip6-addrs:
        fdb3:9162:9bb3:1:1f25:35de:6780:f014
        fda0:d336:e5c3:cd0a:0:ff:fe00:fc11
        fda0:d336:e5c3:cd0a:0:ff:fe00:fc38
        fda0:d336:e5c3:cd0a:0:ff:fe00:fc10
        fda0:d336:e5c3:cd0a:0:ff:fe00:1c00
        fda0:d336:e5c3:cd0a:a453:971e:f19d:ad3a
        fe80:0:0:0:5893:80cc:d45d:9164
id:25 rloc16:0x6400 ext-addr:521e27e6c018672e ver:4
    3-links:{ 07 24 26 46 }
    ip6-addrs:
        fdb3:9162:9bb3:1:935c:db6c:16c2:2c2
        fda0:d336:e5c3:cd0a:0:ff:fe00:6400
        fda0:d336:e5c3:cd0a:6d9e:388f:eca1:6966
        fe80:0:0:0:501e:27e6:c018:672e
id:24 rloc16:0x6000 ext-addr:fe91a0c8f1e8d7c6 ver:4 - leader
    3-links:{ 07 25 26 46 }
    ip6-addrs:
        fdb3:9162:9bb3:1:3145:5358:1143:efa6
        fda0:d336:e5c3:cd0a:0:ff:fe00:fc00
        fda0:d336:e5c3:cd0a:0:ff:fe00:6000
        fda0:d336:e5c3:cd0a:8bc3:929e:b00c:1de8
        fe80:0:0:0:fc91:a0c8:f1e8:d7c6
id:26 rloc16:0x6800 ext-addr:3a8b19d7144a7de3 ver:4
    3-links:{ 07 24 25 46 }
    ip6-addrs:
        fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02
        fda0:d336:e5c3:cd0a:0:ff:fe00:6800
        fda0:d336:e5c3:cd0a:e4c4:3dbd:ec2c:766b
        fe80:0:0:0:388b:19d7:144a:7de3
Done

root@openwrt:~# ot-ctl ping fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02
16 bytes from fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02: icmp_seq=4 hlim=64 time=19ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 19/19.0/19 ms.
Done

root@openwrt:~# ping -6 fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02 -c1
PING fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02 (fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02): 56 data bytes

^C
--- fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02 ping statistics ---
1 packets transmitted, 0 packets received, 100% packet loss
root@openwrt:~# ping -6 fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02
PING fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02 (fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02): 56 data bytes

^C
--- fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02 ping statistics ---
2 packets transmitted, 0 packets received, 100% packet loss

root@openwrt~# ip -6 route get fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02
fdb3:9162:9bb3:1:ff3e:ba39:36ff:be02 from :: via fe80::2a0:3ff:fe97:8a dev eth0 proto static src fd1b:6a0b:4b57:e874:8419:1690:c416:329a metric 512 pref medium

root@openwrt:~# ps|grep otbr-agent
 6566 root      5576 S    /usr/sbin/otbr-agent -d 7 -B br-lan -I wpan0 spinel+hdlc+uart:///dev/ttyCH343USB0?uart-baudrate=1000000&uart-flow-control trel://br-lan
14832 root      1260 S    grep otbr-agent
```
**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id [simplicity_sdk v2024.6.1](https://github.com/SiliconLabs/simplicity_sdk/releases/tag/v2024.6.1-0)
2. IEEE 802.15.4 hardware platform EFR32MG21F768

lancersky2017, 2024-09-06:
OK. I found a route missing. It worked after I added it. But I don't know why he lost it.
```
ip -6 route add fdb3:9162:9bb3:1::/64 dev wpan0  metric 256 
```

jwhui, 2024-09-06:
On posix, the ULA address is configured on the network interface here:
https://github.com/openthread/openthread/blob/e63e9ce86b4f22f95155e1e2e9da1d300389e7c6/src/posix/platform/netif.cpp#L431

Note that the on-link prefix is configured in the same function:
https://github.com/openthread/openthread/blob/e63e9ce86b4f22f95155e1e2e9da1d300389e7c6/src/posix/platform/netif.cpp#L451

Can you confirm that the correct ULA and on-link prefix is configured on the wpan0 interface?

lancersky2017, 2024-09-06:
In the log above, the `ot-ctl meshdiag topology ip6-addrs` command can show the IP information. And the `extaddr` of TBR is `96a1d6878b79b792`
I checked the wpan0 interface and there is no OMR address.
```
root@openwrt:~# ot-ctl ipaddr
fda0:d336:e5c3:cd0a:0:ff:fe00:fc10
fda0:d336:e5c3:cd0a:0:ff:fe00:b800
fda0:d336:e5c3:cd0a:a66e:6189:7425:b50b
fe80:0:0:0:94a1:d687:8b79:b792
fdb3:9162:9bb3:1:fa4c:7ba2:9177:32ee
Done
root@openwrt:~# ifconfig wpan0
wpan0     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet6 addr: fda0:d336:e5c3:cd0a:a66e:6189:7425:b50b/64 Scope:Global
          inet6 addr: fda0:d336:e5c3:cd0a:0:ff:fe00:fc10/64 Scope:Global
          inet6 addr: fda0:d336:e5c3:cd0a:0:ff:fe00:b800/64 Scope:Global
          inet6 addr: fe80::94a1:d687:8b79:b792/64 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1280  Metric:1
          RX packets:17187 errors:0 dropped:18 overruns:0 frame:0
          TX packets:249 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:1986339 (1.8 MiB)  TX bytes:47619 (46.5 KiB)
```

jwhui, 2024-09-06:
No OMR address configured on the wpan0 interface is why Linux ping is not working.

@bukepo , any ideas on how the netif and OT can get out of sync on the IPv6 address configuration?

closed 2025-05-21 by jwhui