openthread/openthread #10725: [Thread in Mobile] List of PRs and discussion

Damian-Nordic, 2024-09-18:
This issue is used to track all PRs and discussions related to the implementation of Thread in Mobile functionality defined in https://threadgroup.atlassian.net/browse/SPEC-1267 that enables direct communication between two sleepy devices.

The prototype of the functionality has been fully implemented and demonstrated on Nordic SoCs. The entire change set is available at https://github.com/nrfconnect/sdk-openthread/commit/collab-thread-s2s but it will be contributed in parts according to the below list to enable more effective code review process and API design discussions.

List of PRs (posted or planned; the list may evolve over time):
- Multipurpose wake-up frame support
- Dataset CSL Wake-up Channel TLV support
- Wake-up frame periodic sniffing
- Wake-up frame sequence transmission
- 100ms unit in child supervision interval/timeout
- MLE on wake-up frame reception (WED side)
- MLE on wake-up sequence transmission (WC side)
- eCSL support in POSIX platform

zhanglongxia, 2024-09-18:
The prototype requires the radio driver's support. Is there a guide for us to get the radio driver to build this prototype on nrf52840DK? And there are lots of compile errors when I build it on simulation platform via the command `./script/cmake-build simulation -DOT_MTD=ON -DOT_CSL_PERIPHERAL=ON -DOT_FTD=OFF -DOT_CSL_RECEIVER=ON`.

Damian-Nordic, 2024-09-18:
@zhanglongxia
Right, the prototype was developed using nRF Connect SDK - nRF radio bits have been pushed to https://github.com/nrfconnect/sdk-nrfxlib/commit/bc4e35b83096b60ccdc301aed2ccc8db8c702fa0, but you would also need Zephyr patches that haven't been published yet (we're working on that). I'll let you know when building the prototype is possible.

We will take care of the simulation errors while contributing subsequent pieces - so far we have only built it as a part of NCS and using the POSIX platform.

radua-nxp, 2024-12-13:
Has anybody built Wireshark with the Multipurpose frame parsing fix for windows? Or are there any 'nightly' builds?

radua-nxp, 2024-12-13:
I just found it; thanks and sorry to bother.

jwhui, 2026-11-13:
Closing as superseded. The earlier `SPEC-1267` wake-up and P2P prototype tracked here has been removed in favor of the updated Thread Direct (`[thread-direct]`) architecture.

closed 2026-11-13 by jwhui