openthread/openthread #10764: CLI command `dns resolve4` causes infinite loop of DNS queries, in NxDomain cases

EskoDijk, 2024-09-27:
**Describe the bug** A clear and concise description of what the bug is.
The command `dns resolve4 somename.org` locks the CLI and leads to an infinite loop of DNS queries to the BR. In case the DNS response RCode back to the client is 'NxDomain'.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: latest commit (tried in simulation) and ot-reference-release (tried on Nordic)
2. IEEE 802.15.4 hardware platform: both simulation (OTNS2) and Nordic NRF52840
3. Build steps: standard for OTNS2 / ot-reference-release respectively
4. Network topology: one BR with one Router attached

**Expected behavior** A clear and concise description of what you expected to happen.
No retry if the DNS query response is received by the client, even if the RCode indicates an error.

**Console/log output** If applicable, add console/log output to help explain your problem.
(can be added if needed)

**Additional context** Add any other context about the problem here.
Problem comes up in cert tests while issuing `dns resolve4 ipv4only.arpa` from the reference device client.

abtink, 2024-09-27:
Will take a look at it later when I get a chance. I will be less available today (Friday) and next week.

abtink, 2024-09-27:
I believe I've found the issues. It's a typo involving an incorrect query type. I've submitted a fix:

- [https://github.com/openthread/openthread/pull/10766](https://github.com/openthread/openthread/pull/10766)

I'd like to add a test for this as well, but I've submitted the PR for now to get the fix in place.

abtink, 2024-09-27:
Added a second commit that adds a test case on the same PR. Having two commits makes it easier to cherry-pick the fix.

EskoDijk, 2024-09-30:
@abtink Oh thanks, that's really super-quick!

EskoDijk, 2024-09-30:
@abtink After applying the fix to the OTBR reference-release nrf52840 dongle build, it works! Now a new issue came up which I've created as Github issue: https://github.com/openthread/openthread/issues/10770 
This was in context of debugging a cert test (11.5).

closed 2024-09-30 by jwhui

EskoDijk, 2024-10-02:
PR was merged, closing this issue