openthread/openthread #10772: [mebdtls] issue with generated `openthread-mbedtls-config.h`

abtink, 2024-10-01:
After updating MacOS to Sonoma 14.7 which also updated the clang toolchain. I noticed the following error message generated when building using `./test/toranj/build.sh all`:
```
FAILED: examples/apps/cli/ot-cli-mtd 
...
Undefined symbols for architecture arm64:
  "_mbedtls_ssl_conf_sig_algs", referenced from:
      ot::MeshCoP::SecureTransport::Setup(bool) in libopenthread-mtd.a[133](secure_transport.cpp.o)
ld: symbol(s) not found for architecture arm64
c++: error: linker command failed with exit code 1 (use -v to see invocation)
```

This points to a undefined function `mbedtls_ssl_conf_sig_algs()` which is used from `MeshCoP::SecureTransport::Setup(bool)`.

abtink, 2024-10-01:
The issue seems to be related with the generated `openthread-mbedtls-config.h`.

The `mebdtls/CMakeLists.txt` file contains the following:
https://github.com/openthread/openthread/blob/09698fae24521ab6f10f7df3fab5b1619f43bfca/third_party/mbedtls/CMakeLists.txt#L51-L65

This uses `unifdef` tool to remove `#if` checks from [third_party/mbedtls/mbedtls-config.h](https://github.com/openthread/openthread/blob/09698fae24521ab6f10f7df3fab5b1619f43bfca/third_party/mbedtls/mbedtls-config.h) and generate `openthread-mbedtls-config.h` which is then used as config header when building mbedtls library.

The issue is that in generated `openthread-mbedtls-config.h` we still see some of `#if` checks remaining:
```
#if OPENTHREAD_CONFIG_COAP_SECURE_API_ENABLE || OPENTHREAD_CONFIG_TLS_ENABLE
#define MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED
#endif
```
The issue seems to be related to limitation of `unifdef`:
```
     When evaluating an expression, unifdef does not expand macros first.  The
     value of a macro must be a simple number, not an expression.  A limited
     form of indirection is allowed, where one macro's value is the name of
     another.
```

In particular with the use of `OPENTHREAD_CONFIG_TLS_ENABLE` which is itself defined as:
```c++
/**
 * @def OPENTHREAD_CONFIG_TLS_ENABLE
 *
 * Define as 1 to enable support for TLS over TCP.
 */
#ifndef OPENTHREAD_CONFIG_TLS_ENABLE
#define OPENTHREAD_CONFIG_TLS_ENABLE (OPENTHREAD_CONFIG_TCP_ENABLE || OPENTHREAD_CONFIG_BLE_TCAT_ENABLE)
#endif
```

abtink, 2024-10-01:
Submitted the PR below to fix the build failure:
- [https://github.com/openthread/openthread/pull/10773](https://github.com/openthread/openthread/pull/10773)

However, it would be good to see if we can find a different (hopefully better) way to generate `mbedtls-config` that does not use `unifdef`.

jwhui, 2026-11-04:
Resolved by #10773 and #13826 (which removed the `unifdef` preprocessing step for `openthread-mbedtls-config.h`).

closed 2026-11-04 by jwhui