openthread/openthread #10775: Question about AIL link up/down event and TREL

Cristib05, 2024-10-03:
**Describe the bug** A clear and concise description of what the bug is.
Having an application which implements TREL functionality and also making use of OpenThread's native mDNS module to discover peers on the same adjacent network, what would be the correct behavior/implementation when a link up/down is reported by application?
Typically, I have used `otPlatInfraIfStateChanged()` so far, but when it comes to TREL module, should this be also handled inside this callback, or application has to use `otPlatTrelHandleDiscoveredPeerInfo()` and set `mRemoved` field to `true` so that `RemovePeerEntry` will be called for each instance in `trel_interface.cpp`. 
I was thinking that another option might be to call something like `Get<Trel::Interface>.HandleMdnsCoreStateChange()` in `Core::SetEnabled() `.
I am thinking that a link down event does not guarantee that link up will be performed on the same network, so in this case, TREL peer list will contain stale data.

If any of the proposed ideas is good, I could open a PR on this.

abtink, 2024-10-03:
Here are the options I suggest:

- Leave it as is and let the OT core multi-radio function detect that it cannot reach neighbors on the TREL link and switch accordingly. This should work independently but may cause some message drops.
- Have the platform implementation call `otPlatTrelHandleDiscoveredPeerInfo()` with `mRemoved = true` for each previously discovered peer. If the interface is up again later, platform would need to restart the mDNS query and report the discovered TREL peers again, which seems reasonable to ensure we don't keep stale information.

The `otPlatInfraIfStateChanged()` callback is (at least for now) primarily related to the `BORDER_ROUTING` function and I dont think will impact any TREL functionality. It might be good to keep it that way.

Another idea is to define a new `otPlatTrel` callback for the platform to signal to the OT core that the TREL interface is offline. This would essentially act as a shortcut to remove all peers instead of calling `otPlatTrelHandleDiscoveredPeerInfo()` for each one.

Cristib05, 2024-10-07:
Thanks for your response @abtink!

closed 2024-12-13 by Cristib05