openthread/openthread #10777: [dns-client] API and CLI support for DNS queries with arbitrary RRType

EskoDijk, 2024-10-03:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
Somewhat: 
* in cert tests (11.x) we like to send DNS queries for exotic RRTypes to be able to test a DNS server.
* in the future, some Thread product vendors may need to query the internet DNS for non-standard RRTypes (not A, not AAAA). 
   * e.g. TXT records for internet hostnames contain information about the domain such as unique IDs used for validation purposes. Future Thread devices may need such information. Or use other RRTypes like MX, SOA, OPENPGP, SIG/KEY etc.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
It would be nice to have a single API function to perform arbitrary RRType DNS queries by a client. This single API can be used for both production use and cert testing (because it's not a 'testing mode' as such - these queries are just allowed in normal use.)

This issue is a placeholder to propose/discuss the concept and solutions for this prior to making a PR. I'm not planning to submit a PR on the short term now (in October), but maybe later on. Or maybe someone else would like to do this :)
Do we agree this could be useful/desired?  Presumably it would not add to the build size if the API function isn't used by the node.

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
(for cert tests: more complicated sending of DNS queries using 'udp send -x' with a hex string)

**Additional context** Add any other context or screenshots about the feature request here.
-

abtink, 2024-10-03:
> Do we agree this could be useful/desired? Presumably, it would not add to the build size if the API function isn't used by the node.

From what I recall, we considered adding this before and decided it could add complexity and possible code overhead. While adding this sounds reasonable (at a minimum to allow more testing), I expect it not to be simple. It would likely require some design changes in `Dns::Client`, impacting the overall build size (regardless of whether we hide this feature behind a build config flag). `Dns::Client` was designed to be lean and lightweight, as it's intended to run on all categories of devices, including resource-limited ones.

I'll examine the code again to see if I can come up with any simple ideas.

If the goal is testing, another option is to implement or use a separate DNS client function that runs independently of OpenThread. Existing tools or implementations might already provide such functionality. Alternatively, as Esko suggested,  `udp send -x` with custom messages can be used.

abtink, 2025-03-24:
@abtink , submitted the following PR: 
- https://github.com/openthread/openthread/pull/11359

EskoDijk, 2025-03-28:
@abtink Thanks, I'm reviewing this now.

closed 2025-04-11 by jwhui