openthread/openthread #10783: [joiner] update joiner ID from eui64 on joiner start

OmegaRelay, 2024-10-04:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.

We're using OpenThread through the nRF connect sdk and we want to be able to set the extended address in runtime after OpenThread initialisation. We can update the network interface address using a variety of methods but none of these update the joiner ID which is only set during initialization or clearing the discerner.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
 
 - A C API function to call `Joiner::SetIdFromIeeeEui64()` something like: 
```c
void otJoinerSetIdFromIeeeEui64(void) { AsCoreType(aInstance).Get<MeshCoP::Joiner>().SetIdFromIeeeEui64(); }
```
**Or**
 - if mDiscerner is empty in `Joiner::Start` update `mId` with the ieeeEui64 like:
```c
Error Joiner::Start(...)
{

    ...

    if (!mDiscerner.IsEmpty())
    {
        SteeringData::CalculateHashBitIndexes(mDiscerner, filterIndexes);
    }
    else
    {
        // Update mId
        SetIdFromIeeeEui64();
        SteeringData::CalculateHashBitIndexes(mId, filterIndexes);
    }

    SuccessOrExit(error = Get<Mle::DiscoverScanner>().Discover(Mac::ChannelMask(0), Get<Mac::Mac>().GetPanId(),
                                                               /* aJoiner */ true, /* aEnableFiltering */ true,
                                                               &filterIndexes, HandleDiscoverResult, this));

    ...
}
```

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.

Currently we're using a workaround where we have to set the discerner with a dummy value and then clear it after setting our extended address to update the joiner id with the new eui64.

**Additional context** Add any other context or screenshots about the feature request here.

abtink, 2024-10-04:
Conceptually, the EUI64 returned from `otPlatRadioGetIeeeEui64()` should be fixed (hardcoded) and remain unchanged at run-time.

```c++
/**
 * Gets the factory-assigned IEEE EUI-64 for this interface.
 *
 * @param[in]  aInstance    The OpenThread instance structure.
 * @param[out] aIeeeEui64  A pointer to the factory-assigned IEEE EUI-64.
 */
void otPlatRadioGetIeeeEui64(otInstance *aInstance, uint8_t *aIeeeEui64);
```

My suggestion is to consider a change of the EUI64 a drastic change that triggers an OT stack re-initialization. Would this work? This would be my recommended solution.

Regarding the other suggestions:

-   Adding an API like `otJoinerSetIdFromIeeeEui64()` feels out-of-place and could lead to incorrect use (e.g., calling it while the discerner is in use would disrupt the joiner process).
-   Recalculcating `mId` from `Start()` seems like a better approach. We should be careful to change this in `GetId()` as well, so the returned ID valud is always calculated and consistent.

OmegaRelay, 2024-11-11:
Hi @abtink, Thanks for your reply, thanks for the suggestion to reinitialize the stack, we have implemented this and it works fine, thanks

closed 2024-11-11 by OmegaRelay