openthread/openthread #10784: [radio] clarify behavior of `otPlatRadioReceiveAt()`

abtink, 2024-10-04:
The `otPlatRadioReceiveAt()` function allows scheduling future receive windows and delegates timing to the radio platform.

```c++
/**
 * Schedule a radio reception window at a specific time and duration.
 *
 * @param[in]  aChannel    The radio channel on which to receive.
 * @param[in]  aStart      The receive window start time relative to the local
 *                         radio clock, see `otPlatRadioGetNow`. The radio
 *                         receiver SHALL be on and ready to receive the first
 *                         symbol of a frame's SHR at the window start time.
 * @param[in]  aDuration   The receive window duration, in microseconds, as
 *                         measured by the local radio clock. The radio SHOULD be
 *                         turned off (or switched to TX mode if an ACK frame
 *                         needs to be sent) after that duration unless it is
 *                         still actively receiving a frame. In the latter case
 *                         the radio SHALL be kept in reception mode until frame
 *                         reception has either succeeded or failed.
 *
 * @retval OT_ERROR_NONE     Successfully scheduled receive window.
 * @retval OT_ERROR_FAILED   The receive window could not be scheduled.
 */
otError otPlatRadioReceiveAt(otInstance *aInstance, uint8_t aChannel, uint32_t aStart, uint32_t aDuration);
```

This issue is to discuss and clarify how multiple calls to `otPlatRadioReceiveAt()` should work. Suppose `otPlatRadioReceiveAt()` is called and successfully schedules reception for a specific start time and duration. Here are some situations to consider:

1)  `otPlatRadioReceiveAt()` is called again before the previous start time.
2)  `otPlatRadioReceiveAt()` is called after the previous start time but within the previous window.
3)  `otPlatRadioReceiveAt()` is called after the end of the previously scheduled window.

My assumption is that in case (1), the new call to `otPlatRadioReceiveAt()` replaces the previous schedule, effectively canceling it.

Cases (2) and (3) are less clear. The question is how early after a previous schedule we can prepare the next one.

My suggestion for cases (2) and (3) is that since the previous start time has already passed, the previous receive schedule is already being executed by the radio and should not be replaced or impacted. The new call to `otPlatRadioReceiveAt()` would then schedule a future receive window.

Thoughts? @edmont, @Damian-Nordic  @LuDuda 

It would be interesting to know how existing implementations of this API in radio platforms behave in these cases.

abtink, 2024-11-06:
@LuDuda, @Damian-Nordic, @edmont, do you have any opinions or suggestions on this?

I would like us to clarify and document this in `radio.h` so we can assume consistent behavior from the radio platform (and help radio platform developers).

Damian-Nordic, 2024-11-07:
@abtink I totally agree with your suggestions.

Note that currently nRF does not implement (1), that is, an attempt to reschedule a slot that has not started yet will fail. The reason is that in a regular CSL this was simply not needed. Though in "Thread in Mobile" prototype, which must handle CST IEs, we had to do exactly what you suggested.

closed 2025-02-13 by jwhui