openthread/openthread #10787: Thread router failed to attach to the high-weighting Thread partition

jinpeng1989, 2024-10-05:
**Describe** At the beginning, a TBR and 6 REEDs are in the same Thread partition, which we will name **Thread partition A** for discussion purposes. Next, a REED creates a new **Thread partition B**, and other REEDs are attached to **partition B**, leaving only TBR in **partition A**. Partition A has a higher priority than Partition B. In theory, when the network stabilizes, REED in partition B should return to partition A, but it doesn't.
**Thread Partition A**: Weighting - 65 Partition ID - 0x6790d027
**Thread Partition B**: Weighting - 64 Partition ID - 0x73c71532
The Weighting value of Partition A is **65**, Partition B is **64**. since 65 is greater than 64, Partition A has a higher priority than Partition B. 

Wireshark packets: [merging-partition-issue.zip](https://github.com/user-attachments/files/17265514/merging-partition-issue.zip)
Network key: ffb392385d66960bdf1c71d8a68c98c2

Filter conditions:
**TBR2** (wpan.addr64 eq d6:9d:a1:2f:88:c8:c5:be) && (mle.cmd == 4)
**REED-1** (wpan.addr64 eq f2:7e:f0:7b:30:2a:8e:c9) && (mle.cmd == 4)
**REED-2** (wpan.addr64 eq 26:ba:3d:ee:93:4a:fc:84) && (mle.cmd == 4)
**REED-3** (wpan.addr64 eq aa:57:93:67:ab:87:1c:ef) && (mle.cmd == 4)
**REED-4** (wpan.addr64 eq 86:5c:52:3b:0a:f5:f1:b2) && (mle.cmd == 4)
**REED-5** (wpan.addr64 eq 0a:0d:ec:96:08:c6:66:b6) && (mle.cmd == 4)
**REED-6** (wpan.addr64 eq 86:a4:11:56:f1:4d:1a:67) && (mle.cmd == 4)

[image]

jinpeng1989, 2024-10-05:
Based on the previous analysis, the REEDs in partition B should try to attach to partition A. Yes, these REEDs are trying to attach to partition A. The following figure shows the REED-5 as an example. The REED-5 tried to attach to the TBR, and completed the full attachment handshake procedure. Strangely, REED-5 seemed to think it had failed to attach to TBR and kept retrying. One possibility is that TME checks the "Child ID Response" message returned by the TBR, determines that it is invalid, and aborts the attachment procedure. Can you help analyze it?

Filter conditions:
REED-1 (wpan.addr64 eq d6:9d:a1:2f:88:c8:c5:be) && (wpan.addr64 eq f2:7e:f0:7b:30:2a:8e:c9) && (mle.cmd == 11 || mle.cmd == 12)
REED-2 (wpan.addr64 eq d6:9d:a1:2f:88:c8:c5:be) && (wpan.addr64 eq 26:ba:3d:ee:93:4a:fc:84) && (mle.cmd == 11 || mle.cmd == 12)
REED-3 (wpan.addr64 eq d6:9d:a1:2f:88:c8:c5:be) && (wpan.addr64 eq aa:57:93:67:ab:87:1c:ef) && (mle.cmd == 11 || mle.cmd == 12)
REED-4 (wpan.addr64 eq d6:9d:a1:2f:88:c8:c5:be) && (wpan.addr64 eq 86:5c:52:3b:0a:f5:f1:b2) && (mle.cmd == 11 || mle.cmd == 12)
REED-5 (wpan.addr64 eq d6:9d:a1:2f:88:c8:c5:be) && (wpan.addr64 eq 0a:0d:ec:96:08:c6:66:b6) && (mle.cmd == 11 || mle.cmd == 12)
REED-6 (wpan.addr64 eq d6:9d:a1:2f:88:c8:c5:be) && (wpan.addr64 eq 86:a4:11:56:f1:4d:1a:67) && (mle.cmd == 11 || mle.cmd == 12)
[image]

jinpeng1989, 2024-10-05:
This TBR has a version TLV of 5, which is Thread version 1.3.1 or 1.4.0, and REEDs has a Thread version of 4, which is Thread version 1.3.0. 
Less than or equal to the 1.3.0 of the Thread specification (Thread version 4), the weight value of the Thread network partition is always 64, greater than or equal to the 1.3.1 of the Thread specification (Thread version 5), the weight of the Thread network partition has other values.
This was not observed when we used Thread version 4's TBR. This phenomenon was first observed using Thread version 5's TBR.
**So my first doubt is, does the old OpenThread stack have compatibility issues with the new weight value?**

abtink, 2024-10-05:
@jinpeng1989 , I spent some time reviewing the pcap you shared to help provide you with some information and next steps.

> So my first doubt is, does the old OpenThread stack have compatibility issues with the new weight value?

This is unlikely. The fact that the REED device does try to attach to the preferred partition with the higher weight indicates that the weight is being used correctly.

The issue appears to be that even though the REED devices go through the attachment steps (exchanging Parent Request/Response and even Child ID Request and Response), they seem to reject or drop the Child ID Response. I checked the captured Child ID Response messages, and they all look good and contain all the proper TLVs (note that Leader Data, which includes weight, does not matter during the attachment process).

As a next step, I suggest investigating why the REED device is not attaching successfully. It may be helpful to collect logs from these devices.

jinpeng1989, 2024-10-08:
This is the REED log. Although it is from another test system, it may be helpful for analysis.
`
[0;32mI(134148555) OPENTHREAD:[N] Mle-----------: Different partition (peer:1737543719, local:1942426930)[0m
[0;32mI(134148555) OPENTHREAD:[N] Mle-----------: Attempt to attach - BetterPartition [0m
[0;32mI(134164435) OPENTHREAD:[N] Mle-----------: Different partition (peer:1737543719, local:1942426930)[0m
[0;32mI(134164435) OPENTHREAD:[N] Mle-----------: Attempt to attach - BetterPartition [0m
[0;33mW(134165125) OPENTHREAD:[W] Platform------: Handle transmit done failed: ChannelAccessFailure[0m
[0;32mI(134174535) OPENTHREAD:[N] Mle-----------: Different partition (peer:1737543719, local:1942426930)[0m
[0;32mI(134174535) OPENTHREAD:[N] Mle-----------: Attempt to attach - BetterPartition [0m
[0;32mI(134185485) OPENTHREAD:[N] Mle-----------: Different partition (peer:1737543719, local:1942426930)[0m
[0;32mI(134185485) OPENTHREAD:[N] Mle-----------: Attempt to attach - BetterPartition [0m
`

[TME-ReceivedTofile-COM10-9_19_2024_2-59-24_PM.zip](https://github.com/user-attachments/files/17286985/TME-ReceivedTofile-COM10-9_19_2024_2-59-24_PM.zip)

jwhui, 2024-10-08:
> [TME-ReceivedTofile-COM10-9_19_2024_2-59-24_PM.zip](https://github.com/user-attachments/files/17286985/TME-ReceivedTofile-COM10-9_19_2024_2-59-24_PM.zip)

There are a lot of `NoBufs` errors:

```
[0;33mW(134439125) OPENTHREAD:[W] Platform------: Error processing result: NoBufs[0m
[0;33mW(134439125) OPENTHREAD:[W] Platform------: Error waiting response: NoBufs[0m
```

Have you tried using a different RCP from a different vendor?

jinpeng1989, 2024-10-08:
There are six REEDs in the system, all of them the same product, and they all perform the same. We will test it with another REED product to see if this issue occurs.

jwhui, 2024-10-08:
> There are six REEDs in the system, all of them the same product, and they all perform the same. We will test it with another REED product to see if this issue occurs.

Those kinds of `NoBufs` errors typically occur due to bugs in the platform implementation outside of OpenThread.

jwhui, 2025-01-07:
Closing stale issue.

closed 2025-01-07 by jwhui