openthread/openthread #10790: [mdns] Question related to Bonjour Conformance Test

Cristib05, 2024-10-07:
Hi,
In the context of mDNS conformance, we have found little to nothing in terms of conformance testing, except for Apple's test suite, Bonjour Conformance Testing.
Would this be something to take into account, I mean not necessarily to test OpenThread's mDNS implementation in such an environment described by this test suite, but in terms of functionalities that are implemented.
For example, we have seen that probe tie-breaking is not currently implemented, but TSR is mentioned as a possible future implementation, which makes more sense in the greater picture. Reply aggregation is as well, a possible missing feature.
Legacy Unicast, which has been added lately in this core code is also mentioned in this test suite.
Are those missing features something that you have taken into consideration for further implementation? Are contributions welcome on this topic?

abtink, 2024-10-07:
@Cristib05, contributions are always welcome. Providing more functions would definitely be helpful.

Regarding conformance testing, I'm unsure if it's possible and, if so, how easy or hard it would be. However, having certain conformance tests execute as part of the OT GitHub CI to validate the native mDNS implementation would be fantastic.

EskoDijk, 2024-11-12:
On TSR: it's considered for a future update for mDNS as sent by the Advertising Proxy, and for mDNS responses when querying by the Discovery Proxy. See IETF draft: https://datatracker.ietf.org/doc/html/draft-tllq-tsr 
This work should progress more in IETF before using it, I think.