openthread/openthread #10885: TBR fails to process IPv6 multicast packets

lancersky2017, 2024-10-31:
**Describe the bug** A clear and concise description of what the bug is.
Hi, Experts
We have a TBR product built on OpenWrt. The Thread `backboneIfName` is set to `br-lan`. 
In the following topology diagram, the test of TBR processing IPv6 multicast data fails. The multicast data sent by Ubuntu cannot be received by Thread ED, but the IPv6 multicast data sent by Thread ED can be received by Ubuntu.
```
	TBR
	/  \
   Ubuntu  Thread ED
```

Does it support such usage scenario?

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id **Silicon Labs GSDKv4.4.3**

lancersky2017, 2024-10-31:
The receiving and sending scripts run on Ubuntu are as follows,
```
# send_ipv6_multicast.py
import socket
import struct
import sys

def send_ipv6_multicast(message, multicast_group, port):
    sock = socket.socket(socket.AF_INET6, socket.SOCK_DGRAM)

    sock.setsockopt(socket.IPPROTO_IPV6, socket.IPV6_MULTICAST_HOPS, 1)

    message_bytes = message.encode('utf-8')

    sock.sendto(message_bytes, (multicast_group, port))
    
    print(f"Sent message: {message} to {multicast_group}:{port}")

    sock.close()

if __name__ == "__main__":
    multicast_group, port = sys.argv[1:]
    message = 'Hello, IPv6 Multicast!'
    send_ipv6_multicast(message, multicast_group, int(port))

# recv_ipv6_multicast.py
import ctypes
import ctypes.util
import socket
import struct
import sys

libc = ctypes.CDLL(ctypes.util.find_library('c'))
ifname, group = sys.argv[1:]
addrinfo = socket.getaddrinfo(group, None)[0]
assert addrinfo[0] == socket.AF_INET6
s = socket.socket(addrinfo[0], socket.SOCK_DGRAM)
s.bind(("::", 8888))
group_bin = socket.inet_pton(addrinfo[0], addrinfo[4][0])
interface_index = libc.if_nametoindex(ifname.encode('ascii'))
mreq = group_bin + struct.pack('@I', interface_index)
s.setsockopt(socket.IPPROTO_IPV6, socket.IPV6_JOIN_GROUP, mreq)
print("Subscribed %s on interface %s." % (group, ifname))
while True:
        data, addr = s.recvfrom(1024)
        print("Receive from ", addr, data)
        if data == b"exit":
                s.sendto(b"Good bye!\n", addr)
                continue
```

sunytt, 2024-10-31:
Hi,

What multicast address did you use to send packets from Ubuntu? Did the ED subscribe to the address?

lancersky2017, 2024-10-31:
Hi @sunytt 
I send the command under ubuntu,
```
python3 send_ipv6_multicast.py ff05::abcd 8888
```
And the ED had subscribe to the address.
```
> ipmaddr
ff05:0:0:0:0:0:0:abcd
ff33:40:fd7c:5552:245e:15b7:0:1
ff32:40:fd7c:5552:245e:15b7:0:1
ff02:0:0:0:0:0:0:2
ff03:0:0:0:0:0:0:2
ff02:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:fc
Done
```
I can catch IPv6 multicast packets on TBR br-lan interface. But it is not forwarded to wpan0. 
```
~# tcpdump -i br-lan ip6
09:50:27.654851 IP6 ddb4:242b:97db:0:739f:172f:3244:9a14.59239 > ff05::abcd.8888: UDP, length 22
```
I tried adding ip6tables rules but it didn't work.
```
-A FORWARD -i br-lan -o wpan0 -j ACCEPT
-A FORWARD -i wpan0 -o br-lan -j ACCEPT
```

lancersky2017, 2024-10-31:
The ED `netstat` command output,
```
> netstat 
| Local Address                                   | Peer Address                                    |
+-------------------------------------------------+-------------------------------------------------+
| [0:0:0:0:0:0:0:0]:8888                          | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:49154                         | [fd7c:5552:245e:15b7:75c2:2ad3:c4d4:4a09]:53536 |
| [0:0:0:0:0:0:0:0]:49153                         | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:61631                         | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:19788                         | [0:0:0:0:0:0:0:0]:0                             |
Done
```

sunytt, 2024-10-31:
Could you check the following output from the TBR cli?

```
> bbr state

> bbr mgmt mlr listener
```

lancersky2017, 2024-10-31:
The output as follow,
```
root@openwrt:~# ot-ctl bbr state
Primary
Done
root@openwrt:~# ot-ctl bbr mgmt mlr listener
Done
```

sunytt, 2024-10-31:
It looks like ED's multicast subscription did not get registered at the BBR, did the ED subscribe to the multicast address after TBR is up and running as a primary BBR?

lancersky2017, 2024-10-31:
Yeah. ED attaches to the network after TBR is successfully started.
Will the `ot-ctl bbr mgmt mlr listener` command output the IPv6 multicast addresses added by ED?
I installed ot-br-posix on Ubuntu. After otbr-agent was running properly, I added an IPv6 multicast address on ED by `ipmaddr` command. Then I used the command `sudo ot-ctl bbr mgmt mlr listener` on Ubuntu but did not see anything.

sunytt, 2024-11-01:
Yes, if the address is registered, it'll show up in `bbr mgmt mlr listener`.

Are the ED and TBR built with `-DOT_MLR=ON` option or have the `OPENTHREAD_CONFIG_MLR_ENABLE` defined to 1? This is needed to support the multicast listener registration feature.

lancersky2017, 2024-11-01:
When the ED turns on `OT_MLR` option, I can see the IPv6 multicast address registered on TBR. However, when IPv6 multicast packets are sent on Ubuntu, they cannot be received on ED.
```
### ED
> ipmaddr
ff05:0:0:0:0:0:0:abcd
ff33:40:fd53:785f:e20e:c3a4:0:1
ff32:40:fd53:785f:e20e:c3a4:0:1
ff02:0:0:0:0:0:0:2
ff03:0:0:0:0:0:0:2
ff02:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:1
ff03:0:0:0:0:0:0:fc
Done
> netstat
| Local Address                                   | Peer Address                                    |
+-------------------------------------------------+-------------------------------------------------+
| [0:0:0:0:0:0:0:0]:8888                          | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:49154                         | [fd53:785f:e20e:c3a4:5374:97de:363d:db09]:53535 |
| [0:0:0:0:0:0:0:0]:49153                         | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:61631                         | [0:0:0:0:0:0:0:0]:0                             |
| [0:0:0:0:0:0:0:0]:19788                         | [0:0:0:0:0:0:0:0]:0                             |
Done

### TBR
root@openwrt:~# ot-ctl bbr state
Primary
Done
root@openwrt:~# ot-ctl bbr mgmt mlr listener
ff05:0:0:0:0:0:0:abcd 3172
Done
root@openwrt:~# ip maddr
1:      lo
        inet  224.0.0.1
        inet6 ff02::1
        inet6 ff01::1
2:      eth1
        link  01:00:5e:00:00:01
        inet  224.0.0.1
        inet6 ff05::2
        inet6 ff01::2
        inet6 ff02::2
        inet6 ff02::1
        inet6 ff01::1
3:      eth0
        link  33:33:00:00:00:01
        link  33:33:00:00:00:02 users 2
        link  01:00:5e:00:00:01
        inet  224.0.0.1
        inet6 ff05::2
        inet6 ff01::2
        inet6 ff02::2
        inet6 ff02::1
        inet6 ff01::1
5:      br-lan
        link  33:33:00:00:00:01
        link  33:33:00:00:00:02
        link  01:00:5e:00:00:01
        link  01:00:5e:00:00:fb
        link  33:33:ff:00:00:01
        link  33:33:00:00:00:fb
        link  33:33:00:01:00:02
        link  33:33:00:01:00:03
        link  33:33:ff:2f:8a:45
        link  33:33:ff:00:00:00
        link  33:33:00:00:00:03
        link  33:33:00:00:ab:cd
        inet  224.0.0.251 users 2
        inet  224.0.0.1
        inet6 ff05::abcd
        inet6 ff32:40:fd53:785f:e20e:c3a4:0:3
        inet6 ff02::1:ff00:0 users 3
        inet6 ff02::1:ff2f:8a45 users 2
        inet6 ff05::1:3
        inet6 ff02::1:2
        inet6 ff02::fb users 2
        inet6 ff02::1:ff00:1
        inet6 ff02::2 users 2
        inet6 ff02::1
        inet6 ff01::1
6:      wlan0
        link  33:33:00:00:00:01 users 2
        link  33:33:00:00:00:02 users 2
        link  01:00:5e:00:00:01 users 2
        link  33:33:00:00:00:fb users 2
        link  33:33:ff:37:e6:b3 users 2
        link  33:33:ff:00:00:00 users 2
        inet  224.0.0.1
        inet6 ff02::1:ff00:0
        inet6 ff02::1:ff37:e6b3
        inet6 ff02::fb users 2
        inet6 ff02::2
        inet6 ff02::1
        inet6 ff01::1
7:      wpan0
        inet  224.0.0.1
        inet6 ff03::2
        inet6 ff33:40:fd53:785f:e20e:c3a4:0:1
        inet6 ff32:40:fd53:785f:e20e:c3a4:0:1
        inet6 ff03::fc
        inet6 ff03::1
        inet6 ff02::16
        inet6 ff02::2 users 2
        inet6 ff02::1 users 2
        inet6 ff01::1

### The Ubuntu can PING to TBR and ED
$ ping -6 fd16:6bdf:1d17:1:7d9e:2bd1:2e70:b895 -c1
PING fd16:6bdf:1d17:1:7d9e:2bd1:2e70:b895(fd16:6bdf:1d17:1:7d9e:2bd1:2e70:b895) 56 data bytes
64 bytes from fd16:6bdf:1d17:1:7d9e:2bd1:2e70:b895: icmp_seq=1 ttl=64 time=1.10 ms

--- fd16:6bdf:1d17:1:7d9e:2bd1:2e70:b895 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 1.100/1.100/1.100/0.000 ms

$ ping -6 fd16:6bdf:1d17:1:f077:ec71:31e3:aefd -c1
PING fd16:6bdf:1d17:1:f077:ec71:31e3:aefd(fd16:6bdf:1d17:1:f077:ec71:31e3:aefd) 56 data bytes
64 bytes from fd16:6bdf:1d17:1:f077:ec71:31e3:aefd: icmp_seq=1 ttl=63 time=34.5 ms

--- fd16:6bdf:1d17:1:f077:ec71:31e3:aefd ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 34.530/34.530/34.530/0.000 ms

```

sunytt, 2024-11-01:
I can see `ff05::abcd` gets added to br-lan, the multicast address subscription should be working.

In your script there is this line: `sock.setsockopt(socket.IPPROTO_IPV6, socket.IPV6_MULTICAST_HOPS, 1)`, this is setting the multicast packets with hoplimit=1, and will prevent packets from being forwarded. Can you change it to a larger number, 32 for example?

lancersky2017, 2024-11-01:
Yeah, It works! Thanks so much!

closed 2024-11-01 by lancersky2017