openthread/openthread #10903: Why echoed Address Registration TLV is not parsed by children?

edmont, 2024-11-06:
Maybe this has been discussed in the past but I can't find such discussion. There is such requirement in Thread spec:

> An MTD that registered a unicast address to the Parent, but the Parent did not echo this address back in the Address Registration TLV in the Child ID Response, SHOULD NOT use this unicast address.

However I can't see such code in MLE (`HandleChildIdResponse` or `HandleChildUpdateResponse`). Wouldn't be beneficial to stop using not echoed back addresses?

abtink, 2024-11-06:
I am not aware of any reason for this. Most likely, the behavior was not needed, as a parent would generally have enough space to accept all addresses, and therefore was not implemented on the child's side.

Adding this seems reasonable to me.

Do you want to add this? I can also help implement it later when I have some free cycles.

jwhui, 2024-11-07:
The reason is as what @abtink mentioned already - the assumption was that the parent could always store all of the child addresses. I agree, it's something we can add.

abtink, 2024-11-08:
> I was wondering if having this feature could have prevented [https://github.com/openthread/openthread/issues/10737](https://github.com/openthread/openthread/issues/10737).

Not necessarily. If the child detected an address registration failure with the parent and then stopped using the address (which is the missing behavior you proposed to add), then it would not have an OMR address and would not be able to communicate with AIL. So it would still fail but fail differently :).

There are a bunch of different issues raised within #10737. The relevant aspect to this discussion corresponds to [this analysis](https://github.com/openthread/openthread/issues/10737#issuecomment-2364719977) I did. This is already fixed by [[mle] enhance MTD child IPv6 address registration #10743](https://github.com/openthread/openthread/pull/10743), where we track and notice a lowpan context ID change and ensure the MTD child will register its addresses again upon this.

edmont, 2024-11-11:
> and then stopped using the address

That's the question about the missing behavior: do we need to just remove not echoed back addresses or also re-trigger `Slaac::AddAddresses()`?

abtink, 2024-11-11:
> do we need to just remove not echoed back addresses or also re-trigger Slaac::AddAddresses()?

My suggestion is to add a new mechanism to mark the addresses as "unregistered" (while keeping them in the `Netif` address list) and then update the "address selection" logic so the device won't pick these addresses as the source when sending an IPv6 message.

Removing the address can be somewhat disruptive, as we cannot know what added the address and it may be re-added. The origin of an address can be the SLAAC module provided by OpenThread, an external SLAAC implementation, or the address may be added manually by the user.