openthread/openthread #10926: Functions in the tcplp component may conflict with platform.

gytxxsy, 2024-11-15:
In the [tcplp](https://github.com/openthread/openthread/tree/main/third_party/tcplp) component , there are functions provided for OpenThread TCP. However, I noticed that some function names may conflict with platform implementations, such as `tcp_input`, `tcp_output` and `tcp_close`. As a result, when `OPENTHREAD_CONFIG_TCP_ENABLE` is enabled, certain compilation issues might occur.

marius-preda, 2024-12-11:
Hi @abtink and @jwhui, we faced the same issue when building OTBR with LWIP and enabling OT's TCP module so we had to do some link tricks to overcome the problem. It would be nice if we could find a better solution to this issue, like renaming those functions in OT's TCP module. Thoughts!

Cristib05, 2024-12-11:
There are also duplicate `#defines` between the two TCP stacks, would it be possible to guard OT's implementation defines?

jwhui, 2025-01-16:
I am OK to make changes to avoid these conflicts.

Does someone want to submit a PR with proposed changes?

marius-preda, 2025-01-16:
> I am OK to make changes to avoid these conflicts.
> 
> Does someone want to submit a PR with proposed changes?

Hi @jwhui,  I can submit a PR to address this. Thanks!

marius-preda, 2025-01-17:
Submitted #11174 to address this issue.

closed 2025-01-22 by jwhui